Koenkk/zigbee-herdsman #205: zigbee2mqtt stop working after a time

drJeckyll, 2020-08-08:
I have a couple of Ikea LED1649C5, LED1903C5/LED1835C6 and some buttons E1743. 

My problem is that after a while (may be when some device or couple of then go offline for whatever reason) whole mqtt2zigbee stop working.

Here is log:
```
zigbee2mqtt:info  2020-08-03 10:39:55: MQTT publish: topic 'zigbee2mqtt/bulb5', payload '{"state":"OFF","brightness":1,"update_available":false,"linkquality":99}'
zigbee2mqtt:info  2020-08-03 10:39:56: MQTT publish: topic 'zigbee2mqtt/bulb9', payload '{"state":"OFF","update_available":false,"brightness":1,"linkquality":94}'
zigbee2mqtt:info  2020-08-03 10:39:56: MQTT publish: topic 'zigbee2mqtt/bulb7', payload '{"state":"OFF","brightness":1,"update_available":false,"linkquality":97}'
zigbee2mqtt:info  2020-08-03 10:39:57: MQTT publish: topic 'zigbee2mqtt/bulb9', payload '{"state":"OFF","update_available":false,"brightness":1,"linkquality":97}'
zigbee2mqtt:info  2020-08-03 10:39:59: MQTT publish: topic 'zigbee2mqtt/bulb10', payload '{"state":"OFF","brightness":1,"update_available":false,"linkquality":97}'
zigbee2mqtt:info  2020-08-03 10:39:59: MQTT publish: topic 'zigbee2mqtt/bulb7', payload '{"state":"OFF","brightness":1,"update_available":false,"linkquality":92}'
zigbee2mqtt:info  2020-08-03 10:40:00: MQTT publish: topic 'zigbee2mqtt/bulb10', payload '{"state":"OFF","brightness":1,"update_available":false,"linkquality":99}'
``` 
Unitl this point everything seems to works fine. But after that:
```
zigbee2mqtt:info  2020-08-03 10:40:29: MQTT publish: topic 'zigbee2mqtt/bulb15/availability', payload 'offline'
zigbee2mqtt:error 2020-08-03 10:40:29: Failed to ping 'bulb15'
zigbee2mqtt:info  2020-08-03 10:40:30: MQTT publish: topic 'zigbee2mqtt/bulb13/availability', payload 'offline'
zigbee2mqtt:error 2020-08-03 10:40:30: Failed to ping 'bulb13'
zigbee2mqtt:info  2020-08-03 10:40:40: MQTT publish: topic 'zigbee2mqtt/bulb14/availability', payload 'offline'
zigbee2mqtt:error 2020-08-03 10:40:40: Failed to ping 'bulb14'
zigbee2mqtt:info  2020-08-03 10:40:49: MQTT publish: topic 'zigbee2mqtt/bulb11/availability', payload 'offline'
zigbee2mqtt:error 2020-08-03 10:40:49: Failed to ping 'bulb11'
zigbee2mqtt:info  2020-08-03 10:40:50: MQTT publish: topic 'zigbee2mqtt/bulb3/availability', payload 'offline'
zigbee2mqtt:error 2020-08-03 10:40:50: Failed to ping 'bulb3'
zigbee2mqtt:info  2020-08-03 10:40:50: MQTT publish: topic 'zigbee2mqtt/bulb4/availability', payload 'offline'
zigbee2mqtt:error 2020-08-03 10:40:50: Failed to ping 'bulb4'
zigbee2mqtt:info  2020-08-03 10:40:51: MQTT publish: topic 'zigbee2mqtt/router/availability', payload 'offline'
zigbee2mqtt:error 2020-08-03 10:40:51: Failed to ping 'router'
zigbee2mqtt:info  2020-08-03 10:40:53: MQTT publish: topic 'zigbee2mqtt/bulb12/availability', payload 'offline'
zigbee2mqtt:error 2020-08-03 10:40:53: Failed to ping 'bulb12'
zigbee2mqtt:info  2020-08-03 10:40:54: MQTT publish: topic 'zigbee2mqtt/bulb1/availability', payload 'offline'
zigbee2mqtt:error 2020-08-03 10:40:54: Failed to ping 'bulb1'
zigbee2mqtt:info  2020-08-03 10:40:55: MQTT publish: topic 'zigbee2mqtt/bulb5/availability', payload 'offline'
zigbee2mqtt:error 2020-08-03 10:40:55: Failed to ping 'bulb5'
zigbee2mqtt:info  2020-08-03 10:40:57: MQTT publish: topic 'zigbee2mqtt/bulb9/availability', payload 'offline'
zigbee2mqtt:error 2020-08-03 10:40:57: Failed to ping 'bulb9'
zigbee2mqtt:info  2020-08-03 10:40:59: MQTT publish: topic 'zigbee2mqtt/bulb6/availability', payload 'offline'
zigbee2mqtt:error 2020-08-03 10:40:59: Failed to ping 'bulb6'
zigbee2mqtt:info  2020-08-03 10:40:59: MQTT publish: topic 'zigbee2mqtt/bulb7/availability', payload 'offline'
zigbee2mqtt:error 2020-08-03 10:40:59: Failed to ping 'bulb7'
zigbee2mqtt:info  2020-08-03 10:41:00: MQTT publish: topic 'zigbee2mqtt/bulb10/availability', payload 'offline'
zigbee2mqtt:error 2020-08-03 10:41:00: Failed to ping 'bulb10'
zigbee2mqtt:info  2020-08-03 10:41:00: MQTT publish: topic 'zigbee2mqtt/bulb8/availability', payload 'offline'
zigbee2mqtt:error 2020-08-03 10:41:00: Failed to ping 'bulb8'
zigbee2mqtt:info  2020-08-03 10:41:20: MQTT publish: topic 'zigbee2mqtt/bulb2/availability', payload 'offline'
zigbee2mqtt:error 2020-08-03 10:41:20: Failed to ping 'bulb2'
zigbee2mqtt:info  2020-08-03 12:10:57: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"bulb6"}}'
zigbee2mqtt:error 2020-08-03 12:10:58: Failed to setup reporting for '0x000b57fffe8fc0dd' - Error: ConfigureReporting 0x000b57fffe8fc0dd/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":300,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (SREQ '--> AF - dataRequest - {"dstaddr":22285,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":255,"options":0,"radius":30,"len":11,"data":{"type":"Buffer","data":[16,47,6,0,0,0,16,0,0,44,1]}}' failed with status '(0x02: INVALID_PARAM)' (expected '(0x00: SUCCESS)'))
    at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:290:27)
    at Generator.next (<anonymous>)
    at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:24:58)
zigbee2mqtt:info  2020-08-03 12:10:58: MQTT publish: topic 'zigbee2mqtt/bulb6/availability', payload 'online'
zigbee2mqtt:error 2020-08-03 12:10:58: Failed to read state of 'bulb6' after reconnect
zigbee2mqtt:info  2020-08-03 12:11:58: MQTT publish: topic 'zigbee2mqtt/bulb6/availability', payload 'offline'
zigbee2mqtt:error 2020-08-03 12:11:58: Failed to ping 'bulb6'
zigbee2mqtt:info  2020-08-03 17:14:46: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"bulb11"}}'
zigbee2mqtt:error 2020-08-03 17:14:46: Failed to setup reporting for '0x90fd9ffffe0fb848' - Error: ConfigureReporting 0x90fd9ffffe0fb848/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":300,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (SREQ '--> AF - dataRequest - {"dstaddr":4583,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":105,"options":0,"radius":30,"len":11,"data":{"type":"Buffer","data":[16,152,6,0,0,0,16,0,0,44,1]}}' failed with status '(0x02: INVALID_PARAM)' (expected '(0x00: SUCCESS)'))
    at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:290:27)
    at Generator.next (<anonymous>)
    at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:24:58)
zigbee2mqtt:info  2020-08-03 17:14:46: MQTT publish: topic 'zigbee2mqtt/bulb11/availability', payload 'online'
zigbee2mqtt:error 2020-08-03 17:14:46: Failed to read state of 'bulb11' after reconnect
zigbee2mqtt:info  2020-08-03 17:15:46: MQTT publish: topic 'zigbee2mqtt/bulb11/availability', payload 'offline'
zigbee2mqtt:error 2020-08-03 17:15:46: Failed to ping 'bulb11'
zigbee2mqtt:info  2020-08-03 17:19:46: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"bulb11"}}'
zigbee2mqtt:info  2020-08-03 17:19:46: MQTT publish: topic 'zigbee2mqtt/bulb11/availability', payload 'online'
zigbee2mqtt:error 2020-08-03 17:19:46: Failed to read state of 'bulb11' after reconnect
zigbee2mqtt:info  2020-08-03 17:20:46: MQTT publish: topic 'zigbee2mqtt/bulb11/availability', payload 'offline'
[... 48 lines cut ...]
zigbee2mqtt:error 2020-08-06 02:26:09: Failed to ping 'bulb7'
zigbee2mqtt:info  2020-08-06 06:16:00: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"bulb8"}}'
zigbee2mqtt:error 2020-08-06 06:16:01: Failed to setup reporting for '0x90fd9ffffe0f8720' - Error: ConfigureReporting 0x90fd9ffffe0f8720/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":300,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (SREQ '--> AF - dataRequest - {"dstaddr":47224,"destendpo [... cut ...]
zigbee2mqtt:info  2020-08-06 06:41:01: MQTT publish: topic 'zigbee2mqtt/bulb8/availability', payload 'online'
zigbee2mqtt:error 2020-08-06 06:41:01: Failed to read state of 'bulb8' after reconnect
zigbee2mqtt:info  2020-08-06 06:42:01: MQTT publish: topic 'zigbee2mqtt/bulb8/availability', payload 'offline'
zigbee2mqtt:error 2020-08-06 06:42:01: Failed to ping 'bulb8'
```

It continues that way and mqtt2zigbee don't work any more until I stop it and run it again.

It's the latest production code from git.

Koenkk, 2020-08-08:
I don't know what adapter/firmware you are using, but the following can help to make z2m more stable:
- If using a CC2531/CC2530 try the source routing firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing (it provides more stability)
- Try disabling the availability feature since its very heavy for the network/adapter (https://www.zigbee2mqtt.io/information/availability.html#note-for-cc2530cc2531-users)
- If that doesn't help, it could be that your network is too large for the CC2530/CC2531, in that case you have to upgrade to a more powerful adapter (CC2652/CC1352 based https://www.zigbee2mqtt.io/information/supported_adapters.html#electrolama-zig-a-zig-ah-zzh)

drJeckyll, 2020-08-08:
I'm using CC2530 with default firmware - https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/default

Enabled availability only because zigbee2mqtt stopped work in same way while hopping things to get better, but they don't :)

I have a router too (CC2530 with router firmware - https://github.com/Koenkk/Z-Stack-firmware/tree/master/router/CC2530/bin), because I have 19 devices (13 LED1649C5, 2 LED1903C5/LED1835C6 and 4 buttons E1743).

MattWestb, 2020-08-09:
With 19 router devices and only 4 end devices i don't see need of having 2 extra cc-2530 routers that have lesser MCU performance and wors RF hardware then the Ikeas ICC-A-1  modules.
The CC-253X is not known being good router and can making problem then not forwarding all packages in the mesh and there is the weak point in your mesh. If the distance between the Ikea bulbs is enough for holding the mesh stabile (its depends of how much RF interference its around) i would remove the CC-2530 and if it's to long between the groups of bulbs for the mesh i would putting some Ikea routers (E1746) or control outlets (E1603) that have better RF and MCU preformas and they have the same zigbee stack that its eliminating mutch compatibility problems at the low level of the mesh.
If you is going for trying source routing then adding the "nearest" routers in itch direction from the coordinator first, so they is acting like spreader from the coordinator and from there the rest can connecting to them.

drJeckyll, 2020-08-09:
@MattWestb Actually bulbs are too close to each other - all in may be 25 sq. m. That may cause interference, not other devices around, because around is mostly open field :)
Only reason why I put a router was, because they don't want to pair after 13 bulb or so.

MattWestb, 2020-08-09:
First you is very lucky that not having 8 neighbors that having strong wifi all around you !!
The Zigbee mesh its not so aggressive as WiFi and other RF equipments and not making much interference. If it very dense you can getting one slowly network but not completely blocked.
Microwaves ovens can killing both WiFi and bluetooth completely if its near enough.
The second i think you have triggering one bug [#3177](https://github.com/Koenkk/zigbee2mqtt/issues/3177).
I dont have so much experience of Z2M (only testing with CC2531) but i think they have the same problem like deCONZ. I think its firmware and core code that making pairing failing. deCONZ needs a reboot after some time or some paired device for working. And i think Z2M lis behaving little like that.
You are using a CC2530 and it cant have all (up to 20) router device directly connected to it and its also better having the mesh "spreading out" and getting more child nodes. The mesh is making that self by time. And around 20 router devices in 25 m² its little dense. The bad thing it the CC2530 that its week in CPU, memory and RF power. 
Trying source routing firmware can helping (only 5 routers direct connected) and worth trying out. If trying it out you sud start pairing the bulbs that is nearest the coordinator first so they is direct connected with it and building the mesh out from it.
Having the CC-2530 router i think its only making all worth in you dense system and its very likely way you lost the contact with some of your bulbs.  You can try power the cc-2530 router of for some day and letting the mesh rebuilding and see how its working. And also don't mixing to much with different fabrikat then all have there problems and combining many different you is also getting more problems.

closed 2020-09-16 by github-actions[bot]