Koenkk/zigbee-herdsman #209: npm install --prod bloat from npm-shrinkwrap.json

trankin, 2020-08-12:
I've been doing some testing and working on packaging some applications with this library in it and found that the overall size of the application was about double the size I expected from similar projects.  I found that even though I was using the --prod flag on the npm install, there were a ton of what appeared to be development dependencies being installed... notably typescript was being installed for a whopping 50Mb.

Anyway, I as doing some research and I believe it comes down to the fact that this repository.. and zigbee-herdsman-converters and zigbee2mqtt for that matter are including an npm-shrinkwrap.json file which apparently triggers the installation of those dev dependencies as well.

I found a person performing similar research here: 
https://npm.community/t/devdependencies-installed-when-package-has-shrinkwrap-file/2244

as well as this open issue here:
https://github.com/npm/npm/issues/18776

I noted in reviewing the documentation @ docs.npmjs.com that it is strongly discouraged for library owners to package this file.
https://docs.npmjs.com/files/shrinkwrap.json

Regardless, I just wanted to point out my learnings, I hope to streamline the overall size of the install of zigbee2mqtt and would love your insight into the use of this file and whether we could consider it's removal.

Thanks for your consideration,

Thomas

trankin, 2020-08-12:
Here's a screenshot example of what I'm seeing that I believe is unexpected.   This occurs from an npm install --production in a bare zigbee2mqtt checkout.  

https://sharex.tomrankin.net/ShareX/2020/08/windirstat_djYBTtJvFW.png

Koenkk, 2020-08-13:
When using `npm ci --production` the node_modules folder is 30 MB which I think is fine.

```
zigbee2mqtt/node_modules on  dev [$] 
➜ du -sh
 30M    .
```

trankin, 2020-08-13:
I'll give it a shot.  Thanks.

trankin, 2020-08-13:
Yes, confirmed that gives me a much more reasonable installation size.  Still learning every day.  Thank you.

closed 2020-08-13 by trankin