Koenkk/zigbee-herdsman #216: Trying to add a new adapter

Florian-Krauss, 2020-09-04:
I'm currently working on implementing a new adapter/coordinator to zigbee2mqtt, that is neither from deconz nor from z-stack. Is there documentation on how I can do that or does anyone know what exactly needs to be defined in the code to make it work? 
The device I want to use as the adapter is an AP-515-RW from HPE aruba.

Koenkk, 2020-09-04:
To add support for a new adapter the following interface has to be implemented: https://github.com/Koenkk/zigbee-herdsman/blob/master/src/adapter/adapter.ts

E.g. implementation for z-stack: https://github.com/Koenkk/zigbee-herdsman/tree/master/src/adapter/z-stack

Florian-Krauss, 2020-09-07:
So it's possible to use the z-stack Code for a non-z-stack adapter?

Koenkk, 2020-09-07:
No, but you can use it as a new example when adding support for a new adapter.

Florian-Krauss, 2020-09-07:
Ok, I will try, thank you.

closed 2020-10-15 by github-actions[bot]