Koenkk/zigbee-herdsman #242: ZiGate implementation

G1K, 2020-10-21:
Discussion of the implementation of a new adapter based on the NXP chip JN516x ( JN5168 / JN5169 )

Perhaps this will solve the requests
https://github.com/Koenkk/zigbee2mqtt/issues/743
https://github.com/Koenkk/zigbee2mqtt/issues/409

Hedda, 2020-11-03:
Does https://github.com/Koenkk/zigbee-herdsman/pull/241 and https://github.com/Koenkk/zigbee2mqtt/pull/4737 being merged mean that zigbee-herdsman and zigbee2mqtt now works with all modules with ZiGate firmware including the original ones from https://zigate.fr by the @fairecasoimeme project from Frédéric Dubois?

Do not own a ZiGate or other NXP Zigbee coordinator myself but know it is supported by ZHA for Home Assistant via [zigpy-zigate](https://github.com/zigpy/zigpy-zigate/)

### Known working Zigbee radio modules
- [ZiGate USB-TTL](https://zigate.fr/produit/zigate-ttl/)
- [ZiGate USB-DIN](https://zigate.fr/produit/zigate-usb-din/)
- [PiZiGate (ZiGate module for Raspberry Pi GPIO)](https://zigate.fr/produit/pizigate-v1-0/)

Gamester17, 2020-11-03:
> Perhaps this will solve the requests
> [Koenkk/zigbee2mqtt#743](https://github.com/Koenkk/zigbee2mqtt/issues/743)
> **[Koenkk/zigbee2mqtt#409](https://github.com/Koenkk/zigbee2mqtt/issues/409)**

Also related to https://github.com/Koenkk/zigbee-herdsman/issues/150

Koenkk, 2020-11-03:
These changes are all available in Zigbee2MQTT 1.16.0

Hedda, 2020-11-30:
FYI, https://github.com/Koenkk/zigbee2mqtt.io/pull/474 has updated Zigbee2MQTT website documentation to list ZiGate USB-TTL as a supported adapter:

* https://www.zigbee2mqtt.io/information/supported_adapters.html

PS: Other ZiGate adapters not yet listed there, include ZiGate USB-DIN, PiZiGate, and ZiGate WiFi Pack -> https://zigate.fr/boutique/

Kalypox, 2020-12-14:
Hi everybody,

I'm trying to add my PiZiGate to Zigbee2MQTT, but i have errors:
```
Zigbee2MQTT:info  2020-12-14 16:38:29: Logging to console and directory: '/app/data/log/2020-12-14.16-38-28' filename: log.txt
Zigbee2MQTT:info  2020-12-14 16:38:29: Starting Zigbee2MQTT version 1.16.2-dev (commit #2468d61)
Zigbee2MQTT:info  2020-12-14 16:38:29: Starting zigbee-herdsman (0.13.41)
Zigbee2MQTT:error 2020-12-14 16:38:39: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2020-12-14 16:38:39: Failed to start zigbee
Zigbee2MQTT:error 2020-12-14 16:38:39: Exiting...
Zigbee2MQTT:error 2020-12-14 16:38:39: Error: failed to connect to zigate adapter [object Object] after 10000ms
    at ZiGateAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/zigate/adapter/zigateAdapter.js:125:23)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/adapter/zigate/adapter/zigateAdapter.js:25:65)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.16.2-dev start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.16.2-dev start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-12-14T15_38_39_702Z-debug.log
Using '/app/data' as data directory

> zigbee2mqtt@1.16.2-dev start /app
> node index.js
```

This is my configuration.yaml:
```
# Home Assistant integration (MQTT discovery)
homeassistant: true

# allow new devices to join
permit_join: true

# MQTT settings
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt
  # MQTT server URL
  server: 'mqtt://127.0.0.1:1883'
  # MQTT server authentication, uncomment if required:
  user: Jonathan
  password: xxxxxxxxxxxxx

# Serial settings
serial:
  # Location of CC2531 USB sniffer
  port: /dev/ttyS0
  adapter: zigate
```

devbis, 2020-12-14:
Hey @Kalypox 

Does the baud for your zigate adapter is equal to the one in herdsman? You should try to set the baudrate in the config file.

The second approach is to upgrade the firmware. It heavily relies on the raw Zigbee mode that is under development since 3.1c.
Maybe updating to the latest firmware can solve your problem.

Kalypox, 2020-12-14:
Hi @devbis,
Thanks you a lot !!!
It works, i added:
```
advanced:
  baudrate: 115200
```
[image]

Have a nice day !

Hedda, 2020-12-14:
Note! ZiGate 3.1d firmware or later is recommended for all ZiGate hardware is it contains an important bug-fix  

https://zigate.fr/tag/firmware/

G1K, 2021-01-13:
Added Roadmap to start post

Described current problems in https://github.com/Koenkk/zigbee-herdsman/issues/252#issuecomment-759430863 related to https://github.com/Koenkk/zigbee-herdsman/issues/285

G1K, 2021-01-28:
Most of the points from the roadmap are closed, but there is a problem with the stability of the module firmware when using raw mode in both directions. Perhaps they will be resolved, but for now I recommend trying to work with the firmware from this repository. 
https://github.com/openlumi/JN-ZigbeeNodeControlBridge-firmware/releases/latest
The firmware is collected on [Github Actions](https://github.com/openlumi/JN-ZigbeeNodeControlBridge-firmware/actions/runs/507959672) binaries are in artifacts

And she supports advanced logging, which can help in finalizing the adapter

G1K, 2021-01-28:
I do not have devices to check OTA, but in the sniffer, I see the correct communication that there is no update for the device. If anyone tests please unsubscribe.

G1K, 2021-03-02:
The latest improvements from the end of January are available in the 1.18 release

Hedda, 2022-01-18:
@Koenkk  Maybe re-open this issue for tracking and leave open even if no activity as ZiGate support is still listed as experimental?

Hedda, 2022-01-26:
hmm, why does github-actions bot close this?

Hedda, 2022-02-28:
Maybe permanently remove the stale label somehow?

G1K, 2022-03-02:
Most likely, you need to add the closing issue exempt-issue-labels to the action settings, specify a list of labels in it that do not need to be marked as obsolete. and for issue to use such a label

https://github.com/actions/stale#exempt-issue-labels

https://github.com/Koenkk/zigbee-herdsman/blob/ff2622550423db91ef70bc685f718e25aea39ea4/.github/workflows/stale.yml#L11-L18

Hedda, 2022-03-09:
@Koenkk Maybe add dont-stale label to this issue as well as it should probably also be keep open for the same reason as EZSP?

https://github.com/Koenkk/zigbee-herdsman/issues?q=is%3Aissue+is%3Aopen+label%3Adont-stale

pipiche38, 2023-02-06:
> Most of the points from the roadmap are closed, but there is a problem with the stability of the module firmware when using raw mode in both directions. Perhaps they will be resolved, but for now I recommend trying to work with the firmware from this repository. https://github.com/openlumi/JN-ZigbeeNodeControlBridge-firmware/releases/latest The firmware is collected on [Github Actions](https://github.com/openlumi/JN-ZigbeeNodeControlBridge-firmware/actions/runs/507959672) binaries are in artifacts
> 
> And it supports advanced logging, which can help in finalizing the adapter

Could you clarify which firmware should be use on a ZiGate V1 to act as a coordinator

G1K, 2023-02-18:
@pipiche38 
Any depending on your type of chip and Baudrate for Zigate sticks most likely should be 5168 and 115200. Just do not take the versions with GP. The firmware JN chips looks at magic bats at the beginning of the firmware and should not let the firmware of 5169 be flashed in 5168 and vice versa.
We use firmware on Xiaomi Gateway, there is a chip 5169 as in Xiaomi end devices

But Zigate V1 firmware has seemed to be corrected and work as it should. The only problem that strongly interferes is the absence of PDM backup and periodically the chip reports that after rebooting it is Factory reseted

In the firmware
https://github.com/openlumi/JN-ZigbeeNodeControlBridge-firmware/releases/latest
https://github.com/Alx2000y/zigbee-herdsman/compare/068bbe7636f588394f69f82bc25c8b68a4feada7...fd707e4f8e4814adf88ca91959cc35984db59c24

Realized raw PDM backup memory of NXP JN516x zigbee chip and recovery. Those who wish can try.

pipiche38, 2023-02-18:
I'll put it on my todo list, this looks promising and could be a good alternative to the zigate firmware

Hedda, 2023-02-21:
> ## Road map
> ...
> * [ ]  * Backup and restore ( [mvp](https://github.com/Koenkk/zigbee-herdsman/issues/242#issuecomment-1435711120) )

@G1K Do you have any updates on the backup and restore features for zigate adapter in zigbee-herdsman?

PS Requestion is related to this discussion about cross-hardware migrations -> https://github.com/Koenkk/zigbee2mqtt/discussions/16478

closed 2026-11-03 by Nerivec