Koenkk/zigbee-herdsman #249: Incorrect Tuya protocol fields

drzony, 2020-11-06:
Hi,

After getting some specs from manufacturer and looking at Tuya pages I've found out that current Tuya protocol fields are incorrect, the `dp` field should be `uint8` not `uint16`, the first byte is type of data as specified [here](https://developer.tuya.com/en/docs/iot/device-development/access-mode-mcu/zigbee-general-solution/tuya-zigbee-module-uart-communication-protocol/tuya-zigbee-module-uart-communication-protocol?id=K9ear5khsqoty#title-8-Command%20issued)

Over the weekend (and maybe some of the next week) I'm willing to do a sizeable Tuya rework:
1. Rename the endpoint from `manuSpecificTuyaDimmer` to just `manuSpecificTuya`
2. Fix the endpoint fields
3. Create a list of data points in converters common.ts from all Tuya devices
4. Modify Tuya commands to use proper `data type` and `data point` combo
5. Add full support for Saswell TRVs

Are you willing to accept such a big change? I'm willing to provide additional support in case of problems with my change

Koenkk, 2020-11-07:
Yes please create a pr for this. But it should be tested well.

drzony, 2020-11-09:
I have tested this with Saswell Thermostats and Tuya T&H sensor, unfortunately I don't have access to other Tuya compatible devices.
All data points seem to come together nicely after this rework, so there should be no problems

drzony, 2020-11-09:
I've also added 2 converters that are not used by default on any devices tuya_data_point_dump and tuya_data_point_test, those can be used to help integrate new Tuya compatible devices. Also a python script is included that parses the dump

closed 2020-11-10 by Koenkk

Faustino-ai, 2021-04-05:
Hello @drzony , I am writing here because I am really out of ways to implement Tuya smart door lock. I've followed every single step in the doc https://www.zigbee2mqtt.io/how_tos/how_to_support_new_tuya_devices.html so please help me out a little. I've found out that my smart lock doesn't respond to tuya_data_point_test and tuya_data_point_dump doesn't receive anything as well. I am suspecting this is due to the format being different. I've posted my issues but so far no one was able to help. Please kindly have a look and help me out. Thank you. 
https://github.com/Koenkk/zigbee2mqtt/issues/6835

PR #253 (2020-11-10): Fixed definition of manuSpecificTuya cluster
Fixes #249