Koenkk/zigbee-herdsman #252: Not correct join Gaz Detector SGMHM-I1 via Zigate in Iobroker...

itProfi, 2020-11-08:
Good day! I try join device Gaz Detector SGMHM-I1 via Zigate in Iobroker...and have some errors..
How i can correct this situation?
[image]
[image]
[image]
Аfter restart adapter in Iobroker, device like ok, but not work properly...
[iobroker.2020-11-08.log](https://github.com/Koenkk/zigbee-herdsman/files/5506125/iobroker.2020-11-08.log)

itProfi, 2020-11-08:
Same problem with smart siren HS2WD-E (it's not "vendor":"HEIMAN", it's russian market Tricolor)
[image]
[image]
[iobroker.2020-11-08_siren.log](https://github.com/Koenkk/zigbee-herdsman/files/5506136/iobroker.2020-11-08_siren.log)

Koenkk, 2020-11-08:
With what adapter is this?

itProfi, 2020-11-08:
JN5169 with zigate coordinator firmware

Koenkk, 2020-11-08:
The `Cannot read property length of undefined` indicates an error in the adapter code. @G1K or @kirovilya can you look at this?

itProfi, 2020-11-09:
@Koenkk In my case, i see, that error is in OTA Update...how i can disable temporary  this?

Koenkk, 2020-11-09:
This functionality cannot be disabled currently.

itProfi, 2020-11-09:
> This functionality cannot be disabled currently.

And how i can correct this error?

Koenkk, 2020-11-09:
I expect some fixes are required in the adapter code (https://github.com/Koenkk/zigbee-herdsman/issues/252#issuecomment-723579075)

itProfi, 2020-11-09:
Today i try only xiaomi gateway with jn5169 and zigbee2mqtt and thats all! Not IOB..I allready ask @kirovilya , he don't no, what do with this error. I upgrdae zigbee2mqtt to 1.6.1 and give another error - 
[image]
[image]

itProfi, 2020-11-12:
@Koenkk @kirovilya says, that he don't no, what's the problem...last error is 
[image]
 in Iobroker

itProfi, 2020-12-13:
@Koenkk is it possible correct this?

Koenkk, 2020-12-13:
@itProfi I cannot fix this, has to be fixed in the adapter code (https://github.com/Koenkk/zigbee-herdsman/issues/252#issuecomment-724129233)

itProfi, 2020-12-14:
@Koenkk No, i has lastest code for this adapter and it's not work. @kirovilya says, that he not correct this for at least one more month (( If you want, close this issue..

G1K, 2021-01-13:
I am working on this problem. This is where I came to.

1. We need a separate queue on the APDU, when using a common queue and waiting for a message 0x8000, messages are partially sent to the network. 
2. When pairing, a lot of requests are often generated, for example, on FreePad 20 endpoint + 20 bind. If you use a sequential queue and wait for a message from the network, it works more stable, but still, when using equal requests, the adapter hangs and does not allow messages to be sent to the network. https://github.com/fairecasoimeme/ZiGate/issues/348 (This was always the case before firmware 3.1d for incoming messages). The solution is still unknown, I can only suggest rebooting the adapter, but the number of instances for different firmware is different and I have not yet figured out how to find out when they are over. 
3. The fix of expectations from this PR does not wait for an answer, although it is needed and the herdsman wants to get a result.  https://github.com/Koenkk/zigbee-herdsman/pull/247/files#diff-12256257d3beef78fe73fdda606b50fbb139127f642a50b74894e21eff30414aR546-R547

As I finish the implementation I will do a PR, while there are only temporary sketches.

// I want to make the most of the adapter, and I don't want to put all messages in the general queue if one of the devices falls asleep and does not receive messages.

itProfi, 2021-01-26:
@G1K all do!

closed 2021-01-26 by itProfi