Koenkk/zigbee-herdsman #263: Q: Ikea tradfri repeater repeats any device?

boesing, 2020-11-24:
Hey there, 

I've bought the ikea tradfri repeater to extend my zigbee network. I got it connected but it seems it does not property connect my osram smart+ led strip.
There are 5 Meters between the repeater and the coordinator aswell as 5 Meter between osram light strip.

LQI from repeater is around 80 but the light strip is being unavailable for some time and when available, its around 6 LQI.

So is there probably some kind of incompatibility between ikea and osram?

Thanks in advance

MattWestb, 2020-11-25:
The IKEA devices have pretty good radio and PCB antennas. "old" OSRAM devices have week chips (CC-2530) and also pretty bad antennas (at least the plugs but not verified with all devices). Its also happens that there is corrupting packages so losing connection with other devices.
But if you is having LQI of 6 its nearly possible communicating with it only with very much luck.
The IKEA repeater is trying working as one zigbee 3 router but its depends wat coordinator you having if its working in zigbee 3 mode. In zigbee 3 mode its using all possible links to other routers and using the best. In zigbee 2.X its more only using the strongest router and cant reroute the traffic so good and the OSRAM is very likely not zigbee 3 and dont using the advance routing.
If you having on good coordinator that the repeater can talking with i wold true putting it nearer the led-strips (like coordinator - 7m - repeter- 3m- led-stip) if its possible and then repairing the led-stip so its shusing the stronges router in its near.  Some users with OSRAM smart plug was putting some bulbs as router on the side for getting then working more stable.

On the last Q: All devices have more or less problem with them self and other devices but sadly OSRAM have not fixing their firmware that have nasty bugs and therefore have more then average problems exceptionally with Xiaomi sensors. In the end no devices on the market is 110% good because mother nature is one b . . . h

boesing, 2020-11-25:
My Zigbee network is on v3. 
My Coordinator is a ZZH! stick (based on CC2652 chip).

Yes, I am losing connection to the LED Strip regularly... Thats why I wanted to extend the network by using a repeater (I've also a TUYA Thermostat next to the LED Strip which does not act as a repeater aswell...).

Gonna try to put the repeater more close to the strip. I wasn't aware of re-pairing when using a router. That might help, thanks!

boesing, 2020-11-25:
Nice! Re-pairing the devices I've had low signal strength' worked! Thank you very much!

closed 2020-11-25 by boesing

MattWestb, 2020-11-25:
The ZZH! sould having good antenna so perhaps putting the repeater very near the OSRAM stripes.
If its working as one router and not as one end device (zigbee 2 / 3 compatibility thing) it sould talking with all possible routers and using the best one for communicating.

I have putting one IKEA control outlet in my kitchen that working as router for My tuya TRV, one 2bt remote for the outlet, lumi weather outside the window and one more on the other side the wall in the bedroom and all is working good.

I think the problem is beside bad antenna that the OSRAM not working in zigbee 3 mode and is acting more or less as one end device and not one real router.

Great that you was getting it working and I think its also better for the TRV having one router in the near.

boesing, 2020-11-25:
> I think the problem is beside bad antenna that the OSRAM
Yeah, all OSRAM antennas are annoying. But even my power outlet has better LQI when re-pairing (as it connects to the repeater aswell).

The TRV has better LQI aswell (even without re-pairing, I guess due to their support of Zigbee 3.0).

I just wasnt aware that re-pairing might help. I am just not yet that deep into Zigbee tho.

MattWestb, 2020-11-25:
In my bath with one control outlet 2 Btm remote and one aqara weather.
[image]
(yesterday it was having battery hookup)