Koenkk/zigbee-herdsman #268: herdsman for Deconz hangs in permitJoin()

lxh7, 2020-12-02:
Zigbee2mqtt "hangs" during startup. 

I've added a few additional log lines in various modules to see how far the startup was processed (see below). 
It seems it hangs on the "yield" (I'm a programmer but not very familiar with JavaScript)

Note: I've done a fresh install of zigbee2mqtt on 2020-12-01 and flashed the ConBee II with the most recent firmware to make sure I didn't hit an old bug, but alas no... 

Any suggestions? Please help...

kind regards,
Alex

This is my log (with my own temporary debug entries in it):

```
> zigbee2mqtt@1.16.2 start /opt/zigbee2mqtt
> node index.js

Zigbee2MQTT:debug 2020-12-02 10:59:28: Loaded state from file /opt/zigbee2mqtt/data/state.json
Zigbee2MQTT:info  2020-12-02 10:59:28: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-12-02.10-59-28' filename: log.txt
Zigbee2MQTT:debug 2020-12-02 10:59:28: Removing old log directory '/opt/zigbee2mqtt/data/log/2020-12-01.23-24-10'
Zigbee2MQTT:info  2020-12-02 10:59:28: Starting Zigbee2MQTT version 1.16.2 (commit #04c15f7)
Zigbee2MQTT:info  2020-12-02 10:59:28: Starting zigbee-herdsman (0.13.37)
Zigbee2MQTT:debug 2020-12-02 10:59:28: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyACM0"}}'
CREATED DECONZ ADAPTER
Zigbee2MQTT:info  2020-12-02 10:59:28: zigbee-herdsman started
Zigbee2MQTT:info  2020-12-02 10:59:29: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":104,"product":0,"revision":"0x26680700","transportrev":0},"type":"ConBee2/RaspBee2"}'
Zigbee2MQTT:debug 2020-12-02 10:59:32: Zigbee network parameters: {"channel":0,"extendedPanID":"0xdddddddddddddddd","panID":6754}
Zigbee2MQTT:info  2020-12-02 10:59:32: Currently 0 devices are joined:
Zigbee2MQTT:warn  2020-12-02 10:59:32: `permit_join` set to  `true` in configuration.yaml.
Zigbee2MQTT:warn  2020-12-02 10:59:32: Allowing new devices to join.
Zigbee2MQTT:warn  2020-12-02 10:59:32: Set `permit_join` to `false` once you joined all devices.
Zigbee2MQTT:info  2020-12-02 10:59:32: Zigbee: allowing new devices to join.
Decons.permitJoin() entered
__awaiter callback
calling driver.enqueueSendDataRequest
entering deconz/driver.enqueueSendDataRequest
Promise created
req created
pushed to apsQueue
^CZigbee2MQTT:debug 2020-12-02 10:59:44: Saving state to file /opt/zigbee2mqtt/data/state.json
Zigbee2MQTT:error 2020-12-02 10:59:44: Not connected to MQTT server!
Zigbee2MQTT:error 2020-12-02 10:59:44: Cannot send message: topic: 'zigbee2mqtt/bridge/state', payload: 'offline
Zigbee2MQTT:info  2020-12-02 10:59:44: Disconnecting from MQTT server
(node:32710) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'end' of undefined
    at MQTT.disconnect (/opt/zigbee2mqtt/lib/mqtt.js:93:21)
    at async Controller.stop (/opt/zigbee2mqtt/lib/controller.js:198:9)
(node:32710) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:32710) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
^CTerminated
```
(please note that the first Ctrl+C is handled partially (see entries on 2020-12-02 09:46:06), but the process hangs. An additional Ctrl+C Terminates something, but the process Node has to be killed with kill -9 'process id'

This is the permitJoin() function in deconzAdapter.js with my additional logging:
```
 permitJoin(seconds, networkAddress) {
console.log("Decons.permitJoin() entered");
        return __awaiter(this, void 0, void 0, function* () {
console.log("__awaiter callback");
            const transactionID = this.nextTransactionID();
            const request = {};
            const zdpFrame = [transactionID, seconds, 0]; // tc_significance 1 or 0 ?
            request.requestId = transactionID;
            request.destAddrMode = constants_1.default.PARAM.addressMode.NWK_ADDR;
            request.destAddr16 = networkAddress || 0xFFFC;
            request.destEndpoint = 0;
            request.profileId = 0;
            request.clusterId = 0x36; // permit join
            request.srcEndpoint = 0;
            request.asduLength = 3;
            request.asduPayload = zdpFrame;
            request.txOptions = 0;
            request.radius = constants_1.default.PARAM.txRadius.DEFAULT_RADIUS;
            request.timeout = 5;
            try {
console.log("calling driver.enqueueSendDataRequest");
                yield this.driver.enqueueSendDataRequest(request);
console.log("driver.enqueueSendDataRequest returned");
                if (seconds === 0) {
                    this.joinPermitted = false;
                }
                else {
                    this.joinPermitted = true;
                }
console.log("driver.writeParameterRequest");
                this.driver.writeParameterRequest(constants_1.default.PARAM.Network.PERMIT_JOIN, seconds);
                debug("PERMIT_JOIN - " + seconds + " seconds");
            }
            catch (error) {
                debug("PERMIT_JOIN FAILED - " + error);
                // try again
console.log("error caught: " + error);
                this.permitJoin(seconds, networkAddress);
                //return Promise.reject(); // do not reject
            }
        });
    }
```
This is the enqueueSendDataRequest() function in driver.js with my additional logging:
```
    enqueueSendDataRequest(request) {
console.log("entering deconz/driver.enqueueSendDataRequest");
        const seqNumber = this.nextSeqNumber();
        return new Promise((resolve, reject) => {
console.log("Promise created");
            debug(`push enqueue send data request to apsQueue. seqNr: ${seqNumber}`);
            const ts = 0;
            const requestId = request.requestId;
            const commandId = constants_1.default.PARAM.APS.DATA_REQUEST;
            const req = { commandId, seqNumber, request, resolve, reject, ts };
console.log("req created");
            apsQueue.push(req);
console.log("pushed to apsQueue");
        });
console.log("unreachable: after return in deconz/driver.enqueueSendDataRequest");
    }
```

Environment: Linux Debian on AMD64, i.e. "PC hardware", Deconz ConBee II. Plenty of resources (RAM and disk).
ConBee II on USB extension cable or directly connected to the PC makes no difference.

Koenkk, 2020-12-02:
@ChrisHae can you check this?

lxh7, 2020-12-07:
The ConBee II was configured as a router.... I'll test later today if that solves the issue. 
It would be nice if the device driver had a check for that ;-)

lxh7, 2020-12-07:
zigbee2mqtt now proceeds and even connects to the mqtt server.

closed 2020-12-07 by lxh7