Koenkk/zigbee-herdsman #271: Cannot connect to ConbeeII

matzesutopia, 2020-12-07:
It doesn't matter which serial port I try, but the lamp will not change to green. I already try 

/dev/ttyS0
/dev/ttyACM0
/dev/ttyUSB0

It will not connect to my Conbee 2 stick. I deactivated the deconz adapter.

2020-12-07 23:57:49.420 - info: zigbee.0 (10039) starting. Version 1.4.0 in /opt/iobroker/node_modules/iobroker.zigbee, node: v14.15.1, js-controller: 3.1.6
--
2020-12-07 23:57:49.480 - info: zigbee.0 (10039) Starting Zigbee npm ...
2020-12-07 23:57:49.655 - info: zigbee.0 (10039) Installed Version: iobroker.zigbee@1.4.0
2020-12-07 23:58:00.872 - error: zigbee.0 (10039) Failed to start Zigbee
2020-12-07 23:58:00.875 - error: zigbee.0 (10039) TIMEOUT

2020-12-07 23:58:10.876 - info: zigbee.0 (10039) Try to reconnect. 1 attempts left
--
2020-12-07 23:58:10.878 - info: zigbee.0 (10039) Starting Zigbee npm ...
2020-12-07 23:58:10.902 - info: zigbee.0 (10039) Installed Version: iobroker.zigbee@1.4.0
2020-12-07 23:58:11.032 - error: zigbee.0 (10039) uncaught exception: Port is not open
2020-12-07 23:58:11.033 - error: zigbee.0 (10039) Error: Port is not open
at unixWrite (/opt/iobroker/node_modules/iobroker.zigbee/node_modules/@serialport/bindings/lib/unix-write.js:25:15)
at async /opt/iobroker/node_modules/iobroker.zigbee/node_modules/@serialport/bindings/lib/linux.js:72:7
2020-12-07 23:58:11.036 - info: zigbee.0 (10039) cleaned everything up...
2020-12-07 23:58:11.042 - info: zigbee.0 (10039) Zigbee: disabling joining new devices.
2020-12-07 23:58:11.231 - error: zigbee.0 (10039) Failed to start Zigbee
2020-12-07 23:58:11.233 - error: zigbee.0 (10039) Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'
at Driver. (/opt/iobroker/node_modules/iobroker.zigbee/node_modules/zigbee-herdsman/dist/adapter/deconz/driver/driver.js:153:28)
at Generator.next ()
at /opt/iobroker/node_modules/iobroker.zigbee/node_modules/zigbee-herdsman/dist/adapter/deconz/driver/driver.js:8:71
at new Promise ()
at __awaiter (/opt/iobroker/node_modules/iobroker.zigbee/node_modules/zigbee-herdsman/dist/adapter/deconz/driver/driver.js:4:12)
at SerialPort. (/opt/iobroker/node_modules/iobroker.zigbee/node_modules/zigbee-herdsman/dist/adapter/deconz/driver/driver.js:151:45)
at SerialPort._error (/opt/iobroker/node_modules/iobroker.zigbee/node_modules/@serialport/stream/lib/index.js:198:14)
at /opt/iobroker/node_modules/iobroker.zigbee/node_modules/@serialport/stream/lib/index.js:242:12
2020-12-07 23:58:11.561 - info: zigbee.0 (10039) terminating
2020-12-07 23:58:11.563 - info: zigbee.0 (10039) Terminated (NO_ERROR): Without reason

ndfred, 2020-12-21:
Try specifying the exact right port, for me that is `/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2215203-if00`

smulle48, 2021-01-18:
Try this command the see if it shows up

`ls -l /dev/serial/by-id`

smulle48, 2021-01-19:
(10039) Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'

^^ either you have chosen the wrong port or something else is still using it.

Try to pull it out and put it in another USB port without rebooting to make sure it's not occupied by another process

MattWestb, 2021-01-19:
Modem manager installed that is occupying the compost ??

closed 2021-03-30 by github-actions[bot]