Koenkk/zigbee-herdsman #273: deCONZ driver has a 60ms delay when reading a button press

ndfred, 2020-12-16:
I am trying to make my lights as responsive as possible. However, when I press the button on any of my remotes there is a noticeable delay for the press to appear in the logs. The button I am using here is bound to a light, and the light turns on noticeably quicker than the Z2M logs show.

I pulled some debug logs and you can clearly see the sequence here:

* we get a `DEVICE_STATE` frame from the button indicating something happened, that gets logged exactly when I press the button
* we send the read data request to the button to (I guess) know exactly what happened 79ms after that
* 6ms later the button answers with what changed, and Z2M logs that as a button press

Basically we introduce an artificial 80ms delay when reading button clicks. Looking at [the source code](https://github.com/Koenkk/zigbee-herdsman/blob/master/src/adapter/deconz/driver/driver.ts), even if `adapter_delay` set to 0 we are still setting `PROCESS_APS_QUEUES_DELAY` to 50ms and `READY_TO_SEND_TIMEOUT` to 10ms for a total of 60ms waiting to send any message.

Why do we need to wait that long, when nothing is happening in the network and presumably in the deCONZ stick? Can't we just cut all of these delays to 0 and look at the deCONZ stick's message queue (or whatever indicates it is under load) and only delay sending commands if it is busy / will drop messages if we don't? Or is this for devices in the Zigbee network which might not respond well to a high frequency of network frames?

I would be happy to run a custom version of Z2M with delays set to 0 to see if my network (which is composed of IKEA and Philips bulbs and switches) works well that way, but am not super familiar with node.js and how to set a specific package to dev mode. The way zigbee-herdsman uses TypeScript which then gets transpiled also makes it difficult to do a few changes in the dependent packages after you run `npm ci`.

I tried checking what the behaviour is for the z-stack adapters but the code is laid out differently which makes this a bit more challenging.

This is with the latest dev checkout from today, `adapter_delay` set to 0, an [IKEA E1743](https://www.zigbee2mqtt.io/devices/E1743.html) switch bound to an [IKEA LED driver](https://www.zigbee2mqtt.io/devices/ICPSHC24-30EU-IL-1.html) and the deCONZ adapter with (to my knowledge) the latest firmware.

Details of the debug logs here:

    zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010 +7s
    zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1 +7s
    zigbee-herdsman:deconz:driver query aps data indication +1ms
    zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 25 +79ms
    zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 25 srcAddr: 0xcac7 destAddr: 0xcd profile id: 0x104 cluster id: 0x6 +86ms
    zigbee-herdsman:deconz:frameParser response payload: 1,57,0 +0ms
    zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":57,"manufacturerCode":null,"commandIdentifier":0},"Payload":{}},"address":51911,"endpoint":1,"linkquality":231,"groupID":205}' +9s
  Zigbee2MQTT:debug 2020-12-16 11:35:15: Received Zigbee message from 'office_desk_remote', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 205
    zigbee-herdsman:controller:endpoint DefaultResponse 0xccccccfffee13f8f/1 6(0, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) +9s
    zigbee-herdsman:deconz:adapter zclFrame.payload: +9s
    zigbee-herdsman:deconz:adapter { cmdId: 0, statusCode: 0 } +0ms
    zigbee-herdsman:deconz:adapter no response expected +1ms
    zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +10ms
  Zigbee2MQTT:info  2020-12-16 11:35:15: MQTT publish: topic 'zigbee2mqtt/office_desk_remote', payload '{"action":"off","battery":5,"click":"off","linkquality":231,"update_available":false}'
    zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xcac7 EP:1 SeqNr. 26 request id: 4 +84ms
    zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 4 status: 0 +94ms
    zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +3ms
    zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110 +10ms
    zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +10ms
    zigbee-herdsman:deconz:driver query aps data confirm +0ms
    zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 27 +93ms
    zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0xcac7 request id: 4 confirm status: 0 +96ms
    zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +3ms
    zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint - message send with transSeq Nr.: 57 +194ms

Mentioning #72 as this issue might be of interest to the community, but that master ConBee issue is not very actionable given the number of different problems it covers.

ndfred, 2020-12-16:
I have actually managed to tweak these values by editing `node_modules/zigbee-herdsman/dist/adapter/deconz/driver/driver.js`. The delay isn't gone, and sometimes it is just as long as before, but sometimes I get a super quick response like this one which makes me feel the adapter can handle very short delays and will just sequence its calls however it sees fit:

    zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010 +5s
    zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1 +5s
    zigbee-herdsman:deconz:driver query aps data indication +0ms
    zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 215 +1ms
    zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 215 srcAddr: 0xcac7 destAddr: 0xcd profile id: 0x104 cluster id: 0x6 +7ms
    zigbee-herdsman:deconz:frameParser response payload: 1,71,0 +0ms
    zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":71,"manufacturerCode":null,"commandIdentifier":0},"Payload":{}},"address":51911,"endpoint":1,"linkquality":243,"groupID":205}' +9s
  Zigbee2MQTT:debug 2020-12-16 13:12:32: Received Zigbee message from 'office_desk_remote', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 205
    zigbee-herdsman:controller:endpoint DefaultResponse 0xccccccfffee13f8f/1 6(0, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) +9s
    zigbee-herdsman:deconz:adapter zclFrame.payload: +9s
    zigbee-herdsman:deconz:adapter { cmdId: 0, statusCode: 0 } +0ms
    zigbee-herdsman:deconz:adapter no response expected +1ms
    zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +11ms
  Zigbee2MQTT:info  2020-12-16 13:12:32: MQTT publish: topic 'zigbee2mqtt/office_desk_remote', payload '{"action":"off","battery":5,"click":"off","linkquality":243,"update_available":false}'
    zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xcac7 EP:1 SeqNr. 216 request id: 28 +87ms
    zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 28 status: 0 +96ms
    zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +3ms
    zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110 +8ms
    zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +8ms
    zigbee-herdsman:deconz:driver query aps data confirm +0ms
    zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 217 +92ms
    zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0xcac7 request id: 28 confirm status: 0 +94ms
    zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +2ms
    zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint - message send with transSeq Nr.: 71 +192ms

Koenkk, 2020-12-17:
@ChrisHae can you check this?

ndfred, 2020-12-21:
I've been running in this configuration for a week and it is much more responsive and consistent, with no missed inputs.

However I tried to pair two new [Philips 929002335001](https://www.zigbee2mqtt.io/devices/929002335001.html) bulbs and that failed in this configuration. I then updated to the latest dev checkout, which might have included updated device profiles, but also deleted my changes so hard to know whether the timing tweaks or the outdated checkout were causing pairing to fail.

I will make the same changes again and continue running in that configuration, and will let you know if anything new comes up.

ndfred, 2020-12-23:
Interesting benchmark for us from a [Silab document](https://www.silabs.com/documents/login/application-notes/an1138-zigbee-mesh-network-performance.pdf) describing their latency expectations and retransmit rates:

> For unicast forwarding within these networks, the message is forwarded as soon as the device is ready to send. For multicast forwarding, there are generally networking requirements for how messages are forwarded. For Zigbee devices, a multicast message is forwarded by a device only after a jitter of up to 64 milliseconds. However, the initiating device has a gap of 500 milliseconds before retransmitting the initial message

So up to 64ms per hop when sending a message, and retransmit after 500ms is expected in normal operating conditions.

> It is a known issue that the use of broadcasts in 15.4 networks should be minimized as they can cause temporary flooding of the network. Zigbee networks have a broadcast transaction record to prevent broadcasts from being processed that have already been forwar- ded by the device. This broadcast table must contain a minimum of 9 entries and they are aged out after 9 seconds, meaning on aver- age only 1 broadcast per second is supported in the network.

So one broadcast message per second is ideal, and even there we see evidence of network congestion (250ms latency). 2 seconds delay keeps the network in a good shape (70-80ms latency).

I'll look into mapping what happens when:

* you push a button on a remote
* you send a simple command to a bulb and a group
* a bulb reports its status

And what the ideal timings look like to ensure we don't overwhelm the network, yet have a responsive experience.

Any suggestions on prior research here, including for Conbee and TI sticks, is very welcome!

ndfred, 2020-12-23:
It turns out the delay set in the configuration file are never actually fully honoured:

* `HANDLE_DEVICE_STATUS_DELAY` will always be 100ms
* `PROCESS_APS_QUEUES_DELAY` will always be 100ms
* `READY_TO_SEND_TIMEOUT` will be honoured

That is because [the driver constructor is called before setDelay is called](https://github.com/Koenkk/zigbee-herdsman/blob/a607cae8e235fe5fa97befda5e22eece90c2b2fa/src/adapter/deconz/adapter/deconzAdapter.ts#L59-L60), when really the constructor should take the delay as a parameter and call `setDelay` on that. I'll try and get a PR out that fixes it.

In the meantime I would suggest tweaking the constructor:

* editing `node_modules/zigbee-herdsman/dist/adapter/deconz/driver/driver.js`
* setting `HANDLE_DEVICE_STATUS_DELAY` and `PROCESS_APS_QUEUES_DELAY` to the desired values in `constructor` (I just set them to 100000 and 1, then update `READY_TO_SEND_TIMEOUT` to 0 in `setDelay`)

kind3r, 2020-12-28:
Or the `setDelay` should restart all the `setInterval`s. 
If it were me, I would assign the intervals to variables and do their setup in a separate function that could be called from both the constructor and `setDelay`. This way the `setDelay` can be called at any point in time during runtime and it would still apply the setting.

closed 2021-02-05 by github-actions[bot]