Koenkk/zigbee-herdsman #294: ConBeeII - Zigbee2MQTT: Error while starting zigbee-herdsman

Nusserdt, 2021-01-17:
Hello @Koenkk,

first of all, thanks for all your work! I start with **CC2531** and [zigbee2mqtt container](https://www.zigbee2mqtt.io/information/docker.html) and see how much information you get from your devices. Then I bought a **ConBeeII** and run it in the [deCONZ container](https://github.com/marthoc/docker-deconz) and see how less information you get from your devices. So I decide to run **ConBeeII** with [zigbee2mqtt container](https://www.zigbee2mqtt.io/information/docker.html), but now I get a `Error while starting zigbee-herdsman` and have no clue what to do.

I run everything with docker-compose on a RaspberryPi B3+.

### zigbee2mqtt (docker-compose)
```
  # ----------------------------------------------------------- ZIGBEE2MQTT
  zigbee2mqtt:
    image: koenkk/zigbee2mqtt
    depends_on:
      - mosquitto
    container_name: zigbee2mqtt
    restart: always
    privileged: true
    deploy:
      resources:
        limits:
          memory: 100M
    networks:
      - home-network
    volumes:
      - /run/udev:/run/udev:ro
      - ./volumes/zigbee2mqtt:/app/data
    devices:
      - /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2406315-if00:/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2406315-if00
    environment:
      TZ: Europe/Berlin
    ports:
      - 9002:8080
```

### zigbee2mqtt (configuration.yaml)
```
# zigbee2mqtt.io https://www.zigbee2mqtt.io/information/configuration.html
homeassistant: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: "mqtt://raspberrypi"
  user: myuser
  password: mypassword
serial:
  adapter: deconz
  port: /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2406315-if00
advanced:
  network_key: GENERATE
frontend:
  port: 8080
```

### zigbee2mqtt (debug-log)
```
Using '/app/data' as data directory

> zigbee2mqtt@1.15.0 start /app
> node index.js

Zigbee2MQTT:info  2021-01-09 08:09:26: Logging to console and directory: '/app/data/log/2021-01-09.08-09-26' filename: log.txt
Zigbee2MQTT:debug 2021-01-09 08:09:26: Removing old log directory '/app/data/log/2021-01-09.08-05-26'
Zigbee2MQTT:debug 2021-01-09 08:09:28: Loaded state from file /app/data/state.json
Zigbee2MQTT:info  2021-01-09 08:09:29: Starting Zigbee2MQTT version 1.15.0 (commit #1bccc5d)
Zigbee2MQTT:info  2021-01-09 08:09:29: Starting zigbee-herdsman...
Zigbee2MQTT:debug 2021-01-09 08:09:29: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"deconz","path":"/dev/ttyACM0"}}'

CREATED DECONZ ADAPTER

Zigbee2MQTT:error 2021-01-09 08:09:40: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-01-09 08:09:40: Failed to start zigbee
Zigbee2MQTT:error 2021-01-09 08:09:40: Exiting...
Zigbee2MQTT:error 2021-01-09 08:09:40: undefined

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.15.0 start: `node index.js`
npm ERR! Exit status 1
```

For **deCONZ** install, I executed `sudo gpasswd -a $USER dialout` like suggested. Maybe an issue now? I removed the **deCONZ-container** from my stack.

ChrisHae, 2021-01-22:
I don't know if and how the conbee2 can be run in the zigbee2mqtt container but the configuration.yaml looks correct. What was the specific problem when running it in the deconz container?

And to figure out why it does not start in the z2mqtt container please enable debug log output and post the log from the startup. https://www.zigbee2mqtt.io/information/debug.html

Nusserdt, 2021-01-22:
@ChrisHae thanks a lot for your response, unfortunately this is already the `log_level: debug` log. I mess up this information in my question and fix this info in the **zigbee2mqtt (configuration.yaml)** part. I'am sorry for that.

The debug log print the following lines:
```
Zigbee2MQTT:debug 2021-01-09 08:09:26: Removing old log directory '/app/data/log/2021-01-09.08-05-26'
Zigbee2MQTT:debug 2021-01-09 08:09:28: Loaded state from file /app/data/state.json
Zigbee2MQTT:debug 2021-01-09 08:09:29: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"deconz","path":"/dev/ttyACM0"}}'
```

In general the [deCONZ container](https://github.com/marthoc/docker-deconz) work perfectly well together with the **ConBeeII** stick. I was only missing the information about the **battery** and **link quality** of my `xiaomi aqara temp/hum` sensor which I had with the **zigbee2mqtt** setup. I also need the data from the sensor in MQTT.

ChrisHae, 2021-01-23:
I mean the zigbee-herdsman debug log, when you start z2mqtt with: DEBUG=zigbee-herdsman* npm start
It gives more information but only in STDOUT and not in the logfiles. The link quality reported with the ConBee2 is not the correct one. This will be fixed in FW. But when you start a network scan with z2mqtt the correct LQI will be displayed. I don't know why the battery info is missing. Perhaps the debug log would help. Or I have to test it for myself, I also have a aqara temp/hum sensor here.

Nusserdt, 2021-01-24:
Alright @ChrisHae now I get it. I add `DEBUG: zigbee-herdsman*` to my **zigbee2mqtt** docker-compose environment and the following lines get added to my log:

```
Zigbee2MQTT:info  2021-01-24 06:49:27: Starting zigbee-herdsman...
Zigbee2MQTT:debug 2021-01-24 06:49:27: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"deconz","path":"/dev/ttyACM0"}}'
2021-01-24T05:49:27.907Z zigbee-herdsman:adapter Path '/dev/ttyACM0' is valid for 'DeconzAdapter'
CREATED DECONZ ADAPTER
2021-01-24T05:49:27.922Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[9,137,223,208,156,193,192,78,141,205,121,233,183,56,233,160],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyACM0","adapter":"deconz"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json","adapter":{"concurrent":null}}'
2021-01-24T05:49:27.928Z zigbee-herdsman:deconz:driver Opening with /dev/ttyACM0
2021-01-24T05:49:27.959Z zigbee-herdsman:deconz:driver Serialport opened
2021-01-24T05:49:27.964Z zigbee-herdsman:controller:log Started with result 'resumed'
2021-01-24T05:49:27.964Z zigbee-herdsman:controller:log Injected database: true, adapter: true
2021-01-24T05:49:27.970Z zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 1 paramId: 1
2021-01-24T05:49:33.564Z zigbee-herdsman:deconz:frameParser device state: 10100010
2021-01-24T05:49:33.566Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2021-01-24T05:49:37.973Z zigbee-herdsman:deconz:driver Timeout for request - CMD: 0xa seqNr: 1
Zigbee2MQTT:error 2021-01-24 06:49:37: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-01-24 06:49:37: Failed to start zigbee
Zigbee2MQTT:error 2021-01-24 06:49:37: Exiting...
Zigbee2MQTT:error 2021-01-24 06:49:37: undefined
```

Nusserdt, 2021-02-24:
Issue still exist, please don't close.

NinioDev, 2021-03-03:
I have exactly the same issue, I'm using Conbee II with Zigbee2MQTT, I have activated the debug (log_level: debug), and here is the logs (with errors) I got:
```
Zigbee2MQTT:debug 2021-03-03 17:04:47: Loaded state from file /config/zigbee2mqtt/state.json
Zigbee2MQTT:info  2021-03-03 17:04:47: Logging to console and directory: '/config/zigbee2mqtt/log/2021-03-03.17-04-45' filename: log.txt
Zigbee2MQTT:debug 2021-03-03 17:04:47: Removing old log directory '/config/zigbee2mqtt/log/2021-03-03.17-02-20'
Zigbee2MQTT:info  2021-03-03 17:04:47: Starting Zigbee2MQTT version 1.18.1 (commit #1.18.1)
Zigbee2MQTT:info  2021-03-03 17:04:47: Starting zigbee-herdsman (0.13.71)
Zigbee2MQTT:debug 2021-03-03 17:04:47: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"deconz","path":"/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2413881-if00"}}'
CREATED DECONZ ADAPTER
Zigbee2MQTT:error 2021-03-03 17:04:57: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-03-03 17:04:57: Failed to start zigbee
Zigbee2MQTT:error 2021-03-03 17:04:57: Exiting...
Zigbee2MQTT:error 2021-03-03 17:04:57: undefined
```
Any idea what is the case?

steps I did:
1) Flash my Conbee II following the instructions [here ](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually) to the latest version:
2) Here is my configuration (I have created mqtt user for this, I also used the homeassistant user of mosquitto broker, same results):
```
data_path: /config/zigbee2mqtt
external_converters: []
devices: devices.yaml
groups: groups.yaml
homeassistant: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://192.168.1.191:1883'
  user: mqtt
  password: mqtt
serial:
  adapter: deconz
  port: >-
    /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2413881-if00
advanced:
  log_level: debug
  pan_id: 6754
  channel: 11
  network_key:
    - 1
    - 3
    - 5
    - 7
    - 9
    - 11
    - 13
    - 15
    - 0
    - 2
    - 4
    - 6
    - 8
    - 10
    - 12
    - 13
  availability_blocklist: []
  availability_passlist: []
device_options: {}
blocklist: []
passlist: []
queue: {}
frontend:
  port: 8099
experimental: {}
socat:
  enabled: false
  master: 'pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777'
  slave: 'tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5'
  options: '-d -d'
  log: false
baudrate: 38400
rtscts: false
```
What am I missing, I cant make my zigbee2mqtt work :(
Thank you

Nusserdt, 2021-03-21:
I could solve my problem by completely reinstall my Raspberry Pi 3B+.

1. flush the Pi with the now Imager Software from https://www.raspberrypi.org/software/
2. login to your Pi and make the preconfiguration (Timezone, WLAN, SSH connection)
3. `sudo apt-get update && sudo apt-get upgrade`
4.  `sudo rpi-update`
5.  `curl -fsSL https://get.docker.com -o get-docker.sh`
6. `sudo sh get-docker.sh`
7. `sudo usermod -aG docker pi`
8. `sudo pip3 -v install docker-compose`
9. `git clone MY_REPOSITORY`
10. `sudo reboot # IMPORTANT FOR DOCKER`
11. `docker-compose up -d`

**Code changes:**
- I switch back from the hacky device port `/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2406315-if00` to `/dev/ttyACM0` (I update my question)
- I had to add `listener 1883` to my `mosquitto.conf` to fix an runtime error

**Home-Assistent (ConBeeII + Zigbee2Mqtt)**
[image]

closed 2021-04-29 by github-actions[bot]