Koenkk/zigbee-herdsman #305: Make it possible to save in an external database

wilmardo, 2021-02-03:
Same issue as https://github.com/Koenkk/zigbee2mqtt/issues/4947

As stated in the linked issue this would be a welcome addition for users who write on SD card like storage.

My usecase is a bit specific but nonetheless a bit of explanation.
I am trying to get a stateless Zigbee2MQTT implementation so it can failover between Kubernetes nodes (external CC2530 TCP serial) without the need of a volume (since distributed storage is hard on low powered devices). 
I am using Kubernetes configmaps for the configuration so only the `database.db` and the `coordinator_backup.json` are blocking a completely stateless approach.
If these could be stored in an external database then can even run the image without any volume.

wilmardo, 2021-03-08:
Not stale.

With something like [Sequelize](https://sequelize.org/) it would be possible to maintain the current SQLite implementation as backwards compatible option.

wilmardo, 2021-04-22:
Still not stale though haha

Toastyyy3, 2024-08-23:
I suppose there is still no support for external databases?

Koenkk, 2024-09-25:
@judahrand changes are welcome, but note that it's not just the `database.db` and `coordinator_backup.json`, all files in the `data` folder should be persisted (also `state.json`)

judahrand, 2024-09-26:
Fantastic! I took an initial look at this last night and it (obviously) won't be a trivial change but I do think there could be _significant_ value to using a 'real' database in terms of stability, extensibility ([litestream](https://github.com/benbjohnson/litestream), Postgres, etc.), and, I suspect, even performance. 

I'm not a Typescript developer normally so I'm not familiar with the ORM landscape in Typescript but from my investigation last night I think that using [Drizzle](https://github.com/drizzle-team/drizzle-orm) probably makes the most sense. It seems to aim to be as lightweight as possible seems to fit well for `zigbee-herdsman`. Initially, I suspect it would also be most straightforward to only support Sqlite. Other databases could be supported later if the desire was present. 

There are a few design considerations to make too, I think. One that jumps to mind is how the 'backups' would work. For Coordinator backups this could probably just be versioned rows in a table and the newest row is always used by default or a given row could be marked as 'active'. A limited number of these backups could be kept at any given time and they could be timestamped. 

When it comes to the backup logic when a `reset` is performed it becomes a little more difficult. For Sqlite initially the same logic can be used where we just copy the entire database file to a backed up version. However, this isn't really possible when using a remote database (and in this case one could argue that this sort of disaster recovery should be the database's responsibility not the application's). I wonder if in that case the functionality could be dropped? 

I'm sure there are other decisions that would need to be made but from what I can tell, as long as the API for Controller is maintained and a migration path for the current files exists then other breaking changes should be acceptable. Is this correct?

Koenkk, 2024-09-28:
I'm not convinced that a SQL database will have any impact on performance, the whole database is loaded in memory and all devices are resolved through a very efficient [lookup](https://github.com/Koenkk/zigbee-herdsman/blob/5d02efe44826401521b85259dde5e3ca4421e312/src/controller/model/device.ts#L646). A typical database is also nothing in terms of size, mine is 102KB (+- 80 devices).

An external database will also not solve the problem of needing persistent storage, z2m still stores `configuration.yaml` and `state.json` which needs to be writable.

I think a better approach here is to allow the Z2M data directory to be stored on a remote filesystem. If we use for example [OpenDAL](https://opendal.apache.org/) you could store the directory on almost any remote file system, e.g. via sftp or even OneDrive! 😄

judahrand, 2024-09-28:
That's a fair point on performance. I hadn't initially realized that the whole database is loaded into memory. There are other advantages to using SQLite at least, however, imo. 

> An external database will also not solve the problem of needing persistent storage, z2m still stores configuration.yaml and state.json which needs to be writable.

Does the `configuration.yaml` need to be writable and persistent? That seems a surprising property for a config file which I'd expect to be entirely user controlled. That `state.json` could easily be migrated to live in the same SQLite database I think.

I've got a branch which implements a SQLite database which almost passes all the tests so I'll get that working and put it up at least for consideration.

Koenkk, 2024-09-28:
> Does the configuration.yaml need to be writable and persistent? 

Yes, e.g. the `devices` is updated whenever you pair a new device and for example when updating settings through the frontend.

Nerivec, 2024-09-29:
I think orm is much too complex for the 100 or so lines that Z2M needs. There is practically no use for the features since, like Koenkk said, everything is cached in memory on startup and read/parsed from there at runtime, not from the database. _The tiny size allows this without creating RAM issues, and makes it a hard-to-beat option as far as speed is concerned._

As for writes, a quick bench on current vs drizzle orm branch shows a 2x drop in performance minimum. No doubt the overhead alone is a killer :/

_Coordinator backups in the database is definitely a no-go for me. This is not portable._

judahrand, 2024-09-29:
> As for writes, a quick bench on current vs drizzle orm branch shows a 2x drop in performance minimum. No doubt the overhead alone is a killer :/

What benchmark are you running? I suspect it can be improved substantially - the current implementation is the simplest lift and shift possible.

> Coordinator backups in the database is definitely a no-go for me. This is not portable.

I'm not sure I'm understanding you here. Why is it not portable? Easy to get back to the same JSON by pulling it out of a db table.

Nerivec, 2024-09-29:
> What benchmark are you running? I suspect it can be improved substantially - the current implementation is the simplest lift and shift possible.

The usual, tinybench, with a `save` action (nothing else but a `device.save()`, I bypassed the rest of the logic).
I don't think it can be improved by much, since every runtime action is in-memory, not in-database, only saves actually touch the database at runtime, no querying, no filtering, etc.. The only thing the orm would actually end up being used for, is load everything in-memory on startup, then periodically write to fs (_trying to do anything else versus the current in-RAM implementation would result in a further massive drop in performance_). Hence the orm overhead being a killer against a plain fs read/write.

> Easy to get back to the same JSON by pulling it out of a db table.

Not without dependencies and with extra parsing required. Current file is natively supported and follows the open backup format.
https://github.com/zigpy/open-coordinator-backup

closed 2024-12-19 by github-actions[bot]