Koenkk/zigbee-herdsman #313: Seems like a bug in the code

peterbb, 2021-02-20:
In `src/adapter/zigate/driver/buffaloZiGate.ts` there is the following code:
```javascript
const mac = this.readUInt8();
result.alternatePanCoordinator = !!(mac.mac & 0b00000001);
```

That doesn't seem to make any sense: readUInt8 returns a number, which doesn't have any property `mac`. Am I missing something? I don't have hardware to test it.

Koenkk, 2021-02-21:
@G1K can you check this?

G1K, 2021-02-21:
@peterbb Yes, you are right, these are flaws in the code.
1) MACCAPABILITY is not used anywhere in the application, parsing has been inherited.
2) In `write`, you can remove it, it was added due to the difference in the IEEADDR byte order that was transmitted by zigate and the one that comes from the zigbee network, but gradually I had to get away from zigate messages and use RAW mode, so this is also an unused piece of code.
3) In the third case, return is unnecessary, but does not break the application.
Yes, I know the code is far from ideal, and it is even used (ts-ignore, istanbul ignore) by analogy with neighboring adapters for example deconz, but it works)

All pull requests that improves the adapter code are welcome.

closed 2021-04-01 by github-actions[bot]