Koenkk/zigbee-herdsman #319: [WIP]: EFR32 EZSP adapter implementation and test

kirovilya, 2021-03-03:
After [accepting the alpha version of the code](https://github.com/Koenkk/zigbee-herdsman/pull/317), now I propose to write comments here until we complete the development.

The adapter is designed to work with EFR32 chips using the EmberZNet v8 protocol. Recommended firmware NCP 6.7.8.

The work with devices with such chips was tested:
* Re-flashed Sonoff Zigbee Bridge gateway https://www.digiblur.com/2020/07/how-to-use-sonoff-zigbee-bridge-with.html
* Flashed devboard EByte E180-ZG120B https://github.com/zigpy/zigpy/discussions/584
* Flashed usb stick based on SM-011 https://modkam.ru/?p=1889 (description in Russian, use the translator)
* Defaro Sprut stick https://spruthub.ru/device/defaro-sprut-stick-2-pro (description in Russian, use the translator)

Settings:
* specify **port** parameter or gateway address
* set **adapter** to "ezsp"

Worked:

- [X] Pairing/Leaving
- [X] Bind/Unbind
- [X] ZCL messages/commands
- [X] Network map

TODO:

- [ ] Improving the stability work of the network
- [ ] Groups
- [ ] Touchlink 
- [ ] Backup/Restore
- [ ] OTA update
- [ ] Tests
- [ ] ZGP (Zigbee Green Power)

I urge the developers to participate in the finalization of the adapter. I'm not an expert in TypeScript and EmberZNet, so I ask for help.

MattWestb, 2021-03-03:
Have you getting some user with EM357 or EM3581 that have updating the firmware to EZSP6.7.8.0 to testing if they is working ?

It shall working if Ssilabs have making all OK then the protocol is is independent of the hardware its running.

I posing one request for testing in EM35X forum.

If going well you have getting some more thousand possible user of Z2M :-)))

Great work done going from alpha stage and and implanting more functionality !!!

kirovilya, 2021-03-03:
@MattWestb no, i haven't. I think it should work in the same way there, too. the protocol is the same.

MattWestb, 2021-03-03:
I have asking our brave EM35X users if some can testing it and im 95% sure its working OK :-))

Hedda, 2021-03-03:
> The adapter is designed to work with EFR32 chips using the EmberZNet v8 protocol. Recommended firmware NCP 6.7.8.
> 
> The work with devices with such chips was tested:
> 
> * Re-flashed Sonoff Zigbee Bridge gateway https://www.digiblur.com/2020/07/how-to-use-sonoff-zigbee-bridge-with.html
> * Flashed devboard EByte E180-ZG120B [zigpy/zigpy#584](https://github.com/zigpy/zigpy/discussions/584)
> * Flashed usb stick based on SM-011 https://modkam.ru/?p=1889 (description in Russian, use the translator)
> * Defaro Sprut stick https://spruthub.ru/device/defaro-sprut-stick-2-pro (description in Russian, use the translator)

EZSP adapter can also be tested with these as long as have updated firmware  to Silicon Labs EmberZNet 6.7.x coordinator or later:

- Zigbee 3.0 dongle from ITead (coming soon?) https://community.home-assistant.io/t/zigbee-3-0-usb-dongle-from-itead-based-on-silicon-labs-efr32mg21-soc/271144
- Elelabs Zigbee USB Adapter (ELU013) https://elelabs.com/products/elelabs-usb-adapter.html
- Elelabs Zigbee Raspberry Pi Shield (ELR023) https://elelabs.com/products/elelabs-zigbee-shield.html
-  TubesZB style ESP8266/ESP32 based DIY Zigbee gateways running ESPHome with serial-bridge configuration https://github.com/zigpy/zigpy/discussions/584
- Tuya TYGWZ01 / Lidl / Silvercrest Smart Gateway hacked with DIY firmware https://github.com/zigpy/zigpy/discussions/650
- DIY "Billy EZSP" using IKEA TRÅDFRI ICC-A-1 Module ripped from an IKEA Tradfri Series device with hacked firmware https://github.com/MattWestb/IKEA-TRADFRI-ICC-A-1-Module
- Nortek GoControl QuickStick Combo Model HUSBZB-1 (note that this is based on Silabs older EM3581 "Ember" family chip)
- Telegesis ETRX357USB (note that this is based on Silabs older EM357 / EM357x "Ember" family chip)

USB-stick firmware can be updated with Elelabs EZSP Firmware Update Utility or the Zigbee firmware updater image by @walthowd

https://github.com/Elelabs/elelabs-zigbee-ezsp-utility

https://github.com/walthowd/husbzb-firmware

https://github.com/zha-ng/EZSP-Firmware

https://github.com/grobasoz/zigbee-firmware/

Direct link to the wiki page where @MattWestb documented recommended application firmware version for EZSP Coordinators:

https://github.com/zigpy/zigpy/wiki/Coordinator-Firmware-Updates#recommended-firmware-version-for-ezsp-coordinators

danielk117, 2021-03-06:
Tried it with 5 thermostat SEA801. Works basically 👍  but there are some problems. Z2M restarts every few minutes with the following message...
```
│Mar 06 15:03:25 raspberrypi npm[14963]: /opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:67                                      │
│Mar 06 15:03:25 raspberrypi npm[14963]:                     waiter.reject(new Error(message));                                                       │
│Mar 06 15:03:25 raspberrypi npm[14963]:                                   ^                                                                          │
│Mar 06 15:03:25 raspberrypi npm[14963]: Error: {"frameId":85,"sequence":261} after 30000ms                                                           │
│Mar 06 15:03:25 raspberrypi npm[14963]:     at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:67:35)       │
│Mar 06 15:03:25 raspberrypi npm[14963]:     at listOnTimeout (node:internal/timers:556:17)                                                           │
│Mar 06 15:03:25 raspberrypi npm[14963]:     at processTimers (node:internal/timers:499:7)                                                            │
│Mar 06 15:03:25 raspberrypi npm[14963]: npm ERR! code 1                                                                                              │
│Mar 06 15:03:25 raspberrypi npm[14963]: npm ERR! path /opt/zigbee2mqtt                                                                               │
│Mar 06 15:03:25 raspberrypi npm[14963]: npm ERR! command failed                                                                                      │
│Mar 06 15:03:25 raspberrypi npm[14963]: npm ERR! command sh -c node index.js                                                                         │
│Mar 06 15:03:25 raspberrypi npm[14963]: npm ERR! A complete log of this run can be found in:                                                         │
│Mar 06 15:03:25 raspberrypi npm[14963]: npm ERR!     /home/pi/.npm/_logs/2021-03-06T14_03_25_317Z-debug.log                                          │
│Mar 06 15:03:25 raspberrypi systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE                                      │
│Mar 06 15:03:25 raspberrypi systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
```

kirovilya, 2021-03-06:
@danielk117 Thanks for testing. After accepting the latest changes, you can check again - the reconnection in case of errors has been implemented and some pending moments have been removed.

Koenkk, 2021-03-06:
Changes will be available in few hours in the dev branch (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

danielk117, 2021-03-06:
Thank you both! 😃  I'm already on dev branch and will try the latest version now.

danielk117, 2021-03-06:
@kirovilya 
Sorry man, same errors for now.

````
│Mar 06 21:02:06 raspberrypi npm[12167]: Assertion failed                                                                                             │
│Mar 06 21:02:06 raspberrypi npm[12167]: Assertion failed                                                                                             │
│Mar 06 21:02:42 raspberrypi npm[12167]: /opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:67                                      │
│Mar 06 21:02:42 raspberrypi npm[12167]:                     waiter.reject(new Error(message));                                                       │
│Mar 06 21:02:42 raspberrypi npm[12167]:                                   ^                                                                          │
│Mar 06 21:02:42 raspberrypi npm[12167]: Error: {"frameId":97,"sequence":267} after 10000ms                                                           │
│Mar 06 21:02:42 raspberrypi npm[12167]:     at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:67:35)       │
│Mar 06 21:02:42 raspberrypi npm[12167]:     at listOnTimeout (node:internal/timers:556:17)                                                           │
│Mar 06 21:02:42 raspberrypi npm[12167]:     at processTimers (node:internal/timers:499:7)                                                            │
│Mar 06 21:02:42 raspberrypi npm[12167]: npm ERR! code 1                                                                                              │
│Mar 06 21:02:42 raspberrypi npm[12167]: npm ERR! path /opt/zigbee2mqtt                                                                               │
│Mar 06 21:02:42 raspberrypi npm[12167]: npm ERR! command failed                                                                                      │
│Mar 06 21:02:42 raspberrypi npm[12167]: npm ERR! command sh -c node index.js                                                                         │
│Mar 06 21:02:42 raspberrypi npm[12167]: npm ERR! A complete log of this run can be found in:                                                         │
│Mar 06 21:02:42 raspberrypi npm[12167]: npm ERR!     /home/pi/.npm/_logs/2021-03-06T20_02_42_328Z-debug.log
````
npm debug log:
````

0 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
1 info using npm@7.3.0
2 info using node@v15.5.0
3 timing config:load:defaults Completed in 4ms
4 timing config:load:file:/usr/lib/node_modules/npm/npmrc Completed in 5ms
5 timing config:load:builtin Completed in 5ms
6 timing config:load:cli Completed in 4ms
7 timing config:load:env Completed in 0ms
8 timing config:load:file:/opt/zigbee2mqtt/.npmrc Completed in 2ms
9 timing config:load:project Completed in 4ms
10 timing config:load:file:/home/pi/.npmrc Completed in 0ms
11 timing config:load:user Completed in 1ms
12 timing config:load:file:/usr/etc/npmrc Completed in 0ms
13 timing config:load:global Completed in 0ms
14 timing config:load:cafile Completed in 1ms
15 timing config:load:validate Completed in 1ms
16 timing config:load:setUserAgent Completed in 2ms
17 timing config:load:setEnvs Completed in 2ms
18 timing config:load Completed in 25ms
19 verbose npm-session 3488c9d57fec89ec
20 timing npm:load Completed in 44ms
21 timing command:run-script Completed in 1232065ms
22 timing command:start Completed in 1232071ms
23 verbose stack Error: command failed
23 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/@npmcli/promise-spawn/index.js:64:27)
23 verbose stack     at ChildProcess.emit (node:events:376:20)
23 verbose stack     at maybeClose (node:internal/child_process:1063:16)
23 verbose stack     at Process.ChildProcess._handle.onexit (node:internal/child_process:295:5)
24 verbose pkgid zigbee2mqtt@1.18.1-dev
25 verbose cwd /opt/zigbee2mqtt
26 verbose Linux 5.4.83-v7l+
27 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
28 verbose node v15.5.0
29 verbose npm  v7.3.0
30 error code 1
31 error path /opt/zigbee2mqtt
32 error command failed
33 error command sh -c node index.js
34 verbose exit 1
````

danielk117, 2021-03-07:
As written before, the problem still exists, but i think it happens less often.
I Found another error, which wasn't there with the CC2530. It happend a couple of times since yesterday. 

Around 07:45 i was trying to set `zigbee/thermostat_wohnzimmer/set/current_heating_setpoint` to `17`. The device itself does the job, but Z2M says after 2 minutes (at 07:47:46) the `current_heating_setpoint` is `22` (which wasn't true).
I think the message `Assertion failed` has to do with that.
```
Mar 07 07:41:43 raspberrypi npm[3136]: Zigbee2MQTT:info  2021-03-07 07:41:43: MQTT publish: topic 'zigbee/thermostat_flur', payload '{"anti_scaling":"ON","away_mode":"OFF","battery_low":false,"child_lock":"LOCKED","current_heating_setpoint":15,"frost_detection":"OFF","linkquality":176,"local_temperature":14.7,"local_temperature_calibration":0,"preset_mode":"none","system_mode":"heat","window_detection":"OFF"}'
Mar 07 07:45:33 raspberrypi npm[3136]: Zigbee2MQTT:info  2021-03-07 07:45:33: MQTT publish: topic 'zigbee/thermostat_kinderzimmer', payload '{"anti_scaling":"OFF","away_mode":"OFF","battery_low":false,"child_lock":"LOCKED","current_heating_setpoint":21,"frost_detection":"OFF","linkquality":120,"local_temperature":22.1,"local_temperature_calibration":0,"preset_mode":"none","system_mode":"heat","window_detection":"OFF"}'
Mar 07 07:45:43 raspberrypi npm[3136]: Zigbee2MQTT:info  2021-03-07 07:45:43: MQTT publish: topic 'zigbee/thermostat_kinderzimmer', payload '{"anti_scaling":"OFF","away_mode":"OFF","battery_low":false,"child_lock":"LOCKED","current_heating_setpoint":21,"frost_detection":"OFF","linkquality":116,"local_temperature":22.1,"local_temperature_calibration":0,"preset_mode":"none","system_mode":"heat","window_detection":"OFF"}'
Mar 07 07:45:54 raspberrypi npm[3136]: Zigbee2MQTT:info  2021-03-07 07:45:54: MQTT publish: topic 'zigbee/thermostat_kinderzimmer', payload '{"anti_scaling":"OFF","away_mode":"OFF","battery_low":false,"child_lock":"LOCKED","current_heating_setpoint":21,"frost_detection":"OFF","linkquality":116,"local_temperature":22.1,"local_temperature_calibration":0,"preset_mode":"none","system_mode":"heat","window_detection":"OFF"}'
Mar 07 07:45:59 raspberrypi npm[3136]: Zigbee2MQTT:info  2021-03-07 07:45:59: MQTT publish: topic 'zigbee/thermostat_kinderzimmer', payload '{"anti_scaling":"OFF","away_mode":"OFF","battery_low":false,"child_lock":"LOCKED","current_heating_setpoint":21,"frost_detection":"OFF","linkquality":116,"local_temperature":22.1,"local_temperature_calibration":0,"preset_mode":"none","system_mode":"heat","window_detection":"OFF"}'
Mar 07 07:46:05 raspberrypi npm[3136]: Assertion failed
Mar 07 07:46:05 raspberrypi npm[3136]: Assertion failed
Mar 07 07:46:21 raspberrypi npm[3136]: Zigbee2MQTT:info  2021-03-07 07:46:21: MQTT publish: topic 'zigbee/thermostat_kinderzimmer', payload '{"anti_scaling":"OFF","away_mode":"OFF","battery_low":false,"child_lock":"LOCKED","current_heating_setpoint":21,"frost_detection":"OFF","linkquality":116,"local_temperature":22.1,"local_temperature_calibration":0,"preset_mode":"none","system_mode":"heat","window_detection":"OFF"}'
Mar 07 07:47:46 raspberrypi npm[3136]: Zigbee2MQTT:info  2021-03-07 07:47:46: MQTT publish: topic 'zigbee/thermostat_wohnzimmer', payload '{"anti_scaling":"ON","away_mode":"OFF","battery_low":false,"child_lock":"LOCKED","current_heating_setpoint":22,"frost_detection":"OFF","linkquality":188,"local_temperature":22.2,"local_temperature_calibration":0,"preset_mode":"none","system_mode":"heat","window_detection":"OFF"}'
```
Any idea?

kirovilya, 2021-03-07:
@danielk117 these logs are very few ... deeper logging is needed - https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

kirovilya, 2021-03-07:
I think that "Assertion failed" is a restart of the connection, when a lot of errors have accumulated, the adapter restarts the connection and re-runs the initialization procedure. "Assertion failed" is not a critical error in this case. Most likely inside there are more critical errors that are simply not visible here, but they are visible in the debug logs

danielk117, 2021-03-07:
While checking the logs, the following error occured multiple times while everything is working normal.
```
Mar 07 12:26:22 raspberrypi npm[26687]: 2021-03-07T11:26:22.700Z zigbee-herdsman:adapter:ezsp:error Watchdog heartbeat timeout Error: {"frameId":5,"sequence":263} after 10000ms
Mar 07 12:26:22 raspberrypi npm[26687]:     at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:67:35)       
Mar 07 12:26:22 raspberrypi npm[26687]:     at listOnTimeout (node:internal/timers:556:17)                  
Mar 07 12:26:22 raspberrypi npm[26687]:     at processTimers (node:internal/timers:499:7)
```
I was trying to set `current_heating_setpoint` to `28` at 12:40:58. 
The devices doesn't react and the debugging logs show the following...
```
Mar 07 12:40:58 raspberrypi npm[26687]: 2021-03-07T11:40:58.188Z zigbee-herdsman:adapter:ezsp:uart <-- [60bdb1a96b2a157977904b25455493099d4e274ddece6763e87e] [96,189,177,169,107,42,21,121,119,144,75,37,69,84,147,9,157,78,39,77,222,206,103,99,232,126]
Mar 07 12:40:58 raspberrypi npm[26687]: 2021-03-07T11:40:58.188Z zigbee-herdsman:adapter:ezsp:uart Recv DATA frame (6,0,0): 60bdb1a96b2a157977904b25455493099d4e274ddece6763e87e
Mar 07 12:40:58 raspberrypi npm[26687]: 2021-03-07T11:40:58.188Z zigbee-herdsman:adapter:ezsp:uart Send ACK  frame (7)
Mar 07 12:40:58 raspberrypi npm[26687]: 2021-03-07T11:40:58.188Z zigbee-herdsman:adapter:ezsp:uart --> [87009f7e] [135,0,159,126]
Mar 07 12:40:58 raspberrypi npm[26687]: 2021-03-07T11:40:58.188Z zigbee-herdsman:adapter:ezsp:log <=== Frame: ff90013f0000cb2e040100ef010140010000e6330000
Mar 07 12:40:58 raspberrypi npm[26687]: 2021-03-07T11:40:58.188Z zigbee-herdsman:adapter:ezsp:log <=== Application frame 63 (messageSentHandler) received: 00cb2e040100ef010140010000e6330000
Mar 07 12:40:58 raspberrypi npm[26687]: 2021-03-07T11:40:58.188Z zigbee-herdsman:adapter:ezsp:log <=== Application frame 63 (messageSentHandler)   parsed: 0,11979,EmberApsFrame: {"profileId":260,"clusterId":61184,"sourceEndpoint":1,"destinationEndpoint":1,"options":320,"groupId":0,"sequence":230},51,0,
Mar 07 12:40:58 raspberrypi npm[26687]: 2021-03-07T11:40:58.291Z zigbee-herdsman:adapter:ezsp:uart <-- [70bdb1a97d312a15b65894a524ab5593499c89eb6426e09874f0cf308bfc5258a5ebc9de6f8ed379237e] [112,189,177,169,125,49,42,21,182,88,148,165,36,171,85,147,73,156,137,235,100,38,224,152,116,240,207,48,139,252,82,88,165,235,201,222,111,142,211,121,35,126]
Mar 07 12:40:58 raspberrypi npm[26687]: 2021-03-07T11:40:58.291Z zigbee-herdsman:adapter:ezsp:uart Recv DATA frame (7,0,0): 70bdb1a9112a15b65894a524ab5593499c89eb6426e09874f0cf308bfc5258a5ebc9de6f8ed379237e00
Mar 07 12:40:58 raspberrypi npm[26687]: 2021-03-07T11:40:58.291Z zigbee-herdsman:adapter:ezsp:uart Send ACK  frame (0)
Mar 07 12:40:58 raspberrypi npm[26687]: 2021-03-07T11:40:58.291Z zigbee-herdsman:adapter:ezsp:uart --> [8070787e] [128,112,120,126]
Mar 07 12:40:58 raspberrypi npm[26687]: 2021-03-07T11:40:58.291Z zigbee-herdsman:adapter:ezsp:log <=== Frame: ff9001450000040100ef010100010000c7cccfcb2effff0d095302002c670200040000012cbe
Mar 07 12:40:58 raspberrypi npm[26687]: 2021-03-07T11:40:58.291Z zigbee-herdsman:adapter:ezsp:log <=== Application frame 69 (incomingMessageHandler) received: 00040100ef010100010000c7cccfcb2effff0d095302002c670200040000012cbe
Mar 07 12:40:58 raspberrypi npm[26687]: [288B blob data]
Mar 07 12:40:58 raspberrypi npm[26687]: [3B blob data]
Mar 07 12:40:58 raspberrypi npm[26687]: [1B blob data]
Mar 07 12:40:58 raspberrypi npm[26687]: [2B blob data]
Mar 07 12:40:58 raspberrypi npm[26687]: 2021-03-07T11:40:58.292Z zigbee-herdsman:adapter:ezsp processMessage: {"messageType":0,"apsFrame":{"profileId":260,"clusterId":61184,"sourceEndpoint":1,"destinationEndpoint":1,"options":256,"groupId":0,"sequence":199},"lqi":204,"rssi":-49,"sender":11979,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[9,83,2,0,44,103,2,0,4,0,0,1,44]}}
Mar 07 12:40:58 raspberrypi npm[26687]: 2021-03-07T11:40:58.293Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":83,"manufacturerCode":null,"commandIdentifier":2},"Payload":{"status":0,"transid":44,"dp":103,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,1,44]}},"Command":{"ID":2,"parameters":[{"name":"status","type":32},{"name":"transid","type":32},{"name":"dp","type":32},{"name":"datatype","type":32},{"name":"fn","type":32},{"name":"data","type":65}],"name":"setDataResponse"}},"address":11979,"endpoint":1,"linkquality":204,"groupID":0}'
Mar 07 12:40:58 raspberrypi npm[26687]: Zigbee2MQTT:debug 2021-03-07 12:40:58: Received Zigbee message from 'thermostat_wohnzimmer', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,1,44],"type":"Buffer"},"datatype":2,"dp":103,"fn":0,"status":0,"transid":44}' from endpoint 1 with groupID 0
Mar 07 12:40:58 raspberrypi npm[26687]: 2021-03-07T11:40:58.295Z zigbee-herdsman:controller:endpoint DefaultResponse 0x847127fffe23af74/1 61184(2, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
[... cut ...]
Mar 07 12:43:08 raspberrypi npm[26687]: 2021-03-07T11:43:08.634Z zigbee-herdsman:adapter:ezsp:log ===> Send data    setExtendedTimeout: (4d00017e0074af23feff27718401)
Mar 07 12:43:08 raspberrypi npm[26687]: 2021-03-07T11:43:08.635Z zigbee-herdsman:adapter:ezsp:uart Send DATA frame (5,4,0): 4d00017e0074af23feff27718401
Mar 07 12:43:08 raspberrypi npm[26687]: 2021-03-07T11:43:08.635Z zigbee-herdsman:adapter:ezsp:uart waiting (6)
Mar 07 12:43:08 raspberrypi npm[26687]: 2021-03-07T11:43:08.635Z zigbee-herdsman:adapter:ezsp:uart --> [540f21a92a2a611d7a6ab502dbd193ad887e] [

kirovilya, 2021-03-07:
@danielk117 next time use pastebin or file... do not post long logs as comment

MattWestb, 2021-03-07:
If its large but not too large you can using this markdown:
```
<details>
  <summary>Click for folding out the log</summary>

Your long log.

</details>
```
<details>
  <summary>Click for folding out the log</summary>

```
Test log:
0
1
2
3
4
5
6
7
8
9
A
B
C
D
E
F
```
</details>

Or do one Zip file with the log file and attaching it in the comment.

danielk117, 2021-03-07:
Next time i use pastebin. I didn’t pay attention to the length 😉

kirovilya, 2021-03-09:
No, sorry. I am busy at main work

fouge, 2021-03-09:
Hello everyone,

I am trying the EZSP adapter (thank you for your work!) and I have an error while pairing. The procedure is starting correctly but then the device leaves the network:

```
Zigbee2MQTT:info  2021-03-09 17:59:25: Device 'arke' joined
Zigbee2MQTT:info  2021-03-09 17:59:25: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"arke","ieee_address":"0xf4ce361fc2bf0e3e"},"type":"device_joined"}'
Zigbee2MQTT:info  2021-03-09 17:59:25: Starting interview of 'arke'
Zigbee2MQTT:info  2021-03-09 17:59:25: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"arke","ieee_address":"0xf4ce361fc2bf0e3e","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2021-03-09 17:59:25: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"arke"},"type":"device_connected"}'
Zigbee2MQTT:info  2021-03-09 17:59:25: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"arke"},"type":"pairing"}'
Zigbee2MQTT:debug 2021-03-09 17:59:25: Received Zigbee message from 'arke', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-03-09 17:59:25: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2021-03-09 17:59:25: Received Zigbee message from 'arke', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-03-09 17:59:25: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2021-03-09 17:59:25: Received Zigbee message from 'arke', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-03-09 17:59:25: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:debug 2021-03-09 17:59:25: Received Zigbee message from 'arke', type 'readResponse', cluster 'genBasic', data '{"zclVersion":2}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-03-09 17:59:25: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:warn  2021-03-09 17:59:25: Device '0xf4ce361fc2bf0e3e' left the network
Zigbee2MQTT:info  2021-03-09 17:59:25: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"ieee_address":"0xf4ce361fc2bf0e3e"},"type":"device_leave"}'
Zigbee2MQTT:info  2021-03-09 17:59:25: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0xf4ce361fc2bf0e3e"},"type":"device_removed"}'
```

My SDK documentation is telling me:

```
this signal will be generated if the device tries to join legacy Zigbee network and the TCLK
exhange cannot be completed.
```
 
Any recommendations?

Best,
Cyril

MattWestb, 2021-03-09:
Then you is trying pairing one no Zigbee 3 device with one "true" Zugbee 3 coordinator..

Try with one "real" Zigbee 3 device like (most 95%) IKEA devices.

MattWestb, 2021-03-09:
"zclVersion":2 = Zigbee 3 or ZLL with Zigbee  PRO = can / need updating the trust center link key.
ZCL 1 = old HA or old not Zigbee PRO ZLL = cant changing trust center link key.

And then its up to the policy of the trust center in your coordinator is having if its accepting no / less security devices or not.

kirovilya, 2021-03-09:
@MattWestb can you tell me what config / policies it is set for? maybe my parameters are wrong

MattWestb, 2021-03-09:
I dont knowing them only the workflow then pairing real Zegbee 3 devices.
Look in bellows how its made for the V8 (its getting much for lower versions).

If you is putting the level to low then some Zigbee 3 devices cant complete joining /not getting one updated trust center link key) and to high legacy and Xiaomi devices cant joining cant requesting one updated trust center link key.

fouge, 2021-03-10:
I have the same error using bellows, but not when using Z3Gateway CLI and forming a Z3.0 network ([instructions](https://www.silabs.com/community/wireless/zigbee-and-thread/knowledge-base.entry.html/2018/04/24/how_to_form_zigbee3-zrzB)).

I'll keep you updated,
Best,
Cyril

Hedda, 2021-03-11:
> **I have the same error using bellows**, but not when using Z3Gateway CLI and forming a Z3.0 network ([instructions](https://www.silabs.com/community/wireless/zigbee-and-thread/knowledge-base.entry.html/2018/04/24/how_to_form_zigbee3-zrzB)).

Can I suggest that you report that as an issue to bellows developers as well(?) -> https://github.com/zigpy/bellows/issues

kirovilya, 2021-03-11:
let's play again with some changes...

MattWestb, 2021-03-11:
I have seen one god Silab dock with the security setting and explaining for some month ago but i cant finding it for the moment. I looking little more if i can finding it.

But dont setting the security to low then its going to being problem later then users is start using Zigbee R23 devices that is coming.

danielk117, 2021-03-12:
> let's play again with some changes...

Seems to solve all my problems! No errors since yesterday evening. 👍🏻

kirovilya, 2021-03-12:
@danielk117  I do not believe ;))

danielk117, 2021-03-15:
@kirovilya You’re right 😉 after some days of running the latest build, your changes fixes the restarts of z2m. It doesn‘t solve the „values not set on device“ or „values setted but no response from device“.

kirovilya, 2021-03-15:
I knew it. we need to figure out why the network is falling apart ... I have no idea how to figure it out yet.

Hedda, 2021-03-15:
Note! EmberZNet USB Zigbee adapters/dongles/sticks are highly recommended instead of using a remotely connected Ember module over a serial-to-ip bridge / serial forwarding server like ser2net, but if you still insist on using a remotely connected Ember module over a serial-to-ip bridge then the recommendation is to use a bridge model that features wired Ethernet instead of WiFi.

Regardless, it is not a good idea to use WiFi based solution such as Sonoff ZBBrige when developing or testing this EZSP adapter in Zigbee2MQTT or IoBroker as it for a fast is infamous for not giving a stable connection for production or troubleshooting this now.

FYI, Home Assistant ZHA and zigpy developers are now strongly recommending not to use any WiFi based EZSP bridges/gateways.

https://github.com/zigpy/bellows#warning-about-zigbee-to-wifi-bridges

The reason Ember remote bridges over serial-to-IP proxy server is not recommended is that clients using the EZSP serial protocol requires a robust connection between the EmberZNet Zigbee stack running on EFR32 MCU and the serial port of the Zigbee radio. With solutions such as *ITEAD Sonoff ZBBridge* or a Ser2Net serial proxy connection over a WiFi network it is expected to see `NCP entered failed state. Requesting APP controller restart` in the logs. This is a normal part of the operation and indicates there was a drop in communication which caused packet loss to occurred between the zigpy radio library and the Zigbee radio. The use of serial network proxies/bridges/servers over WiFi is therefore not recommended when wanting a stable Zigbee environment with Silicon Labs Ember based Zigbee radios. The developers of the [bellows radio library for the zigpy project](https://github.com/zigpy/bellows) has more information about this if needed.

kirovilya, 2021-03-15:
https://www.itead.cc/zigbee-3-0-usb-dongle.html

Hedda, 2021-03-16:
> very likely its not debug locked (but not verified). 

I asked ITead via mail before and got the reply that it will not be debug locked + should also feature the standard Gecko bootloader.

kirovilya, 2021-03-16:
Guys, can you have experience in the number of connected devices (for efr32 chip): directly to the coordinator and everything on the network?
I see that when configuring, you can set the configuration values, but it depends on the amount of memory on the chip.

MattWestb, 2021-03-16:
@kirovilya Its also depends if the parameter in the firmware and some cant being overridden. ZHA is left must untouched = using the firmware defaults and using the rest of ram for buffers.

On god collections of Silabs Zigbee KBs  if you not have finding it before: https://silabs-prod.adobecqms.net/community/wireless/zigbee-and-thread/knowledge-base.entry.html/2020/04/01/zigbee_thread_knowledgebasearticleslist-ih5r

kirovilya, 2021-03-16:
@MattWestb hmm ... and what are the indicators now? how many devices were there?
Thanks for the link

MattWestb, 2021-03-16:
Zihpy/bellows have making one "EZSP PRO" for NCP with V8 firmware with extended devices see: https://github.com/zigpy/bellows/pull/397

Hedda, 2021-03-16:
> @MattWestb hmm ... and what are the indicators now? how many devices were there?

I believe that bellows limited the number of children directly connected to the coordinator by default was 32 if not using "EZSP PRO" firmware? But now it sounds in https://github.com/zigpy/bellows/pull/397 as if they increased that default to 64 for non-Pro firmware or? And for "PRO"?

I understand that, as already noted, the hard limit is set in parameter for Zigbee NCP before compiling application firmware, meaning that the person who compiles the Zigbee coordinator firmware that you on your EFR32 hardware will use has to either know or guess the capability of the MCU hardware to estimate the amount of Zigbee devices that it can control, and as you said; much of that depends on amount om RAM on the chip used.

This gets further complicated as the EFR32 Series 1/2 contains a bunch of available SoCs that all have different flash and RAM, ex:

https://www.silabs.com/wireless/zigbee

https://www.silabs.com/wireless/zigbee/efr32mg12-series-1-socs

https://www.silabs.com/wireless/zigbee/efr32mg13-series-1-socs

https://www.silabs.com/wireless/zigbee/efr32mg13-series-1-modules

https://www.silabs.com/wireless/zigbee/efr32mg21-series-2-socs

https://www.silabs.com/wireless/zigbee/efr32mg22-series-2-socs

https://www.silabs.com/wireless/zigbee/efr32mg21-series-2-modules

MattWestb, 2021-03-16:
For my the best shall being no (ZERO) SED connected to the coordinator with EZSP then it can making problems for the SED then the coordinator is having "lazy periods".

The same problem is de(F)CONZ having and SEDs is working much better having one "normal router" as its parent (I have doing some 100 hours sniffing of IKEA controlling devices wit EZSP coordinators).

The best practice in Zigbee i starting having one backbone with good router so you is getting  on stable mesh and then  adding the SED to near by routers then its working  great.

MattWestb, 2021-03-16:
@Hedda Its not so easy saying wot is best or not.
Reed that and think it thru and you is knowing tittle more: https://www.silabs.com/community/wireless/zigbee-and-thread/knowledge-base.entry.html/2014/03/24/guidelines_for_large-pgj4

digiblur, 2021-03-16:
> Its using the same EZSP firmware as the Sonoff ZBB but the firmware is not signed so very likely its not debug locked (but not verified). Also possible using our "free" cocked firmware on it without flashing new boot loader with SWD.
> The latest firmware is without both hard and software flow control https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/Zigbee3.0_Dongle.
> 
> But still can having problem with comports in docker containers.

I have the stick and used it with z2m. If there are any other tests you want to do let me know.

gabeale, 2021-03-17:
Dear @digiblur can you please share the detailed steps for testing z2m with EZSP? In your guide at the following link: https://www.digiblur.com/2020/07/how-to-use-sonoff-zigbee-bridge-with.html I see only the configuration for ZHA.
Thanks in advance! :-)

kirovilya, 2021-03-17:
@gabeale :) https://www.digiblur.com/2021/03/zigbee2mqtt-with-sonoff-zigbee-bridge.html

digiblur, 2021-03-17:
> @gabeale :) https://www.digiblur.com/2021/03/zigbee2mqtt-with-sonoff-zigbee-bridge.html

Thanks for the catch!

gabeale, 2021-03-18:
@digiblur, thanks for the detailed instructions. I have followed this guide for Supervised HA but I'm not able to add in the configuration options the adapter setting (i.e. adapter: ezsp) because when I save it desappears. On the other hand Sonoff Zigbee bridge was working properly with ZHA (ref. previous guide), any suggestion?
P.S. RPI is running with Home Assistant OS 5.12, core-2021.3.4, Zigbee2mqtt Edge:latest add-on from "dwelch2101/zigbee2mqtt-edge-aarch64" repository

digiblur, 2021-03-18:
> @digiblur, thanks for the detailed instructions. I have followed this guide for Supervised HA but I'm not able to add in the configuration options the adapter setting (i.e. adapter: ezsp) because [...]

Make sure you have the latest z2m edge if you are running Supervisor based.

kirovilya, 2021-03-18:
Now you can test groups

kirovilya, 2021-03-18:
Yesterday I tried to perform an OTA update of the firmware for the IKEA bulb ... the update began, transferred about 4000 bytes and then the problems of sending packets began - the bulb stopped responding (or the stick stopped sending packets). Then the update stopped and after a while the work was restored.
There is a suspicion of some buffer settings or we need to pause ... until you figure it out.
Tested on EByte E180-ZG120B module

MattWestb, 2021-03-18:
@kirovilya Is your E180 110% working OK with its EZSP firmware ?
Its sounds like some parameters in the firmware is not good and its getting locked and not acking all packages.

I can being bad parameters setting in Z2M but or is being in the NCP firmware.

Have some one with other adapter trying doing OTA updating ?

Or trying doing one fast test with your E180 with ZHA and you can see its working OK doing OTA updating (you can modding ZHA doing firmware download 2 for testing OTA updating in Z2M).

If the problem is in the NCP firmware its very bad if you is trying finding somthing that is (not) bad in your Z2M code!!

kirovilya, 2021-03-18:
@MattWestb I think the firmware for the E180 is working, the rest of the actions work.
I think that I set some configuration parameters (in the z2m adapter) incorrectly.
later I will try on another stick and on the sonoff bridge

MattWestb, 2021-03-18:
Then its very OK only that you is not hunting things in the wrong place and going in the wall :-))

If you need downgrading firmware on IKEA device in ZHA you finding the patch here: https://github.com/home-assistant/core/issues/47177#issuecomment-797056118

deCONZ can do the same with manual selecting the firmware to updating the device with.

I have using both method and they is working well.

digiblur, 2021-03-18:
Anyone test the HUSBZB-1 with this after upgrading it to 6.7.8 firmware? I tried the other day along with another user but it just made Z2M hang at start up indefinitely. I will give it a shot again this weekend.

MattWestb, 2021-03-18:
For the EZSP 6.7.8.0 looks being stable on all devices i have getting firmware upgrades for and EZSP on the application APT level compatible but no machine cod compatible so all shall working the same with Z2M.

The only large problem ZHA is having is with Sonoff ZBB / ERF32MG2 devices that is making strange things.

Did you flashing the 115K2  or the 57K6 comport speed firmware on the HUSBZB-1 ?

digiblur, 2021-03-18:
> Did you flashing the 115K2 or the 57K6 comport speed firmware on the HUSBZB-1 ?

Both.  Which is recommended? (asking for a friend that is going to try it this weekend)

MattWestb, 2021-03-18:
The hardware dont have any problem with both but the 115K2 is getting the data thru the  pipe 100% faster = good. 
(the packet trace is running on 1M6 without hardware flow control on the chip so the weak part is the USB side)

If have the coordinator installed in ZHA you need patching the config for getting it working if you is changing the comport speed or you must delete and reinstall the ZHA integration with the correct comport speed. [Instruction for changing the comport speed ](https://github.com/zigpy/zigpy/wiki/Coordinator-Firmware-Updates#changing-the-the-baudrate-after-zha-integration-is-installed) fold the text out with the arrow.

digiblur, 2021-03-18:
I wasn't using ZHA.

MattWestb, 2021-03-18:
Then changing the comport speed in Z2M config and all shall working well :-)))

digiblur, 2021-03-18:
> Then changing the comport speed in Z2M config and all shall working well :-)))

Set the port to 57600 and z2m doesn't start, immediately fails saying it can't form the network.  The same port and device work fine on ZHA though.  

```
Assertion failed

Assertion failed

[31mZigbee2MQTT:error[39m 2021-03-18 17:48:13: Error while starting zigbee-herdsman

[31mZigbee2MQTT:error[39m 2021-03-18 17:48:13: Failed to start zigbee

[31mZigbee2MQTT:error[39m 2021-03-18 17:48:13: Exiting...

[31mZigbee2MQTT:error[39m 2021-03-18 17:48:13: Error: Failure forming network:139
```
```
Status: Downloaded newer image for walthowd/husbzb-firmware:latest
{"ports": [{"stackVersion": "6.7.8-373", "deviceType": "zigbee", "pid": "8A2A", "port": "/dev/ttyUSB2", "vid": "10C4"}]}
Found zigbee port at /dev/ttyUSB2 running 6.7.8-373
```

oriolism, 2021-03-19:
Im having issues pairing the IKEA Remote E1524/E1810. I can see the device on the webpage but it wont detect any button on the logs. Anyone has this working with the Sonoff ZbBrige? 

Zigbee2MQTT:info  2021-03-18 19:52:27: Configuring '0x14b457fffe4f0a12'

Zigbee2MQTT:error 2021-03-18 19:52:27: Failed to configure '0x14b457fffe4f0a12', attempt 2 (Error: Bind 0x14b457fffe4f0a12/1 genOnOff from '901' failed (Incorrect value passed)

    at new EmberEUI64 (/app/node_modules/zigbee-herdsman/dist/adapter/ezsp/driver/types/named.js:73:23)

    at EZSPAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/ezsp/adapter/ezspAdapter.js:375:33)

    at Generator.next (<anonymous>)

    at /app/node_modules/zigbee-herdsman/dist/adapter/ezsp/adapter/ezspAdapter.js:27:71

    at new Promise (<anonymous>)

    at __awaiter (/app/node_modules/zigbee-herdsman/dist/adapter/ezsp/adapter/ezspAdapter.js:23:12)

    at Object.func (/app/node_modules/zigbee-herdsman/dist/adapter/ezsp/adapter/ezspAdapter.js:373:52)

    at Queue.<anonymous> (/app/node_modules/zigbee-herdsman/dist/utils/queue.js:29:46)

    at Generator.next (<anonymous>)

    at /app/node_modules/zigbee-herdsman/dist/utils/queue.js:8:71)

Zigbee2MQTT:info  2021-03-18 19:52:27: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"TRADFRI remote control","friendly_name":"0x14b457fffe4f0a12","model":"E1524/E1810","supported":true,"vendor":"IKEA"},"type":"pairing"}'

kirovilya, 2021-03-19:
@digiblur Can you record debug logs? https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

[741 more replies]

closed 2024-11-23 by Nerivec