Koenkk/zigbee-herdsman #330: Support for missing ZCL foundation commands

aesedepece, 2021-03-15:
File `src/zcl/definition/foundation.ts` states:

https://github.com/Koenkk/zigbee-herdsman/blob/878388fb767d39a8f560179895a18345322f7f6a/src/zcl/definition/foundation.ts#L154-L165

Reading these lines, I understand that when `zigbee-shepherd` was refactored into `zigbee-herdsman` most recent ZCL foundation / global commands were intentionally left out just for the sake of simplicity and no real demand for those commands.

However, as more and more manufacturers adopt ZigBee, specially its quite opinionated 3.x version along the ZigBee Cluster Library, we're seeing an increasing number of devices making use of these ZCL commands, specially `readStruct`, `writeStruct` and `writeStructRsp`.

I found out about this the hard way when trying to implement the `toZigbee`converter for [Livolo covers][livolo], but I can imagine that many other devices are currently missing support for advanced settings in their Z2M integrations also because of this limitation.

I think it is time to support those commands, and I'd be happy to work on them.

I can't call myself an expert in the ZigBee stack at all, but I've been lurking the space for 10 years and I'm familiar with sniffing 802.15.4 traffic, well-seasoned in reverse-engineering stuff, and have written more TypeScript than I'd like to admit, so I feel confident I can come up with something that lives up to the quality standards of this project.

So, before I jump into implementing anything, I wanted to ask the devs here, specially @Koenkk, whether this is a good idea, or whether am I missing any reason why this shouldn't be worked on.

Cheers!

[livolo]: https://github.com/Koenkk/zigbee-herdsman-converters/pull/2325

Koenkk, 2021-03-16:
The reason that these were not added yet was indeed that we didn't have any devices which use these commands. If you have a device which does it would be great if you can implement it!

aesedepece, 2021-03-17:
Great, I'll go with `0x0f` / `writeStruct, as I can extensively test it with my Livolo cover.

I guess you prefer not to implement unused commands for the sake of keeping the code base as compact and easy to maintain as possible. Is this right? 

Btw, what's your take on the discovery commands? With more and more manufacturers adopting the 3.0 stack and certifying their products, it is to be expected that they support these commands, and that the capabilities of their devices should be perfectly discoverable and configurable with existing converters, so that new devices that are compliant with the ZCL specs don't necessarily need to be explicitly added to Z2M for them to work. I imagine that this is the way the ZHA integration works in HA. Is this maybe out of the scope or against the design decisions of this project?

Koenkk, 2021-03-17:
> I guess you prefer not to implement unused commands for the sake of keeping the code base as compact and easy to maintain as possible. Is this right?

Exactly, these commands are probably only used by very few devices. 99,99% of the devices use the already supported commands, Livolo is for sure an exception as they completely deviate from the ZCL spec.

> Is this maybe out of the scope or against the design decisions of this project?

It's out of scope for this issue indeed.
The main thought behind this is to have a list of devices which are confirmed to work and to provide a better integration. Note that currently mainly devices like TuYa are being added, these do not follow the ZCL spec so discovery is not possible.

aesedepece, 2021-03-18:
Thanks for the clarifications.

As soon as I start working on `writeStruct`, I'll close this too broad of an issue and rather open one that is specific to that command, so that the discussion can be more focused and we don't have open ended issues that stay unsolved forever.

aesedepece, 2021-03-20:
Superseded by #333 for now :rocket:

closed 2021-03-20 by aesedepece