Koenkk/zigbee-herdsman #332: Devices not identified (Tuya and 4Noks)

kuumaur, 2021-03-19:
Hi Guys,
it try to pair some devices from Tuya and 4-Noks.
I have created already issues at
https://github.com/ioBroker/ioBroker.zigbee/issues/1021
https://github.com/ioBroker/ioBroker.zigbee/issues/996

I have had some conversation with @asgothian, but we did not have luck.
His recommendation was to create an Issue here as last hope.
For the Tuya Door-/Window-Sensor i see after unsuccessful pairing this in ioBroker/zigbee-0:
    type:EndDevice
    nwk:38603
    manuf id:0
    interviewed:false
    configured:false

@asgothian assumes, that the identification information is correct in some way.
I have already some Aqara Door-Window-Contact-Sensors
The 4-Noks is a socket switch with measurement, like the Bitron AV2010/25 are.
What can I do to come a step closer to a solution?
Is it possible to copy an existing device configuration?

Br Kuumaur

Koenkk, 2021-03-19:
Can you temporarily switch to zigbee2mqtt and provide the herdsman debug log when pairing the device?

See https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

kuumaur, 2021-03-21:
Thanx a lot for your help.
I set up zigbee2mqtt.
I startet zigbee2mqtt by command line with the debug parameters.
I tried to pair both devices (Sensor and socket).
It produced a lot of output. I hope i did it all correctly.
What can i do next?

Here is the output:
root@fhem-prod:/opt/zigbee2mqtt# DEBUG=zigbee-herdsman* npm start

> zigbee2mqtt@1.18.1 start /opt/zigbee2mqtt
> node index.js

Zigbee2MQTT:debug 2021-03-21 13:53:51: Loaded state from file /opt/zigbee2mqtt/data/state.json
Zigbee2MQTT:info  2021-03-21 13:53:51: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2021-03-21.13-53-51' filename: log.txt
Zigbee2MQTT:info  2021-03-21 13:53:51: Starting Zigbee2MQTT version 1.18.1 (commit #a43d1f3)
Zigbee2MQTT:info  2021-03-21 13:53:51: Starting zigbee-herdsman (0.13.71)
Zigbee2MQTT:debug 2021-03-21 13:53:51: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyUSB0"}}'
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[247,241,17,140,208,139,205,233,51,87,98,157,41,243,230,119],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyUSB0"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","adapter":{"concurrent":null,"delay":null}}' +0ms
  zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyUSB0 and {"baudRate":115200,"rtscts":false,"autoOpen":false} +0ms
  zigbee-herdsman:adapter:zStack:znp:log Serialport opened +4ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [0] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [0] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [0] +301ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [0,0] +0ms
  zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload +455ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239] +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +0ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +1s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,6,65,128,0,2,1,2,7,1,192] +2s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [0,0,254,6,65,128,0,2,1,2,7,1,192] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 6 - 2 - 1 - 128 - [0,2,1,2,7,1] - 192 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- SYS - resetInd - {"reason":0,"transportrev":2,"productid":1,"majorrel":2,"minorrel":7,"hwrev":1} +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +6s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,2,97,1,89,6,61] +5s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,2,97,1,89,6,61] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 2 - 3 - 1 - 1 - [89,6] - 61 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - ping - {"capabilities":1625} +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - version - {} +4ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,2,35] +3ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,10,97,2,2,1,2,7,1,66,62,52,1,0,39] +3ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,10,97,2,2,1,2,7,1,66,62,52,1,0,39] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 10 - 3 - 1 - 2 - [2,1,2,7,1,66,62,52,1,0] - 39 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - version - {"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20201026} +4ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:adapter:zStack:adapter Adapter concurrent: 16 +0ms
  zigbee-herdsman:adapter:zStack:adapter Detected znp version 'zStack3x0' ({"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20201026}) +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvRead - {"id":96,"offset":0,"len":1,"value":{"type":"Buffer","data":[85]}} +5ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,33,8,96,0,0,74] +5ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,97,8,0,1,85,62] +13ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,97,8,0,1,85,62] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 3 - 1 - 8 - [0,1,85] - 62 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvRead - {"status":0,"len":1,"value":{"type":"Buffer","data":[85]}} +14ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:startZnp Item 'hasConfigured' is valid +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvRead - {"id":96,"offset":0,"len":1,"value":{"type":"Buffer","data":[85]}} +13ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,33,8,96,0,0,74] +13ms
[... cut ...]
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,14,69,132,0,0,0,0,0,8,5,8,1,5,0,0,0,0,206] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,14,69,132,0,0,0,0,0,8,5,8,1,5,0,0,0,0,206] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 14 - 2 - 5 - 132 - [0,0,0,0,0,8,5,8,1,5,0,0,0,0] - 206 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - simpleDescRsp - {"srcaddr":0,"status":0,"nwkaddr":0,"len":8,"endpoint":5,"profileid":264,"deviceid":5,"deviceversion":0,"numinclusters":0,"inclusterlist":[],"numoutclusters":0,"outclusterlist":[]} +6ms

kuumaur, 2021-03-21:
This is the 2nd try, hopefully it is only the interview of the socket switch 0x000d6f0000d393c9:
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,223,37,201,147,211,0,0,111,13,0,0,0,146] +2m
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,223,37,201,147,211,0,0,111,13,0,0,0,146] +0ms                                                                                                                 
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [223,37,201,147,211,0,0,111,13,0,0,0] - 146 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":9695,"extaddr":"0x000d6f0000d393c9","parentaddr":0} +2m
  zigbee-herdsman:controller:log Device '0x000d6f0000d393c9' joined +6m
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:controller:log Device '0x000d6f0000d393c9' accepted by handler +0ms
  zigbee-herdsman:controller:log New device '0x000d6f0000d393c9' joined +0ms
  zigbee-herdsman:controller:log Creating device '0x000d6f0000d393c9' +0ms
Zigbee2MQTT:info  2021-03-21 14:11:36: Device '0x000d6f0000d393c9' joined
Zigbee2MQTT:info  2021-03-21 14:11:36: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000d6f0000d393c9","ieee_address":"0x000d6f0000d393c9"},"type":"device_joined"}'
  zigbee-herdsman:controller:log Interview '0x000d6f0000d393c9' start +2ms
Zigbee2MQTT:info  2021-03-21 14:11:36: Starting interview of '0x000d6f0000d393c9'
Zigbee2MQTT:info  2021-03-21 14:11:36: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000d6f0000d393c9","ieee_address":"0x000d6f0000d393c9","status":"started"},"type":"device_interview"}'
  zigbee-herdsman:controller:device:log Interview - start device '0x000d6f0000d393c9' +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":9695,"nwkaddrofinterest":9695} +2m
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,223,37,223,37,35] +2m
Zigbee2MQTT:info  2021-03-21 14:11:36: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x000d6f0000d393c9"},"type":"device_connected"}'
Zigbee2MQTT:info  2021-03-21 14:11:36: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x000d6f0000d393c9"},"type":"pairing"}'
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102] +15ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0} +2m
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:adapter Node descriptor request for '9695' failed (Error: AREQ - ZDO - nodeDescRsp after 10000ms), retry +6m
  zigbee-herdsman:adapter:zStack:adapter Discovering route to 9695 +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - extRouteDisc - {"dstAddr":9695,"options":0,"radius":30} +10s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,69,223,37,0,30,128] +10s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,69,0,33] +10s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,69,0,33] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 69 - [0] - 33 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - extRouteDisc - {"status":0} +10s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":9695,"nwkaddrofinterest":9695} +3s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,223,37,223,37,35] +3s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102] +3s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0} +3s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
^CZigbee2MQTT:debug 2021-03-21 14:11:59: Saving state to file /opt/zigbee2mqtt/data/state.json
Zigbee2MQTT:info  2021-03-21 14:11:59: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'
Zigbee2MQTT:info  2021-03-21 14:11:59: Disconnecting from MQTT server
  zigbee-herdsman:controller:log Disable joining +23s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":3,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[25,4,2,11,0,0]}} +10s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,26,36,2,2,253,255,0,0,0,0,0,0,242,0,0,242,33,0,3,0,30,6,0,25,4,2,11,0,0,18] +10s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,2,0,103] +10s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,2,0,103] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 2 - [0] - 103 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequestExt - {"status":0} +10s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,242,3,54] +4ms
[... cut ...]
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - nvRead - {"status":0,"len":12,"value":{"type":"Buffer","data":[17,39,0,0,221,221,221,221,221,221,221,221]}} +5ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:controller:log Wrote coordinator backup to '/opt/zigbee2mqtt/data/coordinator_backup.json' +147ms
  zigbee-herdsman:adapter:zStack:znp:log Port closed +6m
Zigbee2MQTT:info  2021-03-21 14:11:59: zigbee-herdsman stopped
Zigbee2MQTT:info  2021-03-21 14:11:59: Stopped Zigbee2MQTT

kuumaur, 2021-03-21:
This is the entry from data/database.db:
{"id":2,"type":"Unknown","ieeeAddr":"0x000d6f0000d393c9","nwkAddr":9695,"epList":[],"endpoints":{},"interviewCompleted":false,"meta":{},"lastSeen":1616332296521}

kuumaur, 2021-03-21:
This is hopefully only the output from the door sensor 0x00124b001aa69ec5:
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,6,142,197,158,166,26,0,75,18,0,0,0,181] +39s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,6,142,197,158,166,26,0,75,18,0,0,0,181] +0ms                                                                                                                  
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [6,142,197,158,166,26,0,75,18,0,0,0] - 181 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":36358,"extaddr":"0x00124b001aa69ec5","parentaddr":0} +39s
  zigbee-herdsman:controller:log Device '0x00124b001aa69ec5' joined +39s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:controller:log Device '0x00124b001aa69ec5' accepted by handler +1ms
  zigbee-herdsman:controller:log New device '0x00124b001aa69ec5' joined +0ms
  zigbee-herdsman:controller:log Creating device '0x00124b001aa69ec5' +0ms
Zigbee2MQTT:info  2021-03-21 14:30:03: Device '0x00124b001aa69ec5' joined
Zigbee2MQTT:info  2021-03-21 14:30:03: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b001aa69ec5","ieee_address":"0x00124b001aa69ec5"},"type":"device_joined"}'
  zigbee-herdsman:controller:log Interview '0x00124b001aa69ec5' start +11ms
Zigbee2MQTT:info  2021-03-21 14:30:03: Starting interview of '0x00124b001aa69ec5'
Zigbee2MQTT:info  2021-03-21 14:30:03: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b001aa69ec5","ieee_address":"0x00124b001aa69ec5","status":"started"},"type":"device_interview"}'
  zigbee-herdsman:controller:device:log Interview - start device '0x00124b001aa69ec5' +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":36358,"nwkaddrofinterest":36358} +39s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,6,142,6,142,35] +39s
Zigbee2MQTT:info  2021-03-21 14:30:03: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x00124b001aa69ec5"},"type":"device_connected"}'
Zigbee2MQTT:info  2021-03-21 14:30:03: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x00124b001aa69ec5"},"type":"pairing"}'
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102] +19ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0} +39s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,13,69,193,6,142,6,142,197,158,166,26,0,75,18,0,128,183] +512ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,13,69,193,6,142,6,142,197,158,166,26,0,75,18,0,128,183] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 13 - 2 - 5 - 193 - [6,142,6,142,197,158,166,26,0,75,18,0,128] - 183 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - endDeviceAnnceInd - {"srcaddr":36358,"nwkaddr":36358,"ieeeaddr":"0x00124b001aa69ec5","capabilities":128} +533ms
  zigbee-herdsman:controller:log Device announce '0x00124b001aa69ec5' +522ms
Zigbee2MQTT:debug 2021-03-21 14:30:03: Device '0x00124b001aa69ec5' announced itself
Zigbee2MQTT:info  2021-03-21 14:30:03: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b001aa69ec5","ieee_address":"0x00124b001aa69ec5"},"type":"device_announce"}'
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +3ms
Zigbee2MQTT:info  2021-03-21 14:30:03: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x00124b001aa69ec5"},"type":"device_announced"}'
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,27,68,129,0,0,1,0,6,142,1,1,0,30,0,110,182,47,0,0,7,24,1,10,32,0,32,30,6,142,29,33] +95ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,1,0,6,142,1,1,0,30,0,110,182,47,0,0,7,24,1,10,32,0,32,30,6,142,29,33] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 27 - 2 - 4 - 129 - [0,0,1,0,6,142,1,1,0,30,0,110,182,47,0,0,7,24,1,10,32,0,32,30,6,142,29] - 33 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":1,"srcaddr":36358,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":30,"securityuse":0,"timestamp":3126894,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[24,1,10,32,0,32,30]}} +99ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":1,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":32,"dataType":32,"attrData":30}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":36358,"endpoint":1,"linkquality":30,"groupID":0}' +100ms
  zigbee-herdsman:controller:log 'zcl' data is from unknown endpoint '1' from device with network address '36358', creating it... +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +4ms
[... cut ...]
Zigbee2MQTT:debug 2021-03-21 14:30:03: Received Zigbee message from '0x00124b001aa69ec5', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":160}' from endpoint 1 with groupID 0
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,1,0,6,142,1,1,0,30,0,71,185,47,0,0,7,24,3,10,32,0,32,30,6,142,29,5] +2ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 27 - 2 - 4 - 129 - [0,0,1,0,6,142,1,1,0,30,0,71,185,47,0,0,7,24,3,10,32,0,32,30,6,142,29] - 5 +0ms

kuumaur, 2021-03-21:
This is the entry in database.db
{"id":3,"type":"EndDevice","ieeeAddr":"0x00124b001aa69ec5","nwkAddr":36358,"manufId":0,"epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1,1280],"outClusterList":[0,3,1,1280],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":false,"meta":{},"lastSeen":1616333439937}

Koenkk, 2021-03-21:
Seems the device doesn't respond, what coordinator are you using and can you pair it close to the coordinator?

Also next time please provide yours logs via e.g. pastebin.com to avoid this issue getting lengthy.

kuumaur, 2021-03-21:
This is what lsusb tells me:
Cygnal Integrated Products, Inc. CP2102/CP2109 UART Bridge Controller [CP210x family]

kuumaur, 2021-03-21:
Since the test with zigbee2mqtt, i cant bring up the iobroker.zigbee adapter, even after reboot. Do they both influence each other?

kuumaur, 2021-03-21:
This is the log from the door sensor, the distance is just around 30cm.
https://pastebin.com/n1h5xL5J

kuumaur, 2021-03-22:
This is the log from the socket switch, the distance is just around 50cm.
https://pastebin.com/idhuSwiA

Koenkk, 2021-03-22:
> This is the log from the socket switch, the distance is just around 50cm.
> https://pastebin.com/idhuSwiA

I've check this log, but you are stoping z2m to soon, wait until you see `interview_failed` in the log

kuumaur, 2021-03-23:
> > This is the log from the socket switch, the distance is just around 50cm.
> > https://pastebin.com/idhuSwiA
> 
> I've check this log, but you are stoping z2m to soon, wait until you see `interview_failed` in the log

I hope, that a have catched it all:
https://pastebin.com/37DwUwqq

kuumaur, 2021-03-23:
> > This is the log from the socket switch, the distance is just around 50cm.
> > https://pastebin.com/idhuSwiA
> 
> I've check this log, but you are stoping z2m to soon, wait until you see `interview_failed` in the log

And this is the log from the door sensor:
https://pastebin.com/91Me5Qqu

Koenkk, 2021-03-23:
Switch doesn't look very good, door sensor gets further but stops responding.

To debug this further you will have to sniff the traffic of pairing with both Zigbee2MQTT and the original gateway so we can compare them. (https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)

kuumaur, 2021-03-25:
Thank you very much for your patience. I appreciate your work really.
I dont have the original gateway. Do you see a chance to get further if i buy a Gateway?
I dont want to spend more money on a possibly dead horse. I was assuming, that Zigbee is somehow open and compatible. That was the reason, why I did the switch over to Zigbee. As @Asgothian said, there are some Vendors they dont care about standards.
Is it a dead horse?

Thanks in advance
Kuumaur

Koenkk, 2021-03-25:
@kuumaur chances that we can get it working are not very big indeed. Personally not really a fan of the TuYa, I like Xiaomi much more.

kuumaur, 2021-03-26:
Ok, thank you for your open words. I stop here and bury these devices.

closed 2021-03-26 by kuumaur