Koenkk/zigbee-herdsman #338: Configure reporting crashes

asgothian, 2021-03-23:
I have been trying to track down an issue with the zigbee-herdsman where a call to ConfigureReporting fails when the target device returns a "134 UNSUPPORTED ATTRIBUTE".

Unfortunately, this fail results in an uncaught promise rejection which cannot be averted through the calling code.

As far as i was able track this down, the issue lies in the "throw" in line 572 in the function 'ConfigureReporting'.

I have placed the call to endpoint.configureReporting() in a try/catch and was unable to catch the error thrown by the function. Unfortunately, i have not enough experience with TS libraries and their installation from GitHub (if this is possible at all), so i am unable to verify what exactly is going on. I did try to work my way through the compiled js code in my installation but found no way to catch the error raised by the function.

While i understand that throwing an error may be a good method of propagating an error through the code, throwing an error which forces the calling code to crash does not make sense.

I would appreciate
- any hints as to how to get this debugged and fixed
- a fix for this solution.

A.

Koenkk, 2021-03-24:
Can you provide the code you call the configure reporting function with + a log?

asgothian, 2021-03-24:
The log file section. Note that it is with debug info from the zigbee-herdsman.

```
2021-03-23 20:20:56.873  - [34mdebug[39m: zigbee.0 (2761) zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal %s:%i/%i (%i,%i,%i)
2021-03-23 20:20:56.891  - [34mdebug[39m: zigbee.0 (2761) zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100,254,3,68,128,0,1,94,152,254,24,68,129,0,0,6,0,105,49,1,1,0,105,0,125,216,237,0,0,4,8,71,7,0,105,49,29,171]
2021-03-23 20:20:56.891  - [34mdebug[39m: zigbee.0 (2761) zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100,254,3,68,128,0,1,94,152,254,24,68,129,0,0,6,0,105,49,1,1,0,105,0,125,216,237,0,0,4,8,71,7,0,105,49,29,171]
2021-03-23 20:20:56.892  - [34mdebug[39m: zigbee.0 (2761) zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2021-03-23 20:20:56.892  - [34mdebug[39m: zigbee.0 (2761) zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2021-03-23 20:20:56.893  - [34mdebug[39m: zigbee.0 (2761) zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,94,152,254,24,68,129,0,0,6,0,105,49,1,1,0,105,0,125,216,237,0,0,4,8,71,7,0,105,49,29,171]
2021-03-23 20:20:56.893  - [34mdebug[39m: zigbee.0 (2761) zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,94] - 152
2021-03-23 20:20:56.893  - [34mdebug[39m: zigbee.0 (2761) zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":94}
2021-03-23 20:20:56.894  - [34mdebug[39m: zigbee.0 (2761) zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,24,68,129,0,0,6,0,105,49,1,1,0,105,0,125,216,237,0,0,4,8,71,7,0,105,49,29,171]
2021-03-23 20:20:56.894  - [34mdebug[39m: zigbee.0 (2761) zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 24 - 2 - 4 - 129 - [0,0,6,0,105,49,1,1,0,105,0,125,216,237,0,0,4,8,71,7,0,105,49,29] - 171
2021-03-23 20:20:56.895  - [34mdebug[39m: zigbee.0 (2761) zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":12649,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":105,"securityuse":0,"timestamp":15587453,"transseqnumber":0,"len":4,"data":{"type":"Buffer","data":[8,71,7,0]}}
2021-03-23 20:20:56.896  - [34mdebug[39m: zigbee.0 (2761) zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":71,"manufacturerCode":null,"commandIdentifier":7},"Payload":[{"status":0}],"Command":{"ID":7,"name":"configReportRsp","parameters":[{"name":"status","type":32},{"name":"direction","type":32,"conditions":[{"type":"minimumRemainingBufferBytes","value":3}]},{"name":"attrId","type":33,"conditions":[{"type":"minimumRemainingBufferBytes","value":2}]}]}},"address":12649,"endpoint":1,"linkquality":105,"groupID":0}'
2021-03-23 20:20:56.897  - [34mdebug[39m: zigbee.0 (2761) zigbee-herdsman:controller:endpoint DefaultResponse 0x00158d00025c9c87/1 6(7, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2021-03-23 20:20:56.898  - [34mdebug[39m: zigbee.0 (2761) zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-03-23 20:20:56.898  - [34mdebug[39m: zigbee.0 (2761) zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":41163,"destendpoint":1,"srcendpoint":1,"clusterid":8,"transid":95,"options":0,"radius":30,"len":12,"data":{"type":"Buffer","data":[16,76,6,0,0,0,32,0,0,16,14,1]}}
2021-03-23 20:20:56.899  - [34mdebug[39m: zigbee.0 (2761) zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,22,36,1,203,160,1,1,8,0,95,0,30,12,16,76,6,0,0,0,32,0,0,16,14,1,120]
2021-03-23 20:20:56.900  - [34mdebug[39m: zigbee.0 (2761) zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal %s:%i/%i (%i,%i,%i)
2021-03-23 20:20:56.915  - [34mdebug[39m: zigbee.0 (2761) No converter available for 'LED1649C5' with cluster 'genLevelCtrl' and type 'read'
2021-03-23 20:20:56.917  - [34mdebug[39m: zigbee.0 (2761) No converter available for 'LED1649C5' with cluster 'genLevelCtrl' and type 'read'
2021-03-23 20:20:56.919  - [34mdebug[39m: zigbee.0 (2761) No converter available for 'ICPSHC24-10EU-IL-1' with cluster 'genLevelCtrl' and type 'read'
2021-03-23 20:20:56.952  - [34mdebug[39m: zigbee.0 (2761) zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100,254,3]
2021-03-23 20:20:56.952  - [34mdebug[39m: zigbee.0 (2761) zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100,254,3]
2021-03-23 20:20:56.952  - [34mdebug[39m: zigbee.0 (2761) zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2021-03-23 20:20:56.953  - [34mdebug[39m: zigbee.0 (2761) zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2021-03-23 20:20:56.953  - [34mdebug[39m: zigbee.0 (2761) zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3]
2021-03-23 20:20:56.953  - [34mdebug[39m: zigbee.0 (2761) zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":10387,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":96,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,63,11,7,0]}}
[... cut ...]
        this.debug(`group: ${safeJsonStringify(group)}`);
        await entity.endpoint.addToGroup(group.mapped);
        callback();
    }
```

This entire code change was an attempt to resolve an issue where grouped lights would not report on their state change back to the iobroker, and as such show the wrong state / brightness information when switched by a group. The idea was (is) to enable reporting on the devices which support it in order to keep the iobroker state and state from the lamp in sync  if it was switched from within a group.

asgothian, 2021-03-24:
What i forgot to mention - and i no longer have that log file:

I added a debug message at the start of the configureReporting function in endpoint.js in the zigbee-herdsman which just gave me the call stack. I could verify that the above code directly calls the herdsman function with no layer in between.

A.

Koenkk, 2021-03-24:
You should put `await` before `entity.endpoint.configureReporting`.

asgothian, 2021-03-24:
I'll give that a go, but not today. would be great if that fixes it. 

Thanks for the suggestion.

A.

asgothian, 2021-03-24:
Actually, i did manage to test it tonight. It does fix it. I will have to read a lot more about that in order to figure out exactly why it worked.

Thanks a lot for the help.

A.

closed 2021-03-24 by asgothian