Koenkk/zigbee-herdsman #340: High CPU usage on Raspberry Pi Zero with RaspBee II in Docker zigbee2mqtt image

jbieling, 2021-03-25:
I am seeing a high CPU usage of around 30-40% on a Raspberry Pi Zero with a "deconz" adapter (RaspBee II). The setup is using a zigbee-herdsmann 0.13.71 (used by zigbee2mqtt 1.18.1) in a Docker container running on a Raspberry Pi Zero with a RaspBee II adapter. For testing, there was only one sensor (IKEA motion sensor) connected.

Specifying an "adapter_delay" in the "advanced" section of the configuration.yaml for zigbee2mqtt does not change anything, even though with `DEBUG=zigbee-herdsman* npm start` I can see a log entry about the delay being set accordingly.

Some digging led me to drivers.js of the deconz adapter. I found the delays for `PROCESS_QUEUES` and `HANDLE_DEVICE_STATUS_DELAY` are in fact updated in `setDelay`, but the timers/intervals are actually already running with the original delays. Changing the hard-coded default delays for those variables from 5 to 50 works and reduces CPU usage to about 6-8%.

Koenkk, 2021-03-26:
@ChrisHae can you look into this?

riccardo-springtech, 2021-03-30:
I have the same problem, on my Raspberry Pi Zero the cpu load of zigbee2mqtt was always around 30 - 40 %.
This is my setup:
- ConnBee II USB adapter with the latest firmware (0x26680700)
- node.js version v14.16.0
- zigbee2mqtt  1.18.1-dev commit: 010a1408
- seven zigbee devices: switches, bulbs, sockets, temperature sensors
.
Your fix reduces the CPU usage to 5-8%.
Thanks.

jbieling, 2021-05-02:
Compared to using fixed delays as described above, zigbee-herdsmann 0.13.92 (used by zigbee2mqtt 1.18.3) now results in very similar CPU usage. Thanks for the fix!

@ChrisHae, out of interest: why are you using multiple timers instead of a single one (they all have the same interval)? Are there side effects to be expected, if all calls are performed in a single timer? I am asking, because the CPU usage is now higher than in my original post, which is likely due to a higher number of devices (now 19 in total, mostly bulbs, some sensors, ... ). While a higher load can probably be expected with a higher number of devices to be managed, I am also hoping the load could be reduced by reducing the number of timers.

closed 2021-06-10 by github-actions[bot]