Koenkk/zigbee-herdsman #342: Error UnhandledPromiseRejectionWarning

glsf91, 2021-03-29:
I have a Sonoff SNZB-03 motion sensor. I use this firmware https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_other_coordinator_20210120.zip on a CC2652P
(Ebyte E72).
I got this error:

```
Mar 29 17:10:06 ubuntu1 npm[30551]: Zigbee2MQTT:info  2021-03-29 17:10:06: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"SonoffMotionSensor","ieee_address":"0x00124b0022fd35a1"},"type":"device_announce"}'
Mar 29 17:10:06 ubuntu1 npm[30551]: Zigbee2MQTT:info  2021-03-29 17:10:06: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"SonoffMotionSensor"},"type":"device_announced"}'
Mar 29 17:10:06 ubuntu1 npm[30551]: Zigbee2MQTT:info  2021-03-29 17:10:06: MQTT publish: topic 'zigbee2mqtt/SonoffMotionSensor', payload '{"battery_low":false,"linkquality":24,"occupancy":true,"tamper":false}'
Mar 29 17:10:49 ubuntu1 npm[30551]: (node:30600) UnhandledPromiseRejectionWarning: Error: Command 0x00124b0022fd35a1/1 ssIasZone.enrollRsp({"enrollrspcode":0,"zoneid":23}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'MAC transaction expired' (240))
Mar 29 17:10:49 ubuntu1 npm[30551]:     at ZStackAdapter.<anonymous> (/home/john/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:339:27)
Mar 29 17:10:49 ubuntu1 npm[30551]:     at Generator.next (<anonymous>)
Mar 29 17:10:49 ubuntu1 npm[30551]:     at fulfilled (/home/john/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:24:58)
Mar 29 17:10:49 ubuntu1 npm[30551]: (node:30600) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
Mar 29 17:10:49 ubuntu1 npm[30551]: (node:30600) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

```

Is see the device_announce a few times a day for only this device. My other sensors are from Xiaomi.

glsf91, 2021-05-01:
This is still a problem

glsf91, 2021-06-01:
This is still a problem (avoid closing)

glsf91, 2021-07-03:
Prefent stale

glsf91, 2021-08-03:
prevent stale

MattWestb, 2021-08-03:
`device_announce` is made then one device is rejoining the network after being re powered (routers) or have losing / not liking its parent and "jumping" to one new parent (end devices).
I have 2 Philips HUE motion sensors that is "jumping around" all the time and using one of the worse parent they can find in the other end of the apartment but i cant do anything about it until i baying 2 new light sensors for replace them with (i primary using the light  sensor in then and Aqaras for temperature).

I dont knowing if the Sonoff motion sensor is not liking some routers as its parent but i think its normal and dot have large problems like most older Xiaomi sensor is having (but they is leaving and not coming back if they is getting problems).

glsf91, 2021-09-27:
prevent stale

glsf91, 2021-11-29:
prevent stale

closed 2021-12-07 by github-actions[bot]