Koenkk/zigbee-herdsman #354: Add support for setting coordinator as "Primary Trust Center"

H3buss, 2021-04-28:
I'm trying to add support for Schneider Wiser CCTFR6700, an electric heater driver supporting "fil pilot".

During pairing with vendor's hub, this controller performs a _Node Descriptor Request_ on the coordinator, then requests _Trust Center Link Key (0x04)_ from coordinator.

When pairing with an unofficial coordinator ( tried zigate and conbee II ), it stops at _Node Descriptor Request_.
Note that device function partially after that, it can drive the heater and it offers thermostat function, but does not report measured temp and it's status LED is red.

I compared the responses to the _Node Descriptor Request_ and the main difference seems to be _Primary Trust Center_ from _Server Flags_ being **True** in the offical hub's reponse and **False** in DIY coordinators.

Hence the question, how can we set this **_Primary Trust Center_** bit to _**True**_?
Is it even possible with available hardware?

H3buss, 2021-04-28:
After a bit of searching, it seems that this _Primary Trust Center_ bit should be set using the Z-stack firmware.
I'll try it using my CC2531, but I will be blind as I currently use it for sniffing.

closed 2021-04-30 by H3buss

MattWestb, 2021-05-01:
Requesting and getting one updated trust center link key is on Zigbee 3 function.
If the device is have the policy that it must have one updated trust center link key but cant getting on from the trust center its leaving the network then its cant working with the default link key.

One example is IKEA lights with old firmware (ZLL) is not requesting and not getting on updated link key and is happy with that and new zigbee 3  firmware is requesting and using it if the coordinator is supporting it or they is happy with on default link key (policy in the device).

H3buss, 2021-05-02:
I'm really struggling to indentify what goes wrong whith this device, so I'm trying to make the coordinator behave as close as possible to the official hub.

I hacked into the firmware and I managed to get the device request and verifiy this TC link key, but it still does not behave as intended after that (blinking red and not reporting temperature as expected).
I went as afar as making the coordinator return the exact same key as the official hub, but it did not do the trick. I'm still searching.

I found out that for some reason the device makes one association requests, then another one after 20 seconds. This is not the case with the official hub, so this is my current lead.

MattWestb, 2021-05-02:
The updating TC link key is good described in [AN1233: Zigbee Security](https://www.silabs.com/documents/public/application-notes/an1233-zigbee-security.pdf) 3.3.3 Requesting a New Link Key after Joining.

You can also looking if somthing else is going  wrong in the pairing by comparing how it shall working and how the device is doing it in the real.

H3buss, 2021-05-03:
I noticed that I was unable to set the extended PanID with Z-Stack 3.0.x
The stack uses the extended address as extended PanID. I tryied hard coding it in the code but no success.

MattWestb, 2021-05-03:
If you changing is setting in Z2M and the coordinator already have formed on network you can changing all networks parameters but its not taking effect before you doing on leave and start on new network with commands to the coordinator.

Its possible doing it by network update command if the coordinator is supporting it.

And i think the CC-2530/1 with 3.0.X is not possible changing those parameters then its hard coded in the firmware but im not sure of that.

H3buss, 2021-05-03:
I'm setting the Extended PanId using extended address, this is a hack but I'm not sure if this as any effect on the device anyway.

I'm currently investigating on those multiple association requests from the device. I've extended my tests with the official hub and it seems to occur with it as well.  Howeverthe official hub seems to wait for the Device announce, just after assocication before starting its interview process.

When using herdsman, the device re-associate during the interview. Before it does there is a few second where the device seems hanged (not responding to requests). The interview process continues, when the device comes back, ingnoring the re-associations.

How can I wait for device announce before starting interview? Or restart interview on association request?

H3buss, 2021-05-03:
I moved the interview process into onDeviceAnnounce()
The behavior is really close that what I get with the hub. But still no luck with the final behavior of the device.