Koenkk/zigbee-herdsman #377: Pairing not possible

emddummi, 2021-06-04:
Error: Failed to open the network: Error: SRSP - UTIL - ledControl after 6000ms at Timeout._onTimeout (/opt/iobroker/node_modules/zigbee-herdsman/dist/utils/waitress.js:37:35) at listOnTimeout (internal/timers.js:554:17) at processTimers (internal/timers.js:497:7). undefined Pairing started

Zigbee Modul: CC2538 + CC2592

I use IOBROKER and this is an issue for herdsman...

Koenkk, 2021-06-04:
@0x3EC can you confirm the CC2538 firmwares do not support LED?

asgothian, 2021-06-05:
I have seen this issue with a number of cc2538+2592 coordinators as well as some cc2530 coordinators. I tried to get a cc2538 based coordinator myself to pinpoint the issue, but the one i got does support the LED function. 
@koenkk - is this something which is actively done by the herdsman ? If so, can we add an optional parameter to the call to open the network to block the led switching ? I am asking because the zigbee Adapter will attempt to turn off the LED on the coordinator, and as such know if the specific stick will support this feature. We could pass that information on to the Herdsman when opening the network.

A

Koenkk, 2021-06-05:
Fixed now in the latest herdsman version.

@asgothian we determine if an adapter supports led based on: https://github.com/Koenkk/zigbee-herdsman/blob/c0ac12fb5e344b7e4bfc8964e24e10c943a535d8/src/adapter/z-stack/adapter/zStackAdapter.ts#L116 , but this does not seem 100% correct since e.g. the CC2530 on the 20190608 firmware doesn't support it (while the CC2531 does). Fixed by https://github.com/Koenkk/zigbee-herdsman/blob/c0ac12fb5e344b7e4bfc8964e24e10c943a535d8/src/adapter/z-stack/adapter/zStackAdapter.ts#L225

closed 2021-06-05 by Koenkk