Koenkk/zigbee-herdsman #379: Unable to attempt a 2nd start if the first herdsman start fails

asgothian, 2021-06-06:
Hi

Sorry for yet another issue, but the backup refactor has raised a number of issues to the point that i needed to track down an error which has been causing issues for a while now.

Consider the following scenario:
- A Herdsman Instance is created with a set of parameters.
- An attempt is made to start this instance, but during start an error causes this to not work. Important detail here is that the serial port is actually opened in this scenario.
- The startup procedure throws an error.

At this point, the client application is in an unrecoverable state:
- attempting to stop the Herdsman will throw an error at Database save (the log of this error is available below)
- If this exception is unhandled the client application will be terminated.
- If this exception is caught, the client application will remain alive, with parts of the herdsman still running (e.g. the serial port is still open)
- I could not find an option to close the connection to the hardware
- I could not find an option to update the settings and attempt a "start" again

This is causing a few headaches with the iobroker.zigbee adapter - sending it into a reconnect loop which it cannot break free of as the serial port is never closed and then throwing a confusing error which has nothing to do with the actual cause of the issue.

I see two of options to deal with this on the iobroker side, unfortunately, neither of them is ideal:
- Always take down the client application and force a full restart of the client. This is bad for cases where the error should be recoverable, e.g. is caused due to a temporary disconnect (serial port/network port)
- Distinguish between fatal errors and recoverable errors and only take down the adapter when a fatal error occurs - this requires a complex parsing of error messages not designed for this (see issue #376)

Outside of this, the remaining option would be to ensure that the 'stop' command on the herdsman side always shuts down all subsystems.

```
TypeError: Cannot read property 'getEntries' of null
    at Function.loadFromDatabaseIfNecessary (/opt/iobroker/node_modules/zigbee-herdsman/dist/controller/model/device.js:220:55)
    at Function.all (/opt/iobroker/node_modules/zigbee-herdsman/dist/controller/model/device.js:240:16)
    at Controller.databaseSave (/opt/iobroker/node_modules/zigbee-herdsman/dist/controller/controller.js:262:45)
    at Controller.<anonymous> (/opt/iobroker/node_modules/zigbee-herdsman/dist/controller/controller.js:243:18)
    at Generator.next (<anonymous>)
    at /opt/iobroker/node_modules/zigbee-herdsman/dist/controller/controller.js:27:71
    at new Promise (<anonymous>)
    at __awaiter (/opt/iobroker/node_modules/zigbee-herdsman/dist/controller/controller.js:23:12)
    at Controller.stop (/opt/iobroker/node_modules/zigbee-herdsman/dist/controller/controller.js:241:16)
    at ZigbeeController.start (/opt/iobroker/node_modules/iobroker.zigbee/lib/zigbeecontroller.js:122:33)
```

I would like to propose to change the 'stop' function by adding try/catch around specific sections, but am hesitant to make a PR as i cannot test this at this time. I have tried to set up the environment to compile the typescript code but am running into errors i am unsure on how to fix.

A.

Koenkk, 2021-06-07:
I'm not sure at what point it fails (would require more debugging) but setting up a development environment can be done the folllowing way:
- Clone this repo in the node_modules folder of iobroker
- `cd zigbee-herdsman`
- `npm ci`
- `npm run build` (to compile)
- start iobroker

Now if you make changes, stop iobroker, run npm run build again and start iobroker.

asgothian, 2021-06-07:
I'll give this a try tonight, and let you know.

A.

asgothian, 2021-06-14:
Closed - issue should be fixed.

closed 2021-06-14 by asgothian