Koenkk/zigbee-herdsman #382: Add Shade cluster

kds69, 2021-06-08:
Hi All,
In order (to try) to come over lack of support of "Stop" command of Lupus Rolladen relais (LS12128), I would need to be able to read/write shade attributes.
Indeed this is where I guess is hosted missing currentLevel feature + stop for this device (while it should have been in the Level Cluster and OnOff cluster but it is what it is).
I couldn't find anything from Shade cluster mentioned anywhere in the converters. Thus kindly asking if it is there and if not if it can be added.

Note that overall, I am trying to improve the integration of those shutter relays (ideally up to tilt functionality): in particular new hardware version which is not supported yet and levelconfig, which would be part of a near future PR.
I've tested successfully both with external converter. But the above request would be cherry on the cake.
Kind Regards

renne, 2021-06-13:
@kds69 
How does one set the shutter up/down runtime with zigbee2mqtt for the LS12128?

kds69, 2021-06-13:
For this you need to extend Levelconfig to the device (see external device...). Then you can publish below message, delays in 1/10th of a second. In exemple below, down takes 32.6 seconds and up 33.2 seconds:

zigbee2mqtt/entity_id/set
{"level_config": {
        "off_transition_time": 326,
        "on_off_transition_time": 332,
        "on_transition_time": 332
		}
}

renne, 2021-06-13:
Does that set the Up/down runtime in zigbee2mqtt or directly on the device (4 minutes Default)?

closed 2021-07-21 by github-actions[bot]