Koenkk/zigbee-herdsman #389: zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'

dummys, 2021-07-12:
Hi,

In the code: 
https://github.com/Koenkk/zigbee-herdsman/blob/de3630fa30f0de1d3a128e0ece7e8d165867f48b/src/adapter/adapter.ts#L94
I found the error I get and I try to understand if this "debug error" is a problem or not. It seems that it failed to validate the path of the device ? In my case my configuration.yaml has this:

yaml```
serial:
  port: >-
    /dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_4F_24_0F-if00-port0
```

Koenkk, 2021-07-12:
I don't think this is a herdsman bug but rather an issue with the addon (`udevadm` missing as you mentioned). Note that this also requires `--privilliged=true`, I don't know if HA addons support this.

dummys, 2021-07-14:
Ok I see. As the dongle is still working without it, we can safely close it.

closed 2021-07-14 by dummys