Koenkk/zigbee-herdsman #397: Z-Stack 1.2 resets when updating from Zigbee2MQTT 1.18.1 to 1.20.0 with custom epid

Koenkk, 2021-07-30:
When updating from Zigbee2MQTT 1.18.1 to 1.20.0 with a custom epid on a Z-Stack 1.2 firmware (tested with CC2531) a reset happens and all paired devices are lost. This seems to be caused by https://github.com/Koenkk/zigbee-herdsman/pull/303 (cc @castorw)

To reproduce:
- `cd` to your Zigbee2MQTT git repo
- Add the following to `configuration.yaml`

```yaml
advanced:
 ext_pan_id: [0xAB, 0xDF, 0xAA, 0xCE, 0x01, 0xDD, 0xDD, 0xBE]
```
- Execute `git checkout 1.18.1 && npm ci && npm start`
- Stop z2m with CTRL + C
- Execute `git checkout 1.20.0 && npm ci && DEBUG=zigbee-herdsman:adapter:zStack:startup* npm start`
- Observe that a reset is executed since adapter state does not match configuration:

[image]

castorw, 2021-07-31:
@Koenkk I am unable to reproduce this. Followed your instructions to the point and tried several times, even removed `node_modules` while swapping versions and the adapter starts up okay after the update without re-commissioning the network, only complaining about reversed EPID.

[image]

I have used CC2531 stick with `CC2531_DEFAULT_20190608` firmware. Can you provide me with your firmware image?

Koenkk, 2021-08-02:
I also cannot reproduce this anymore with a fresh flashed stick. I think the stick I tried with somehow became corrupted (given that the diff in epid was also very random).

closed 2021-08-02 by Koenkk