Koenkk/zigbee-herdsman #406: 0.13.124 regression (SRSP - SYS - version after 6000ms)

islepnev, 2021-08-11:
Setup: CC2531, zstack 1.2
Problem occurred after zigbee2mqtt upgrade 1.18.0 to 1.21.0. Tried few revisions to find breaking commit:
1.19.0 - Ok
1.20.0 - Ok
1.20.0-39-gf8927a94 - Ok
1.20.0-58-g6e84c149 - Ok
1.20.0-64-gcd033c89 - Ok
1.20.0-66-g05a535cf Update zigbee-herdsman to 0.13.123 - Ok
1.20.0-67-ffea1ed9 Update zigbee2mqtt-frontend to 0.5.16 - Ok
1.20.0-68-g6068cd85 Update zigbee-herdsman to 0.13.124 - FAILS

console log:

```
Zigbee2MQTT:info  2021-08-12 01:17:53: Starting Zigbee2MQTT version 1.20.0-dev (commit #6068cd85)
Zigbee2MQTT:info  2021-08-12 01:17:53: Starting zigbee-herdsman (0.13.124)
Zigbee2MQTT:error 2021-08-12 01:18:14: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-08-12 01:18:14: Failed to start zigbee
Zigbee2MQTT:error 2021-08-12 01:18:14: Check https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start for possible solutions
Zigbee2MQTT:error 2021-08-12 01:18:14: Exiting...
Zigbee2MQTT:error 2021-08-12 01:18:15: Error: SRSP - SYS - version after 6000ms
```

Log for previous revisions:
```
Zigbee2MQTT:info  2021-08-12 01:34:22: Starting Zigbee2MQTT version 1.20.0-dev (commit #ffea1ed9)
Zigbee2MQTT:info  2021-08-12 01:34:22: Starting zigbee-herdsman (0.13.123)
Zigbee2MQTT:info  2021-08-12 01:34:38: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2021-08-12 01:34:38: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":2,"minorrel":0,"product":0,"revision":"","transportrev":2},"type":"zStack12"}'
Zigbee2MQTT:info  2021-08-12 01:34:38: Currently 32 devices are joined:
```

Koenkk, 2021-08-12:
The only commit in 0.13.124 is https://github.com/Koenkk/zigbee-herdsman/commit/779335fd39881ab8361b7fd9ad877be5342d8354

- Can you confirm that after failing to start on 0.13.124, reverting back to 0.13.123 makes z2m start correctly?
  - If yes, please provide the herdsman debug log of the failing start on 0.13.124.

See https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

islepnev, 2021-08-12:
Checked out 0.13.123, starts correctly.
```
root@wb:~/zigbee2mqtt# git describe
1.20.0-67-gffea1ed9
root@wb:~/zigbee2mqtt# grep zigbee-herdsman package.json 
    "zigbee-herdsman": "0.13.123",
    "zigbee-herdsman-converters": "14.0.205",
```

Terminated, cloned git repo, checked out 0.13.124:
```
root@wb:~/zigbee2mqtt-bak# git describe
1.20.0-68-g6068cd85
root@wb:~/zigbee2mqtt-bak# grep zigbee-herdsman package.json
    "zigbee-herdsman": "0.13.124",
    "zigbee-herdsman-converters": "14.0.205",
```
copied data folder and tried to start:
```
root@wb:~/zigbee2mqtt-bak# npm ci && npm start | tee log.txt
...skip...

Zigbee2MQTT:debug 2021-08-12 18:08:10: Loaded state from file /mnt/data/root/zigbee2mqtt-bak/data/state.json
Zigbee2MQTT:info  2021-08-12 18:08:10: Logging to console and directory: '/mnt/data/root/zigbee2mqtt-bak/data/log/2021-08-12.18-07-56' filename: log.txt
Zigbee2MQTT:info  2021-08-12 18:08:11: Starting Zigbee2MQTT version 1.20.0-dev (commit #6068cd85)
Zigbee2MQTT:info  2021-08-12 18:08:11: Starting zigbee-herdsman (0.13.124)
Zigbee2MQTT:debug 2021-08-12 18:08:11: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/mnt/data/root/zigbee2mqtt-bak/data/coordinator_backup.json","databaseBackupPath":"/mnt/data/root/zigbee2mqtt-bak/data/database.db.backup","databasePath":"/mnt/data/root/zigbee2mqtt-bak/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyMOD3","rtscts":false}}'
Zigbee2MQTT:error 2021-08-12 18:08:33: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-08-12 18:08:33: Failed to start zigbee
Zigbee2MQTT:error 2021-08-12 18:08:33: Check https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start for possible solutions
Zigbee2MQTT:error 2021-08-12 18:08:33: Exiting...
Zigbee2MQTT:error 2021-08-12 18:08:33: Error: SRSP - SYS - version after 6000ms
    at Timeout._onTimeout (/mnt/data/root/zigbee2mqtt-bak/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.20.0-dev start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.20.0-dev start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2021-08-12T15_08_33_419Z-debug.log
```

Koenkk, 2021-08-12:
Thanks, can you provide the herdsman debug logging?  
See https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

@castorw could you check this?

islepnev, 2021-08-12:
```
root@wb:~/zigbee2mqtt-bak# DEBUG=zigbee-herdsman* npm start | tee log.txt

> zigbee2mqtt@1.20.0-dev start /mnt/data/root/zigbee2mqtt-bak
> node index.js

Zigbee2MQTT:debug 2021-08-12 18:25:23: Loaded state from file /mnt/data/root/zigbee2mqtt-bak/data/state.json
Zigbee2MQTT:info  2021-08-12 18:25:23: Logging to console and directory: '/mnt/data/root/zigbee2mqtt-bak/data/log/2021-08-12.18-25-10' filename: log.txt
Zigbee2MQTT:info  2021-08-12 18:25:24: Starting Zigbee2MQTT version 1.20.0-dev (commit #6068cd85)
Zigbee2MQTT:info  2021-08-12 18:25:24: Starting zigbee-herdsman (0.13.124)
Zigbee2MQTT:debug 2021-08-12 18:25:24: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/mnt/data/root/zigbee2mqtt-bak/data/coordinator_backup.json","databaseBackupPath":"/mnt/data/root/zigbee2mqtt-bak/data/database.db.backup","databasePath":"/mnt/data/root/zigbee2mqtt-bak/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyMOD3","rtscts":true}}'
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[185,174,223,88,254,27,89,64,155,142,203,102,85,0,51,15],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"rtscts":true,"path":"/dev/ttyMOD3"},"databasePath":"/mnt/data/root/zigbee2mqtt-bak/data/database.db","databaseBackupPath":"/mnt/data/root/zigbee2mqtt-bak/data/database.db.backup","backupPath":"/mnt/data/root/zigbee2mqtt-bak/data/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}' +0ms
  zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyMOD3 and {"baudRate":115200,"rtscts":true,"autoOpen":false} +0ms
  zigbee-herdsman:adapter:zStack:znp:log Serialport opened +58ms
  zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload +526ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239] +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +0ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +1s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,2,97,1,121,1,26] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,2,97,1,121,1,26] +10ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 2 - 3 - 1 - 1 - [121,1] - 26 +9ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - ping - {"capabilities":377} +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +55ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - version - {} +124ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,2,35] +109ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254] +29ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254] +3ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [5,97,2,2,0,2,6,3,99] +10ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,5,97,2,2,0,2,6,3,99] +3ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 5 - 3 - 1 - 2 - [2,0,2,6,3] - 99 +2ms
  zigbee-herdsman:adapter:zStack:znp:error Error while parsing to ZpiObject 'RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 1. Received 5
  zigbee-herdsman:adapter:zStack:znp:error     at boundsError (internal/buffer.js:82:9)
  zigbee-herdsman:adapter:zStack:znp:error     at Buffer.readUInt32LE (internal/buffer.js:216:5)
  zigbee-herdsman:adapter:zStack:znp:error     at BuffaloZnp.readUInt32 (/mnt/data/root/zigbee2mqtt-bak/node_modules/zigbee-herdsman/src/buffalo/buffalo.ts:101:35)
  zigbee-herdsman:adapter:zStack:znp:error     at BuffaloZnp.read (/mnt/data/root/zigbee2mqtt-bak/node_modules/zigbee-herdsman/src/buffalo/buffalo.ts:284:25)
  zigbee-herdsman:adapter:zStack:znp:error     at BuffaloZnp.read (/mnt/data/root/zigbee2mqtt-bak/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/buffaloZnp.ts:161:26)
  zigbee-herdsman:adapter:zStack:znp:error     at Function.readParameters (/mnt/data/root/zigbee2mqtt-bak/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/zpiObject.ts:107:46)
  zigbee-herdsman:adapter:zStack:znp:error     at Function.fromUnpiFrame (/mnt/data/root/zigbee2mqtt-bak/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/zpiObject.ts:73:30)
  zigbee-herdsman:adapter:zStack:znp:error     at Znp.onUnpiParsed (/mnt/data/root/zigbee2mqtt-bak/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:103:38)
  zigbee-herdsman:adapter:zStack:znp:error     at Parser.emit (events.js:314:20)
  zigbee-herdsman:adapter:zStack:znp:error     at Parser.parseNext (/mnt/data/root/zigbee2mqtt-bak/node_modules/zigbee-herdsman/src/adapter/z-stack/unpi/parser.ts:45:26)' +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +965ms
  zigbee-herdsman:adapter:zStack:adapter Failed to get zStack version, assuming 1.2 +0ms
  zigbee-herdsman:adapter:zStack:adapter Adapter concurrent: 2 +4ms
  zigbee-herdsman:adapter:zStack:adapter Detected znp version 'zStack12' ({"transportrev":2,"product":0,"majorrel":2,"minorrel":0,"maintrel":0,"revision":""}) +5ms
  zigbee-herdsman:adapter:zStack:startup beginning znp startup +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - getExtAddr - {} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,4,37] +6s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254] +5s
[... cut ...]
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.20.0-dev start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.20.0-dev start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2021-08-12T15_25_47_811Z-debug.log
root@wb:~/zigbee2mqtt-bak# 

```

umairyounus, 2021-08-12:
I'm having a similar issue, after upgrading to 1.21.0, zigbee-herdsman is not starting. I've downgraded but problem is presistent. Any suggestion I can try?

```
Zigbee2MQTT:debug 2021-08-12 20:20:18: Loaded state from file /app/data/state.json
Zigbee2MQTT:info  2021-08-12 20:20:18: Logging to console and directory: '/app/data/log/2021-08-12.20-20-16' filename: log.txt
Zigbee2MQTT:debug 2021-08-12 20:20:18: Removing old log directory '/app/data/log/2021-08-10.12-33-37'
Zigbee2MQTT:info  2021-08-12 20:20:18: Starting Zigbee2MQTT version 1.20.0 (commit #41b67fd)
Zigbee2MQTT:info  2021-08-12 20:20:18: Starting zigbee-herdsman (0.13.117)
Zigbee2MQTT:debug 2021-08-12 20:20:18: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[25],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyACM1"}}'
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[2,100,79,131,123,190,58,7,173,24,64,26,126,130,18,134],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[25]},"serialPort":{"path":"/dev/ttyACM1"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}' +0ms
  zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyACM1 and {"baudRate":115200,"rtscts":false,"autoOpen":false} +0ms
  zigbee-herdsman:adapter:zStack:znp:log Serialport opened +7ms
  zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload +460ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239] +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +0ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +1s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +6s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +6s
Zigbee2MQTT:error 2021-08-12 20:20:38: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-08-12 20:20:38: Failed to start zigbee
Zigbee2MQTT:error 2021-08-12 20:20:38: Check https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start for possible solutions
Zigbee2MQTT:error 2021-08-12 20:20:38: Exiting...
Zigbee2MQTT:error 2021-08-12 20:20:38: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:93:31)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.20.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.20.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2021-08-12T20_20_38_983Z-debug.log

Koenkk, 2021-08-13:
@islepnev `Error while parsing to ZpiObject 'RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 1. Received 5` could mean a possible corruption. Can you check if reflashing your stick fixes the problem?

@umairyounus if downgrading doesn't fix it is unrelated to this issue, see https://www.zigbee2mqtt.io/information/FAQ.html#error-srsp---sys---ping-after-6000ms

islepnev, 2021-08-13:
> @islepnev `Error while parsing to ZpiObject 'RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 1. Received 5` could mean a possible corruption. Can you check if reflashing your stick fixes the problem?
I wish I could, no cc-debugger at the moment, sorry.

closed 2021-09-20 by github-actions[bot]