Koenkk/zigbee-herdsman #408: "Unhandled promise rejection" on "No network route"

kirovilya, 2021-08-14:
A strange thing happens with one of the devices - when it is removed from the network, unexpectedly another message is sent to it and it does not reach the device - the service drops:
`Data request failed with error: 'No network route' (205)`

During deletion, I press the button - so that the remote control wakes up. I think it is trying to send a DefaultResponse to this device in response to a click and cannot do it.
Detailed log from iobroker.zigbee:
```
2021-08-14 11:09:43.819 deleteDevice message: {"id":"zigbee.0.b4e3f9fffe357966","force":false}
2021-08-14 11:09:43.819 deleteDevice sysid: 0xb4e3f9fffe357966
2021-08-14 11:09:43.819 zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - mgmtLeaveReq - {"dstaddr":57137,"deviceaddress":"0xb4e3f9fffe357966","removechildrenRejoin":0}
2021-08-14 11:09:43.820 zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,11,37,52,49,223,102,121,53,254,255,249,227,180,0,113]
2021-08-14 11:09:43.821 zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,52,0,80]
2021-08-14 11:09:43.822 zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,52,0,80]
2021-08-14 11:09:43.822 zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 52 - [0] - 80
2021-08-14 11:09:43.822 zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - mgmtLeaveReq - {"status":0}
2021-08-14 11:09:43.822 zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-08-14 11:09:49.700 zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,24,68,129,0,0,6,0,49,223,1,1,0,94,0,76,22,13,0,0,4,1,56,253,0,49,223,29,15]
2021-08-14 11:09:49.700 zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,24,68,129,0,0,6,0,49,223,1,1,0,94,0,76,22,13,0,0,4,1,56,253,0,49,223,29,15]
2021-08-14 11:09:49.701 zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 24 - 2 - 4 - 129 - [0,0,6,0,49,223,1,1,0,94,0,76,22,13,0,0,4,1,56,253,0,49,223,29] - 15
2021-08-14 11:09:49.702 zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":57137,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":94,"securityuse":0,"timestamp":857676,"transseqnumber":0,"len":4,"data":{"type":"Buffer","data":[1,56,253,0]}}
2021-08-14 11:09:49.704 zigbee-herdsman:controller:log Received 'raw' data '{"clusterID":6,"data":{"type":"Buffer","data":[1,56,253,0]},"address":57137,"endpoint":1,"linkquality":94,"groupID":0,"wasBroadcast":false}'
2021-08-14 11:09:49.705 handleMessage. {"type":"raw","device":{"ID":2,"_type":"EndDevice","_ieeeAddr":"0xb4e3f9fffe357966","_networkAddress":57137,"_manufacturerID":4098,"_endpoints":[{"ID":1,"profileID":260,"deviceID":0,"inputClusters":[0,1,6],"outputClusters":[25,10],"deviceNetworkAddress":57137,"deviceIeeeAddress":"0xb4e3f9fffe357966","clusters":{"genBasic":{"attributes":{"modelId":"TS0042","manufacturerName":"_TZ3000_t8hzpgnd","powerSource":3,"zclVersion":3,"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"_binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b00192e4303","endpointID":1}],"_configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}},{"ID":2,"profileID":260,"deviceID":0,"inputClusters":[1,6],"outputClusters":[],"deviceNetworkAddress":57137,"deviceIeeeAddress":"0xb4e3f9fffe357966","clusters":{},"_binds":[],"_configuredReportings":[],"meta":{}}],"_manufacturerName":"_TZ3000_t8hzpgnd","_powerSource":"Battery","_modelID":"TS0042","_applicationVersion":66,"_stackVersion":0,"_zclVersion":3,"_hardwareVersion":1,"_dateCode":"","_interviewCompleted":true,"_interviewing":false,"_skipDefaultResponse":false,"meta":{"configured":1},"_lastSeen":1628928589704,"_linkquality":94},"endpoint":"[Circular]","data":{"type":"Buffer","data":[1,56,253,0]},"linkquality":94,"groupID":0,"cluster":"genOnOff","meta":{}}
2021-08-14 11:09:49.705 zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-08-14 11:09:49.706 Received Zigbee message from '0xb4e3f9fffe357966', type 'raw', cluster 'genOnOff', data '{"type":"Buffer","data":[1,56,253,0]}' from endpoint 1 with groupID 0
2021-08-14 11:09:49.706 Type raw device {"type":"device","device":{"ID":2,"_type":"EndDevice","_ieeeAddr":"0xb4e3f9fffe357966","_networkAddress":57137,"_manufacturerID":4098,"_endpoints":[{"ID":1,"profileID":260,"deviceID":0,"inputClusters":[0,1,6],"outputClusters":[25,10],"deviceNetworkAddress":57137,"deviceIeeeAddress":"0xb4e3f9fffe357966","clusters":{"genBasic":{"attributes":{"modelId":"TS0042","manufacturerName":"_TZ3000_t8hzpgnd","powerSource":3,"zclVersion":3,"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"_binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b00192e4303","endpointID":1}],"_configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}},{"ID":2,"profileID":260,"deviceID":0,"inputClusters":[1,6],"outputClusters":[],"deviceNetworkAddress":57137,"deviceIeeeAddress":"0xb4e3f9fffe357966","clusters":{},"_binds":[],"_configuredReportings":[],"meta":{}}],"_manufacturerName":"_TZ3000_t8hzpgnd","_powerSource":"Battery","_modelID":"TS0042","_applicationVersion":66,"_stackVersion":0,"_zclVersion":3,"_hardwareVersion":1,"_dateCode":"","_interviewCompleted":true,"_interviewing":false,"_skipDefaultResponse":false,"meta":{"configured":1},"_lastSeen":1628928589704,"_linkquality":94},"mapped":{"zigbeeModel":["TS0042"],"model":"TS0042","vendor":"TuYa","description":"Wireless switch with 2 buttons","whiteLabel":[{"vendor":"Smart9","model":"S9TSZGB"},{"vendor":"Lonsonho","model":"TS0042"}],"exposes":[{"type":"numeric","name":"battery","property":"battery","access":1,"unit":"%","description":"Remaining battery in %","value_min":0,"value_max":100},{"type":"enum","name":"action","property":"action","access":1,"values":["1_single","1_double","1_hold","2_single","2_double","2_hold"],"description":"Triggered action (e.g. a button click)"},{"type":"numeric","name":"linkquality","property":"linkquality","access":1,"unit":"lqi","description":"Link quality (signal strength)","value_min":0,"value_max":255}],"fromZigbee":[{"cluster":"genOnOff","type":"raw"},{"cluster":"genPowerCfg","type":["attributeReport","readResponse"]}],"toZigbee":[]},"name":"0xb4e3f9fffe357966"} incoming event: {"type":"raw","device":{"ID":2,"_type":"EndDevice","_ieeeAddr":"0xb4e3f9fffe357966","_networkAddress":57137,"_manufacturerID":4098,"_endpoints":[{"ID":1,"profileID":260,"deviceID":0,"inputClusters":[0,1,6],"outputClusters":[25,10],"deviceNetworkAddress":57137,"deviceIeeeAddress":"0xb4e3f9fffe357966","clusters":{"genBasic":{"attributes":{"modelId":"TS0042","manufacturerName":"_TZ3000_t8hzpgnd","powerSource":3,"zclVersion":3,"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"_binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b00192e4303","endpointID":1}],"_configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}},{"ID":2,"profileID":260,"deviceID":0,"inputClusters":[1,6],"outputClusters":[],"deviceNetworkAddress":57137,"deviceIeeeAddress":"0xb4e3f9fffe357966","clusters":{},"_binds":[],"_configuredReportings":[],"meta":{}}],"_manufacturerName":"_TZ3000_t8hzpgnd","_powerSource":"Battery","_modelID":"TS0042","_applicationVersion":66,"_stackVersion":0,"_zclVersion":3,"_hardwareVersion":1,"_dateCode":"","_interviewCompleted":true,"_interviewing":false,"_skipDefaultResponse":false,"meta":{"configured":1},"_lastSeen":1628928589704,"_linkquality":94},"endpoint":"[Circular]","data":{"type":"Buffer","data":[1,56,253,0]},"linkquality":94,"groupID":0,"cluster":"genOnOff","meta":{}}
2021-08-14 11:09:49.727 zigbee-herdsman:controller:endpoint DefaultResponse 0xb4e3f9fffe357966/1 6(253, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2021-08-14 11:09:49.728 Publish {"action":"1_single"} to "b4e3f9fffe357966"
2021-08-14 11:09:49.807 zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,180,49,223,0,28]
2021-08-14 11:09:49.808 zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,180,49,223,0,28]
[... cut ...] its":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))
2021-08-14 11:10:10.083 cleaned everything up...
2021-08-14 11:10:10.086 Zigbee: disabling joining new devices.
2021-08-14 11:10:10.332 terminating
```

Any idea why this is happening? It seems that this has never happened before.

Another user has a similar situation when he simply presses a button on the device, even without removing the device.

Koenkk, 2021-08-15:
What kind of button is this? You can try disabling all default responses with https://github.com/Koenkk/zigbee-herdsman-converters/blob/150735c361e3c3314643299ddbf8aa5a13712ed6/devices/ewelink.js#L92

kirovilya, 2021-08-15:
it is Loratap remote with 2 buttons https://aliexpress.ru/item/1005002846674119.html identified like 'TS0042'.
I think when receiving the leaveInd event, we should clear all the waiters related to this device

I will try to add a handler as you suggest

kirovilya, 2021-08-15:
not helped. these events do not come to device.onZclData
```
2021-08-15 21:45:22.179 zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,24,68,129,0,0,6,0,26,119,1,1,0,89,0,107,57,86,0,0,4,1,14,253,0,26,119,29,109]
2021-08-15 21:45:22.179 zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,24,68,129,0,0,6,0,26,119,1,1,0,89,0,107,57,86,0,0,4,1,14,253,0,26,119,29,109]
2021-08-15 21:45:22.179 zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 24 - 2 - 4 - 129 - [0,0,6,0,26,119,1,1,0,89,0,107,57,86,0,0,4,1,14,253,0,26,119,29] - 109
2021-08-15 21:45:22.179 zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":30490,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":89,"securityuse":0,"timestamp":5650795,"transseqnumber":0,"len":4,"data":{"type":"Buffer","data":[1,14,253,0]}}
2021-08-15 21:45:22.180 zigbee-herdsman:controller:log Received 'raw' data '{"clusterID":6,"data":{"type":"Buffer","data":[1,14,253,0]},"address":30490,"endpoint":1,"linkquality":89,"groupID":0,"wasBroadcast":false}'
2021-08-15 21:45:22.180 handleMessage. {"type":"raw","device":{"ID":2,"_type":"EndDevice","_ieeeAddr":"0xb4e3f9fffe357966","_networkAddress":30490,"_manufacturerID":4098,"_endpoints":[{"ID":1,"profileID":260,"deviceID":0,"inputClusters":[0,1,6],"outputClusters":[25,10],"deviceNetworkAddress":30490,"deviceIeeeAddress":"0xb4e3f9fffe357966","clusters":{"genBasic":{"attributes":{"modelId":"TS0042","manufacturerName":"_TZ3000_t8hzpgnd","powerSource":3,"zclVersion":3,"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"_binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b00192e4303","endpointID":1}],"_configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}},{"ID":2,"profileID":260,"deviceID":0,"inputClusters":[1,6],"outputClusters":[],"deviceNetworkAddress":30490,"deviceIeeeAddress":"0xb4e3f9fffe357966","clusters":{},"_binds":[],"_configuredReportings":[],"meta":{}}],"_manufacturerName":"_TZ3000_t8hzpgnd","_powerSource":"Battery","_modelID":"TS0042","_applicationVersion":66,"_stackVersion":0,"_zclVersion":3,"_hardwareVersion":1,"_dateCode":"","_interviewCompleted":true,"_interviewing":false,"_skipDefaultResponse":true,"meta":{"configured":1},"_lastSeen":1629053122180,"_linkquality":89},"endpoint":"[Circular]","data":{"type":"Buffer","data":[1,14,253,0]},"linkquality":89,"groupID":0,"cluster":"genOnOff","meta":{}}
2021-08-15 21:45:22.181 zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-08-15 21:45:22.181 Received Zigbee message from '0xb4e3f9fffe357966', type 'raw', cluster 'genOnOff', data '{"type":"Buffer","data":[1,14,253,0]}' from endpoint 1 with groupID 0
2021-08-15 21:45:22.181 Type raw device {"type":"device","device":{"ID":2,"_type":"EndDevice","_ieeeAddr":"0xb4e3f9fffe357966","_networkAddress":30490,"_manufacturerID":4098,"_endpoints":[{"ID":1,"profileID":260,"deviceID":0,"inputClusters":[0,1,6],"outputClusters":[25,10],"deviceNetworkAddress":30490,"deviceIeeeAddress":"0xb4e3f9fffe357966","clusters":{"genBasic":{"attributes":{"modelId":"TS0042","manufacturerName":"_TZ3000_t8hzpgnd","powerSource":3,"zclVersion":3,"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"_binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b00192e4303","endpointID":1}],"_configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}},{"ID":2,"profileID":260,"deviceID":0,"inputClusters":[1,6],"outputClusters":[],"deviceNetworkAddress":30490,"deviceIeeeAddress":"0xb4e3f9fffe357966","clusters":{},"_binds":[],"_configuredReportings":[],"meta":{}}],"_manufacturerName":"_TZ3000_t8hzpgnd","_powerSource":"Battery","_modelID":"TS0042","_applicationVersion":66,"_stackVersion":0,"_zclVersion":3,"_hardwareVersion":1,"_dateCode":"","_interviewCompleted":true,"_interviewing":false,"_skipDefaultResponse":true,"meta":{"configured":1},"_lastSeen":1629053122180,"_linkquality":89},"mapped":{"zigbeeModel":["TS0042"],"model":"TS0042","vendor":"TuYa","description":"Wireless switch with 2 buttons","whiteLabel":[{"vendor":"Smart9","model":"S9TSZGB"},{"vendor":"Lonsonho","model":"TS0042"}],"exposes":[{"type":"numeric","name":"battery","property":"battery","access":1,"unit":"%","description":"Remaining battery in %","value_min":0,"value_max":100},{"type":"enum","name":"action","property":"action","access":1,"values":["1_single","1_double","1_hold","2_single","2_double","2_hold"],"description":"Triggered action (e.g. a button click)"},{"type":"numeric","name":"linkquality","property":"linkquality","access":1,"unit":"lqi","description":"Link quality (signal strength)","value_min":0,"value_max":255}],"fromZigbee":[{"cluster":"genOnOff","type":"raw"},{"cluster":"genPowerCfg","type":["attributeReport","readResponse"]}],"toZigbee":[]},"name":"0xb4e3f9fffe357966"} incoming event: {"type":"raw","device":{"ID":2,"_type":"EndDevice","_ieeeAddr":"0xb4e3f9fffe357966","_networkAddress":30490,"_manufacturerID":4098,"_endpoints":[{"ID":1,"profileID":260,"deviceID":0,"inputClusters":[0,1,6],"outputClusters":[25,10],"deviceNetworkAddress":30490,"deviceIeeeAddress":"0xb4e3f9fffe357966","clusters":{"genBasic":{"attributes":{"modelId":"TS0042","manufacturerName":"_TZ3000_t8hzpgnd","powerSource":3,"zclVersion":3, [... cut ...]
2021-08-15 21:45:52.932 Exception-Code: 205: DefaultResponse 0xb4e3f9fffe357966/1 6(253, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))
2021-08-15 21:45:52.938 cleaned everything up...
2021-08-15 21:45:52.940 Zigbee: disabling joining new devices.
2021-08-15 21:45:53.207 terminating
```

kirovilya, 2021-08-15:
it is not 'zcl' - it is 'raw' type

arteck, 2021-08-16:
same error here with .. TS0043
[image]

MattWestb, 2021-08-16:
@kirovilya Then the host have sending one leave command to one device is cant getting and also not answering any command to / from the host system (i think have seeing some device is sending one "left network" after receiving the leave command then then resetting its self to factory new) then its **not** one part the network and dont have one valid network key or valid trustcenter security frame counter so the device looks doing all OK but the host system is making somthing very strange trying talking with one device that it have kicking out of the system = shall not being possible to do (if the host system is making all right).

Koenkk, 2021-08-16:
Does https://github.com/Koenkk/zigbee-herdsman-converters/commit/d5673e709b9597088b501843a2e82b9275459321  fix it?

kirovilya, 2021-08-16:
@Koenkk better - not crashing. thank you
```
2021-08-16 19:25:22.036 deleteDevice message: {"id":"zigbee.0.b4e3f9fffe357966","force":false}
2021-08-16 19:25:22.036 deleteDevice sysid: 0xb4e3f9fffe357966
2021-08-16 19:25:22.036 zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - mgmtLeaveReq - {"dstaddr":13370,"deviceaddress":"0xb4e3f9fffe357966","removechildrenRejoin":0}
2021-08-16 19:25:22.036 zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,11,37,52,58,52,102,121,53,254,255,249,227,180,0,145]
2021-08-16 19:25:22.038 zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,52,0,80]
2021-08-16 19:25:22.038 zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,52,0,80]
2021-08-16 19:25:22.039 zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 52 - [0] - 80
2021-08-16 19:25:22.039 zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - mgmtLeaveReq - {"status":0}
2021-08-16 19:25:22.039 zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-08-16 19:25:24.542 zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,24,68,129,0,0,6,0,58,52,1,1,0,102,0,183,150,247,0,0,4,1,34,253,0,58,52,29,172]
2021-08-16 19:25:24.543 zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,24,68,129,0,0,6,0,58,52,1,1,0,102,0,183,150,247,0,0,4,1,34,253,0,58,52,29,172]
2021-08-16 19:25:24.543 zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 24 - 2 - 4 - 129 - [0,0,6,0,58,52,1,1,0,102,0,183,150,247,0,0,4,1,34,253,0,58,52,29] - 172
2021-08-16 19:25:24.543 zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":13370,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":102,"securityuse":0,"timestamp":16225975,"transseqnumber":0,"len":4,"data":{"type":"Buffer","data":[1,34,253,0]}}
2021-08-16 19:25:24.543 zigbee-herdsman:controller:log Received 'raw' data '{"clusterID":6,"data":{"type":"Buffer","data":[1,34,253,0]},"address":13370,"endpoint":1,"linkquality":102,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
2021-08-16 19:25:24.544 handleMessage. {"type":"raw","device":{"ID":2,"_type":"EndDevice","_ieeeAddr":"0xb4e3f9fffe357966","_networkAddress":13370,"_manufacturerID":4098,"_endpoints":[{"ID":1,"profileID":260,"deviceID":0,"inputClusters":[0,1,6],"outputClusters":[25,10],"deviceNetworkAddress":13370,"deviceIeeeAddress":"0xb4e3f9fffe357966","clusters":{"genBasic":{"attributes":{"modelId":"TS0042","manufacturerName":"_TZ3000_t8hzpgnd","powerSource":3,"zclVersion":3,"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":188}}},"_binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b00192e4303","endpointID":1}],"_configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{},"pendingRequests":[]},{"ID":2,"profileID":260,"deviceID":0,"inputClusters":[1,6],"outputClusters":[],"deviceNetworkAddress":13370,"deviceIeeeAddress":"0xb4e3f9fffe357966","clusters":{},"_binds":[],"_configuredReportings":[],"meta":{},"pendingRequests":[]}],"_manufacturerName":"_TZ3000_t8hzpgnd","_powerSource":"Battery","_modelID":"TS0042","_applicationVersion":66,"_stackVersion":0,"_zclVersion":3,"_hardwareVersion":1,"_dateCode":"","_interviewCompleted":true,"_interviewing":false,"_skipDefaultResponse":false,"_skipTimeResponse":false,"meta":{"configured":1},"_lastSeen":1629131124544,"_linkquality":102,"_deleted":false},"endpoint":"[Circular]","data":{"type":"Buffer","data":[1,34,253,0]},"linkquality":102,"groupID":0,"cluster":"genOnOff","meta":{}}
2021-08-16 19:25:24.544 zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-08-16 19:25:24.544 Received Zigbee message from '0xb4e3f9fffe357966', type 'raw', cluster 'genOnOff', data '{"type":"Buffer","data":[1,34,253,0]}' from endpoint 1 with groupID 0
2021-08-16 19:25:24.544 Type raw device {"type":"device","device":{"ID":2,"_type":"EndDevice","_ieeeAddr":"0xb4e3f9fffe357966","_networkAddress":13370,"_manufacturerID":4098,"_endpoints":[{"ID":1,"profileID":260,"deviceID":0,"inputClusters":[0,1,6],"outputClusters":[25,10],"deviceNetworkAddress":13370,"deviceIeeeAddress":"0xb4e3f9fffe357966","clusters":{"genBasic":{"attributes":{"modelId":"TS0042","manufacturerName":"_TZ3000_t8hzpgnd","powerSource":3,"zclVersion":3,"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":188}}},"_binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b00192e4303","endpointID":1}],"_configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{},"pendingRequests":[]},{"ID":2,"profileID":260,"deviceID":0,"inputClusters":[1,6],"outputClusters":[],"deviceNetworkAddress":13370,"deviceIeeeAddress":"0xb4e3f9fffe357966","clusters":{},"_binds":[],"_configuredReportings":[],"meta":{},"pendingRequests":[]}],"_manufacturerName":"_TZ3000_t8hzpgnd","_powerSource":"Battery","_modelID":"TS0042","_applicationVersion":66,"_stackVersion":0,"_zclVersion":3,"_hardwareVersion":1,"_dateCode":"","_interviewCompleted":true,"_interviewing":false,"_skipDefaultResponse":false,"_skipTimeResponse":false,"meta":{"configured":1},"_lastSeen":1629131124544,"_linkquality":102,"_deleted":false},"mapped":{"zigbeeModel":["TS0042"],"model":"TS0042","vendor":"TuYa","description":"Wireless switch with 2 buttons","whiteLabel":[{"vendor":"Smart9","model":"S9TSZGB"},{"vendor":"Lonsonho","model":" [... cut ...]
2021-08-16 19:25:44.920 zigbee-herdsman:adapter:zStack:adapter Data confirm error (%s:%d,%d,%d)
2021-08-16 19:25:44.920 zigbee-herdsman:controller:endpoint DefaultResponse 0xb4e3f9fffe357966/1 6(253, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))
```

need to add the ability to disable the DefaultResponse there

arteck, 2021-08-16:
big THX @Koenkk

closed 2021-08-16 by kirovilya