Koenkk/zigbee-herdsman #420: Using "Install Code" to pair devices

kirovilya, 2021-09-16:
First encountered a device that has an Install Code - rotary switch Aqara H1.
We need to learn how to pair a such devices.
As I understand it, when starting pairing, we need to specify the scanned key "Install Code".

[image]
[image]
[image]

MattWestb, 2021-09-16:
Look in Zigpy / ZHA github they have implanting it for some coordinator (EFR32 and perhaps TI) but i have not testing it then i dont have devices that using it.

kirovilya, 2021-09-17:
found issue for z-stack https://github.com/zigpy/zigpy-znp/issues/29 and ezsp https://github.com/zigpy/bellows/issues/292
I continue to looking

MattWestb, 2021-09-17:
Bellows CLI can joining with install code and its starting here:
https://github.com/zigpy/bellows/blob/a15b54fc2275a8bb603383062b5d8e8fe562e18e/bellows/cli/application.py#L73

If i remember right is adding the install code as one Trust center Link Key and is being updated if the Zigbee 3 device like updating the TC-Link key.

I think its one good start for looking at the code wot they is doing.

sjorge, 2021-10-17:
Most hue/innr/tint/osram bulbs have them too, the format is always different sadly, but it usually include a key and other data.

https://github.com/Koenkk/zigbee2mqtt/discussions/8498 Has a few of them

closed 2021-11-24 by github-actions[bot]