Koenkk/zigbee-herdsman #427: Unhandled promise rejection in znp

adriaanvanrossum, 2021-09-27:
It seems to happen in this file: https://github.com/Koenkk/zigbee-herdsman/blob/master/src/adapter/z-stack/znp/znp.ts

The result is that there is no way to use zigbee devices after this error. It just stops working. After a restart it all works again.

This is the console output. Sorry for the mess (I'm on my phone):

```
Zigbee2MQTT:info 2021-09-27 23:34:30: MQTT publish: topic
"zigbeezmat
t/agara humidity _sensor_bathroom', payload
"{"battery": 100, "humidity":
65.63, "linkquality": 110, "pressure": 1012, "temperature":21.65, "voltage":
3025}"
(node:7) UnhandledPromiseRejectionWarning: Error: SREQ
"_-> ZD0 - mgmt
PermitJoinReg - ("addrmode": 15, "dstaddr" :65532, "duration":254, "tcsigni
ficance":0}" failed with status
'(0x10: MEM_ERROR)' (expected
"(0x00:
SUCCESS) ')
at np.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter
/z-stack/znp/znp.js:291:27)
at Generator.next (<anonymous>)
at fulfilled (/app/node_modules/zigbee-herdsman/dist/adapter/z-sta
ck/znp/znp.js:24:58)
(node:7) UnhandledPromiseRejectionWarning: Unhandled promise rejection
• This error originated either by throwing inside of an async function
without a catch block, or by rejecting a promise which was not handle
d with .catch). To terminate the node process on unhandled promise re
jection, use the CLI flag '-unhandled-rejections=strict' (see https:/
/nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id
: 418)
(node:7) UnhandledPromiseRejectionWarning: Error: SREQ '--> ZDO - mgmt
PermitJoinReg - ("addrmode": 15, "dstaddr" :65532, "duration":254, "tcsigni
ficance":0}' failed with status
SUCCESS)')
'(0x10: MEM _ERROR)' (expected
"(0x00:
at np.<anonymous> (/app/node _modules/zigbee-herdsman/dist/adapter
/z-stack/znp/znp.js:291:27)
at Generator.next (<anonymous>)
at fulfilled (/app/node_modules/zigbee-herdsman/dist/adapter/z-sta
ck/znp/znp.js:24:58)
(node: 7) UnhandledPromiseRejectionWarning: Unhandled promise rejection
• This error originated either by throwing inside of an async function
without a catch block, or by rejecting a promise which was not handle
d with .catch(). To terminate the node process on unhandled promise re
jection, use the CLI flag
-unhandled-rejections=strict' (see https:/
/nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 419)
```

Koenkk, 2021-09-28:
What coordinator hardware/firmware version are you using?

adriaanvanrossum, 2021-10-03:
I'm using [the zigbee cc2531 usb stick dongle](https://smartgateways.nl/product/zigbee-cc2531-usb-stick-dongle-met-antenne-voor-zigbee2mqtt-antenne/) with firmware `Z-Stack_Home_1.2`. It's not extended by cable. Could that be a problem. After restarts it doesn't work anymore as well.

Koenkk, 2021-10-03:
Looks like it got corrupted (common for the CC2531), try the source routing fw: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing or update to a more powerful adapter https://www.zigbee2mqtt.io/getting_started/what_do_i_need.html#supported-zigbee-adapter