Koenkk/zigbee-herdsman #450: Request ordering during retry

kennylevinsen, 2021-11-08:
Right now, the adapter layer is responsible for performing retries at will. If two requests are ongoing towards an endpoint, they could end up sent out of order - if the requests were "off", "on" or a series of setpoint assignments as a UI is tapped in rapid succession, this will lead to devices being in unexpected states from a user perspective.

It might make sense to let the endpoint layer be responsible for issuing retries, so that per-endpoint request streams have guaranteed ordering, although we will still have to cater to adapter-specific error mitigation attempts in between retries (e.g. the assoc remove logic in ztack).

It seems that we only handle retry ourselves for zstack. I do not know what retry mechanisms or ordering guarantees the other adapters might provide.

Koenkk, 2021-11-08:
@kennylevinsen requests are always handled in order by the queue: https://github.com/Koenkk/zigbee-herdsman/blob/e0b6ceaeca24ede57930f2552b2332b7108ca211/src/adapter/z-stack/adapter/zStackAdapter.ts#L330 , requests to the same device will never be send simultaneously

kennylevinsen, 2021-11-08:
Eh you're right. Missed that - sorry for the noise.

closed 2021-11-08 by kennylevinsen