Koenkk/zigbee-herdsman #460: LED on CC2530 coordinator

kolbai, 2021-11-23:
Can anyone please clarify, if CC2530 flashed with the latest coordinator firmware supports LED control? I see contradictions between [this](https://github.com/Koenkk/zigbee-herdsman/blob/master/src/adapter/z-stack/adapter/zStackAdapter.ts#L245) and [this](https://github.com/Koenkk/Z-Stack-firmware/blob/develop/coordinator/Z-Stack_Home_1.2/firmware.patch#L31). If it is supported, which pin is used for LED? Can we only set LED to either Enabled or Disabled, but cannot set it to Blink or Flash?

Koenkk, 2021-11-23:
With the new firmware, the firmware itself will control the led when joining is permitted/disabled, you cannot control the LED yourself.

kolbai, 2021-11-23:
Which pin is assigned to LED? As I understand, it should be HAL_LED_3, but I see no activity on P1.4.

Koenkk, 2021-11-23:
Not sure, you can find this in the source code of zstack 1.2 code.

kolbai, 2021-11-24:
I have probed each available pin on CC2530, none of them seem to change status when joining is enabled/disabled. Can anyone please confirm that this feature is working? @igogold ?

Koenkk, 2021-11-24:
Ah sorry, I just remember that LED functionality is disabled for the CC2530 in the Z-Stack home stack.

kolbai, 2021-11-24:
Thank you for clarification. That's very unexpected, given that CC2531 does support LED. I guess I was mislead by your changelog [entry](https://github.com/Koenkk/Z-Stack-firmware/blob/develop/coordinator/Z-Stack_Home_1.2/CHANGELOG.md) "Turn on/off leds when joining is enabled/disabled". Please consider adding this functionality to CC2530 in the future.

closed 2021-11-24 by kolbai