Koenkk/zigbee-herdsman #466: Rename all Tuya manu-speicific cluster commands

hacker-cb, 2021-12-04:
Need to rename all existing Tuya commands and add new commands according [official Tuya commands documentation](https://developer.tuya.com/en/docs/iot-device-dev/tuya-zigbee-universal-docking-access-standard?id=K9ik6zvofpzql#subtitle-6-Private%20cluster).

Previous [discussion](https://github.com/Koenkk/zigbee-herdsman/commit/e0b6ceaeca24ede57930f2552b2332b7108ca211).

Other Tuya  links:

- [UART Protocol for Zigbee Three-level Architecture](https://developer.tuya.com/en/docs/iot/zigbee-module-uart-protocol?id=Ka3hdzgitg4kh#title-4-Frame%20format%20description)

hacker-cb, 2021-12-04:
@Koenkk of cause this will break compatibility at leat between `herdsman` and `converters`.

So, please confirm that you are ready to that :)

Koenkk, 2021-12-04:
@hacker-cb I will make sure to merge the prs at the same time.

closed 2022-01-11 by github-actions[bot]