Koenkk/zigbee-herdsman #472: UnhandledPromiseRejectionWarning: Error: Cluster 'genBasic' has no attribute '57351'

willu, 2021-12-10:
I recently purchased a Schnider Electric temp/humidity sensor.  When I tried to get it working I got this:

```
Zigbee2MQTT:debug 2021-12-10 10:58:41: Received Zigbee message from 'XXXX', type 'read', cluster 'genBasic', data '[57351]' from endpoint 1 with groupID 0
(node:7) UnhandledPromiseRejectionWarning: Error: Cluster 'genBasic' has no attribute '57351'
    at Object.getAttribute (/app/node_modules/zigbee-herdsman/src/zcl/utils.ts:98:19)
    at Device.onZclData (/app/node_modules/zigbee-herdsman/src/controller/model/device.ts:234:48)
    at Controller.onZclOrRawData (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:671:20)
    at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:135:70)
    at ZStackAdapter.emit (events.js:400:28)
    at ZStackAdapter.onZnpRecieved (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:796:30)
    at Znp.emit (events.js:400:28)
    at Znp.onUnpiParsed (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:108:18)
    at Parser.emit (events.js:400:28)
    at Parser.parseNext (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/unpi/parser.ts:45:26)
```
If it helps, this is the converter I added (the stack trace is in zigbee-herdsman rather than one of the other projects - hence filing here):
```
; more CLP593011.js 
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['CCT593011_AS'],
    model: 'CLP593011',
    vendor: 'Schneider Electric',
    description: 'Wiser Temperature/Humidity sensor',
    fromZigbee: [fz.battery, fz.schneider_temperature, fz.humidity], 
    toZigbee: [],
    exposes: [e.battery(), e.temperature(), e.humidity()],
};

module.exports = definition;
```

Those fromZigbee, toZigbee and exposes sections are guesses - I didn't get far enough to know if they're right.

The device is this one: https://www.clipsal.com/products/detail?CatNo=CLP593011  (Clipsal is the Australian brand used by Schneider Electric.  i.e. this should be the same device: https://www.se.com/ww/en/product/CLP593011/wiser-temperature-humidity-smart-sensor/ ).

willu, 2021-12-10:
I'm guessing a fix that guards the `getAttribute()` call here: https://github.com/Koenkk/zigbee-herdsman/blob/master/src/controller/model/device.ts#L234 would work.  Something similar to this:
 https://github.com/Koenkk/zigbee-herdsman/commit/71ad17ebbe828e5c5166b688df07339dcce22e3b

I have to run now though - so I'll test that later.

willu, 2021-12-13:
Sigh.  With this diff I no longer get the exception. But the device still doesn't seem to work.

```
diff --git a/src/controller/model/device.ts b/src/controller/model/device.ts
index 680c25b..0be4227 100755
--- a/src/controller/model/device.ts
+++ b/src/controller/model/device.ts
@@ -230,11 +230,13 @@ class Device extends Entity {
 
             if (frame.Cluster.name in attributes && (frame.Cluster.name !== 'genTime' || !this._skipTimeResponse)) {
                 const response: KeyValue = {};
-                for (const entry of frame.Payload) {
-                    const name = frame.Cluster.getAttribute(entry.attrId).name;
-                    if (name in attributes[frame.Cluster.name].attributes) {
-                        response[name] = attributes[frame.Cluster.name].attributes[name];
-                    }
+               for (const entry of frame.Payload) {
+                   if (cluster.hasAttribute(entry.attrId)) {
+                        const name = frame.Cluster.getAttribute(entry.attrId).name;
+                        if (name in attributes[frame.Cluster.name].attributes) {
+                            response[name] = attributes[frame.Cluster.name].attributes[name];
+                       }
+                   }
                 }
 
                 try {
```

During setup I see:
```
Zigbee2MQTT:debug 2021-12-13 21:03:00: Device '0x804b50fffedbd163' announced itself                                                                                     
Zigbee2MQTT:info  2021-12-13 21:03:00: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x804b50fffedbd163","ieee_address":"0x804b50ff
fedbd163"},"type":"device_announce"}'                                                                                                                                   
Zigbee2MQTT:debug 2021-12-13 21:03:01: Received Zigbee message from '0x804b50fffedbd163', type 'read', cluster 'genBasic', data '[57351]' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-12-13 21:03:02: Retrieving state of '0x804b50fffedbd163' after reconnect                                                                         
Zigbee2MQTT:debug 2021-12-13 21:03:02: Received Zigbee message from '0x804b50fffedbd163', type 'read', cluster 'genBasic', data '[57351]' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-12-13 21:03:03: Received Zigbee message from '0x804b50fffedbd163', type 'read', cluster 'genBasic', data '[57351]' from endpoint 1 with groupID 0
```
Then later an OTA request.  But no other data.

The database.db file ends up with this:
```
{"id":13,"type":"EndDevice","ieeeAddr":"0x804b50fffedbd163","nwkAddr":41568,"manufId":4190,"manufName":"Schneider Electric","powerSource":"Battery","modelId":"CCT593011
_AS","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,1,3,32,1026,1029,2821],"outClusterList":[0,3,25],"clusters":{"genBasic":{"attri
butes":{"modelId":"CCT593011_AS","manufacturerName":"Schneider Electric","powerSource":3,"zclVersion":3,"appVersion":3,"stackVersion":6,"hwVersion":0,"dateCode":"201906
19","swBuildId":"1906191"}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x00124b0022a50f51","endpointID":1}],"configuredReportings":[],"meta":{}}},"a
ppVersion":3,"stackVersion":6,"hwVersion":0,"dateCode":"20190619","swBuildId":"1906191","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1639389775748,"use
ImplicitCheckin":false,"defaultSendWhenActive":true}
```

But this may be generic new device issues now.  I'll open a converters but and continue there.

willu, 2021-12-13:
57351 is 0xE007 - I can't find anything about that as a basic cluster attribute anywhere on the web.

willu, 2021-12-14:
I think https://github.com/Koenkk/zigbee-herdsman/pull/475 should help.

Koenkk, 2021-12-14:
Assuming this has been fixed by #475

closed 2021-12-14 by Koenkk