Koenkk/zigbee-herdsman #480: zoneStatusChanged and panelStatusChanged from IAS Cluster should be Commands not CommandsResponse in cluster.js

olix74, 2021-12-19:
Hello,
I'm not completlely sure if I understand the structure of zigbee-herdsman but my assumption would be that zoneStatusChanged and panelStatusChanged should be commands rather than responses since there are no respective commands from the IAS device. Instead the IAS server is pushing them to the device when needed.
Please see https://zigbeealliance.org/wp-content/uploads/2019/12/07-5123-06-zigbee-cluster-library-specification.pdf section 8.3.2.5.5 
As such they should be moved from the CommandsResponse section to commands section in cluster.js to enable proper sending.
What do you think?

Thank you, for all the effort!
Oliver

Koenkk, 2021-12-19:
Wether a command is a `command` or `commandResponse` depends on the direction of the command, (client -> server vs server -> client)

olix74, 2021-12-20:
@Koenkk, hard for me to disagree, since I'm not an expert on the "herdsman" as you are. Nevertheless, I have the following obeservations to consider.
1. Just another example. In the ssIasWd cluster the command from the "server" to the device to "startwarning" is modeled as a `command`. Same is for the panelStatusChanged and zoneStatusChanged. They are also being send by the server to the client but they are modelled as response. So I don't understand the difference. 
2. a response is the answer to a request. A command is not a response, since there is no request. As such panelStatusChanged and zoneStatusChanged are not a response to anything received from the client. In contrast getPanelStatusRsp is a real response to a command/request from the client - so that is clear to me.
3. in endpoint.js there are two functions: comman() and commandResponse(). I assumed command() is used to send commands and commandResponse() to send responses. Right? commandResponse() requires the transaction number of the request, which is not available for panelStatusChanged, since there is no client request. On the other hand "command()" seems to refuse to send anything from the commandResponse cluster.
Sorry to bother you again and thank you for your feedback,
Oliver

Koenkk, 2021-12-20:
I think the terminology of the `command`/`commandsResponse` is indeed not very good (since it is also not written as such in the ZCL). TBH I've never really looked at this when I created zigbee-herdsman from zigbee-shepherd, the original file can be found [here](https://github.com/zigbeer/zcl-id/blob/master/definitions/cluster_defs.json). Could you experiment a bit wether the direction makes a difference for e.g. the `start warning` command?  Based on what I read in the ZCL spec this is not really clear to me:

[image]

olix74, 2021-12-27:
Hello @Koenkk,
I read throuh the spec, reviewed the code and checked with some IAS devices. I think you are right, the main difference between command() and commandResponse() is the "Direction" bit. 

While "response" always sends SERVER_TO_CLIENT, "command" sends from CLIENT_TO_SERVER. 
As such all messages from SERVER_TO_CLIENT currently have to be in the  CommandsResponse section in cluster.js independent from the fact if they are a response or not.

I think the section 2.5.9.1.2 is not relevant here but if you check Figure 8-1 in the spec, you will notice that Client-Server relationship is flipped for IAS_WD (warning) and IAS_ACE (controll) devices and that was the reason why "command" worked for a  "startwarning" to an IAS_WD device but didn't work for "panelStatusChanged" to an IAS_ACE device . 

I still think the naming of "command" and "commandResponse" is kind of misleading, but that shouldn't be a reason to change it right now - maybe add a comment that "command" means "commandToServer" and "response" means "commandToClient".

Thank you again for sharing this project with us/me and have a great 2022.
I think this issue can be closed.
Oliver

closed 2021-12-27 by olix74