Koenkk/zigbee-herdsman #484: crash during onboarding of Tradfri motion sensor

toanju, 2021-12-24:
Hi,

I got a Tradfri motion sensor and had some difficulties to add it. One of the issues it crashed Zigbee2MQTT at least once (log below) before it was able to write anything into the database. Meaning

* paring started
* device id added to configuarion.yaml
* crash of Zigbee2MQTT
* add information to database.db

I could not trace so far the other issue that occured: The pairing itself took many trials until it actually worked.

Logs

```
Dec 23 14:50:24 rpi3 podman[16282]: Zigbee2MQTT:info  2021-12-23 15:50:24: Starting Zigbee2MQTT version 1.22.1 (commit #c2b5229)
Dec 23 14:50:24 rpi3 podman[16282]: Zigbee2MQTT:info  2021-12-23 15:50:24: Starting zigbee-herdsman (0.13.176)
...
Dec 23 15:35:37 rpi3 podman[16282]: Zigbee2MQTT:info  2021-12-23 16:35:37: Starting interview of '0x8cf681fffe516258'
Dec 23 15:35:37 rpi3 podman[16282]: Zigbee2MQTT:info  2021-12-23 16:35:37: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x8cf681fffe516258","ieee_address":"0x8cf681fffe516258","status":"started"},"type":"device_interview"}'
Dec 23 15:35:37 rpi3 podman[16282]: Zigbee2MQTT:info  2021-12-23 16:35:37: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x8cf681fffe516258"},"type":"pairing"}'
Dec 23 15:35:39 rpi3 podman[16282]: Zigbee2MQTT:info  2021-12-23 16:35:39: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x8cf681fffe516258","ieee_address":"0x8cf681fffe516258"},"type":"device_announce"}'
Dec 23 15:35:39 rpi3 podman[16282]: Zigbee2MQTT:info  2021-12-23 16:35:39: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x8cf681fffe516258"},"type":"device_announced"}'
Dec 23 15:35:48 rpi3 podman[16282]: Zigbee2MQTT:warn  2021-12-23 16:35:48: Device '0x8cf681fffe516258' left the network
Dec 23 15:35:48 rpi3 podman[16282]: Zigbee2MQTT:info  2021-12-23 16:35:48: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x8cf681fffe516258","ieee_address":"0x8cf681fffe516258"},"type":"device_leave"}'
Dec 23 15:35:48 rpi3 podman[16282]: Zigbee2MQTT:info  2021-12-23 16:35:48: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x8cf681fffe516258"},"type":"device_removed"}'
Dec 23 15:35:57 rpi3 podman[16282]: TypeError: Cannot read property 'name' of undefined
Dec 23 15:35:57 rpi3 podman[16282]:     at Zigbee.logDeviceInterview (/app/lib/zigbee.ts:144:34)
Dec 23 15:35:57 rpi3 podman[16282]:     at Controller.<anonymous> (/app/lib/zigbee.ts:84:18)
Dec 23 15:35:57 rpi3 podman[16282]:     at Controller.emit (events.js:400:28)
Dec 23 15:35:57 rpi3 podman[16282]:     at Controller.onDeviceJoined (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:531:22)
Dec 23 15:35:57 rpi3 systemd[879]: libpod-3c1aaf9adf396d55cd85126803eee5014a3882825dd4eae5d0529aec9e24fdfa.scope: Consumed 2min 23.787s CPU time.
```

Koenkk, 2021-12-24:
Already fixed by https://github.com/Koenkk/zigbee2mqtt/commit/aa84a140e11618524386dcc717a6f94e58e0319e  (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2021-12-24 by Koenkk

toanju, 2021-12-24:
Thanks for the quick return. Just tested this with the latest dev container. Works like a charm. As well the device interview works now even though it takes some time (approx. 6minutes in my case).

Happy Holidays!