Koenkk/zigbee-herdsman #511: setting stackTune to 20 on CC2652P

TA2k, 2022-03-13:
Im using a 
Sonoff Zigbee 3.0
Firmware: 20220219
CC1352P2_CC2652P_launchpad_coordinator_20220219.hex 
'zStack3x0' ({"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20220219})

I'm trying to set the transmit power to 20

```
zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - stackTune - {"operation":0,"value":"20"}
zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,2,33,15,0,20,56]
zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,97,15,0,111]
zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,97,15,0,111]
zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 1 - 15 - [0] - 111
zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - stackTune - {"value":0}
zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
```

I looks like the value 20 is not applied. Im not sure is this a firmware, herdsman issue or a misinterpretation

TA2k, 2022-04-13:
Is this the right repo for such a issue?

Koenkk, 2022-04-13:
> I looks like the value 20 is not applied.

What makes you think so?

TA2k, 2022-04-13:
`znp:SRSP <-- SYS - stackTune - {"value":0}`

I would expect

`znp:SRSP <-- SYS - stackTune - {"value":20}`

Koenkk, 2022-04-14:
The value doesn't indicate the new transmit power, it indicates whether the operation was successful or not (0 = OK), from the docs:

[image]

closed 2022-04-14 by TA2k