Koenkk/zigbee-herdsman #525: Manufacturer-Specific Attribute duplicates to standard

theimo1221, 2022-04-15:
Good evening,

today I faced several difficulties interacting with Ubisys-J1 Shutter actuator.

According to the device technical specification it's `Window Covering Cluster` has beneath the normal standard attributes a few custom ones:
https://github.com/Koenkk/zigbee-herdsman/blob/fb28dae8fd537e96bbfe9119fcc74d11db08f5b0/src/zcl/definition/cluster.ts#L1805-L1812

Additionally they add some write capabilities to some normal attributes:
[image]
https://github.com/Koenkk/zigbee-herdsman/blob/fb28dae8fd537e96bbfe9119fcc74d11db08f5b0/src/zcl/definition/cluster.ts#L1784
https://github.com/Koenkk/zigbee-herdsman/blob/fb28dae8fd537e96bbfe9119fcc74d11db08f5b0/src/zcl/definition/cluster.ts#L1791
https://github.com/Koenkk/zigbee-herdsman/blob/fb28dae8fd537e96bbfe9119fcc74d11db08f5b0/src/zcl/definition/cluster.ts#L1795-L1798

If I try to write those attributes I need to specify the command as `manufSpec` and provide the manufacturer code (4438) as otherwise the device correctly responds "READ-ONLY (136)".

I tried adding it `ubisysWindowCoveringType: {ID: 0x0000, type: DataType.enum8, manufacturerCode: ManufacturerCode.Ubisys},` but this didn't work for me, only the normal `windowCoveringType` kept beeing listed.

So my question is, how can we define manufacuterer specific overrides which only affect certain commands (like in this case adding writing capabilities to normaly read-only properties).

Best Regards

Thiemo

Koenkk, 2022-04-15:
Can you try with: `await endpoint.write('closuresWindowCovering', {'windowCoveringType': 0}, {manufacturerCode: 0x10f2});`

theimo1221, 2022-04-15:
That works, so apparently the `manufSpec` is irrelevant for this one.

Koenkk, 2022-04-15:
the only thing adding manufacturer  code to clusters.ts does is sending it automatically vs having to manually pass it as an option to the .write()

theimo1221, 2022-04-15:
But as we have to leave it empty for those read calls we can't simply add their manufacturer code to the standard attribute, correct?

Does the cluster definitions have such an option to add manufacturer code to specific command types, or otherwise could we add a new attribute definition like I tried above for 0x0000 with manufacturer code parallel to the standard one (without manufacturer code)?

Koenkk, 2022-04-16:
> we add a new attribute definition like I tried above for 0x0000 with manufacturer code parallel to the standard one

Can be done, but it is better to create a new converter in `toZigbee` to provide a more human friendly interface to setting this (there you can also do a mapping from a friendly string -> enum8)

theimo1221, 2022-04-16:
Within the converter there is already a function covering the calibration for J1 Shutter including writing of that window type:

https://github.com/Koenkk/zigbee-herdsman-converters/blob/677e5c95e527e7ecce6c5c259bea653d254c4b97/converters/toZigbee.js#L4251

And the writing of those attribute is done using some local helper function `writeAttrFromJson`:
`await entity.write('closuresWindowCovering', attributes, manufacturerOptions.ubisys);`

Thanks for the info 👍

closed 2022-04-16 by theimo1221