Koenkk/zigbee-herdsman #540: Bump serialport version to 10.4.0

ercksen, 2022-05-22:
Hey, I am having struggle to use this library on NodeJS versions > 12.

It seems that the version of `serialport` is kind of old and the `@serialport/bindings` package is officially marked as deprecated and replaced by `@serialport/bindings-cpp`.

I have also seen quite some users having problems after updating the NodeJS version. There could also be some information inside the README about version requirements.

The output I get:

`The module 'node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build/Release/bindings.node'
smart-hub-smarthub-backend-1   | was compiled against a different Node.js version using
smart-hub-smarthub-backend-1   | NODE_MODULE_VERSION 72. This version of Node.js requires
smart-hub-smarthub-backend-1   | NODE_MODULE_VERSION 93. Please try re-compiling or re-installing
smart-hub-smarthub-backend-1   | the module (for instance, using `npm rebuild` or `npm install`)`

Trying to bump the version on my own was not a great success. It would be nice to make this project compatible with newer versions.

Please correct me if I'm wrong and just don't know how to install this lib correctly.

ercksen, 2022-05-22:
I finally fixed the issue by installing the correct binaries. The problem were different node versions between installation and execution of the modules so wrong binaries were installed.

But I will keep the issue open in order to decide whether it could be useful anyways to update to the new package.

Koenkk, 2022-05-22:
The current `@serialport/bindings-cpp` has some issues with the prebuilt binaries and therefore we did not update yet:

- https://github.com/serialport/node-serialport/issues/2438
- https://github.com/Koenkk/zigbee2mqtt/issues/11545

closed 2022-06-30 by github-actions[bot]