Koenkk/zigbee-herdsman #553: Disable "CREATED DECONZ ADAPTER" log statement

ercksen, 2022-06-28:
I propose to disable the "CREATED DECONZ ADAPTER" log statement which I unfortunately cannot disable, in `deconzAdapter.js`:62

```
console.log('CREATED DECONZ ADAPTER');
this.frameParserEvent.on('receivedDataPayload', (data) => { this.checkReceivedDataPayload(data); });
this.frameParserEvent.on('receivedGreenPowerIndication', (data) => { this.checkReceivedGreenPowerIndication(data); });
const that = this;
setInterval(() => { that.checkReceivedDataPayload(null); }, 1000);
setTimeout(() => { that.checkCoordinatorSimpleDescriptor(false); }, 3000);
// <end of constructor>
```

If I have understood correctly, the following instructions cannot throw an error. So after the constructor returns, one could add a manual log statement if desired, without changing any semantics. Returning from constructor without throwing is then equal to successful adapter creation. As I am using a custom logging system and use stdout for other purposes, the only option I have is to disable it via editing `node_modules` content.

I think this would add more flexibility as one can not prevent any logging actions that are done inside external code.

Apart from that, thanks for your work! This is a great library!

Koenkk, 2022-06-29:
Removed!

closed 2022-06-29 by Koenkk

ercksen, 2022-06-29:
Thanks for your super-fast reply and for removing!