Koenkk/zigbee-herdsman #572: Trust ZPIR-8000 always TRUE

Xerxes55, 2022-08-30:
asgothian Worte — please refer: https://github.com/ioBroker/ioBroker.zigbee/issues/1359
but I see no issues about it here. Can you solve the problem?

Xerxes55, 2022-10-02:
Please do not delete. There are several posts on the subject. Such as https://github.com/Koenkk/zigbee2mqtt/issues/7063 or https://github.com/Koenkk/zigbee2mqtt/issues/13480. Is there no one who can take care of it? There is a firmware update. Can't this be distributed?

Kryzek, 2022-10-04:
If the v0.1.0 version does not report the "no more motion" on it's own after motion is no longer detected but the v0.1.4 does for me it means there's something wonky on the firmware. But as long as firmware files from Trust are not available (without their gateway device) there is no way to update devices by other means.

closed 2022-11-12 by github-actions[bot]