Koenkk/zigbee-herdsman #590: Sonoff ZigBee USB Dongle Plus-P

MatAtBread, 2022-10-01:
I have one of these devices (the -P version based on the TI CC2652P and CP2102) which I'm trying to use with ioBroker under Windows.

I don't know much about ZigBee, but I'm a decent TS/JS programmer. It fails to detect, looking into the code, it fails against the tests for 

```
const autoDetectDefinitions = [
    { manufacturer: 'Texas Instruments', vendorId: '0451', productId: '16c8' },
    { manufacturer: 'Texas Instruments', vendorId: '0451', productId: '16a8' },
    { manufacturer: 'Texas Instruments', vendorId: '0451', productId: 'bef3' },
    { manufacturer: 'Electrolama', vendorId: '0403', productId: '6015' }, // ZZH
];
```
...in adapter/z-stack/znp/znp.js

The reported serial port is:
```
{
  "path": "COM12",
  "manufacturer": "Silicon Labs",
  "serialNumber": "62a88fbb4d19ec1182f53bcc47486eb0",
  "pnpId": "USB\\VID_10C4&PID_EA60\\62A88FBB4D19EC1182F53BCC47486EB0",
  "locationId": "Port_#0006.Hub_#0003",
  "vendorId": "10C4",
  "productId": "EA60"
}
```
...any ideas? I've read a few things about re-flashing the dongle, but though I'd ask if that's necessary before I randomly try the various bin files scattered around the internet.

Koenkk, 2022-10-01:
@kirovilya what is the identifier for the Plus-**E** dongle?

kirovilya, 2022-10-01:
```
{
  manufacturer: 'ITEAD',
  serialNumber: '20220712182607',
  pnpId: 'usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220712182607-if00',
  locationId: undefined,
  vendorId: '1a86',
  productId: '55d4',
  path: '/dev/ttyACM0'
}
```

MatAtBread, 2022-10-01:
FYI, I've also installed on a Rasp Pi3 and it appears as /dev/ttyUSV0, but also not found during discovery in ioBroker.

```
pi@pi:~ $ lsusb
Bus 001 Device 005: ID 10c4:ea60 Cygnal Integrated Products, Inc. CP2102/CP2109 UART Bridge Controller [CP210x family]
```

dmesg:
```
[  491.565437] usb 1-1.3: new full-speed USB device number 5 using dwc_otg
[  491.700333] usb 1-1.3: New USB device found, idVendor=10c4, idProduct=ea60, bcdDevice= 1.00
[  491.700348] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  491.700358] usb 1-1.3: Product: Sonoff Zigbee 3.0 USB Dongle Plus
[  491.700367] usb 1-1.3: Manufacturer: ITead
[  491.700376] usb 1-1.3: SerialNumber: 62a88fbb4d19ec1182f53bcc47486eb0
[  491.765656] usbcore: registered new interface driver usbserial_generic
[  491.767606] usbserial: USB Serial support registered for generic
[  491.776864] usbcore: registered new interface driver cp210x
[  491.777339] usbserial: USB Serial support registered for cp210x
[  491.777885] cp210x 1-1.3:1.0: cp210x converter detected
[  491.784764] usb 1-1.3: cp210x converter now attached to ttyUSB0
```

MatAtBread, 2022-10-01:
UPDATE: The Rasp Pi3 issue was that iobroker.zigbee failed to install. `npm i` revealed the issue: old python version.

I've now build Python3.9.5 on the Pi and I can `npm i iobroker.zigbee` without error. Running `iobroker add zigbee --host *****` also completes with no error.

Then restarting iobroker and adding the Zigbee adapter errors with:
```
Starting zigbee-herdsman problem : "No path provided and failed to auto detect path"
```

....maybe the dongle is fried? (Fresh out of the box). Does herdsman have a test script that simply tries to access the device without doing anything complex? Or is it just the device ID still not being found (as in Windows)?

MatAtBread, 2022-10-01:
I successfully flashed `CC1352P2_CC2652P_launchpad_coordinator_20220219.hex`, so I guess the device is working, but no difference to the USB IDs or discovery.

Any advice/pointers you can give as to how to proceed would be gratefully accepted.

MatAtBread, 2022-10-01:
Out of desperation, I just hacked the USB info into the definitions array and put a "?." around where it threw with 
```
TypeError: Cannot read properties of undefined (reading 'getClients')
    at Commands.cleanDeviceStates (C:\ioBroker\node_modules\iobroker.zigbee\lib\commands.js:442:55)
```
....and....it worked!! (on Windows).

Koenkk, 2022-10-05:
I checked the id of the Sonoff dongle Plus-P and it uses a generic ID meaning we cannot distinguish this from other identifiers. (the sprut and CC2652RB use exactly the same ID).

```js
  {
    path: '/dev/tty.usbserial-110',
    manufacturer: 'Silicon Labs',
    serialNumber: 'de4374fdea93eb11b514a85b3d98b6d1',
    pnpId: undefined,
    locationId: '01100000',
    vendorId: '10c4',
    productId: 'ea60'
  }
```

> TypeError: Cannot read properties of undefined (reading 'getClients')
    at Commands.cleanDeviceStates (C:\ioBroker\node_modules\iobroker.zigbee\lib\commands.js:442:55)

This is an iobroker error, please create an issue about it in the iobroker issue tracker.