Koenkk/zigbee-herdsman #620: Problems with ZBDongle-E and TRADFRI on/off switch (E1743)

baylf2000, 2022-11-28:
I'm having problems with the Ikea TRADFRI on/off switch (E1743) leaving the network after about five minutes of inactivity with the message 
`Device '0x943469fffe3e27c3' left the network`  when using the ZBDongle-E. I was running z2m on Home Assistant but have since moved to a standalone install for testing. I have other mains powered devices on the network which are perfectly stable.

Things I've done to troubleshoot:
- Directly installed on a PC. (No change)
- Moved to dev branch. (No change)
- Tried several different firmware versions on the device, including the latest official firmware, and the latest 7.1.1.0 from [https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/Zigbee3.0_Dongle-NoSigned/EZSP](url) (No Change)
- Swapped to a CC2531 adapter. This resolves the issue. The device no longer leaves the network, even after many hours.

Apologies if this is not the correct place for this.

baylf2000, 2022-11-29:
Not sure if this helps, but here's some debug info that captures the remote leaving and then rejoining the network.

>  zigbee-herdsman:adapter:ezsp:ezsp Time to watchdog ... 0 +10s
  zigbee-herdsman:adapter:ezsp:ezsp ==> nop: null +0ms
  zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"nop","_id_":5,"_isRequest_":true} +0ms
  zigbee-herdsman:adapter:ezsp:uart --> DATA (4,2,0): 4400010500 +10s
  zigbee-herdsman:adapter:ezsp:uart --> [420621a9512af8947e] +10s
  zigbee-herdsman:adapter:ezsp:uart -?- waiting (5) +0ms
  zigbee-herdsman:adapter:ezsp:uart <-- [2506a1a9512ab2f57e] +10s
  zigbee-herdsman:adapter:ezsp:uart <-- DATA (2,5,0): 2506a1a9512ab2f57e +4ms
  zigbee-herdsman:adapter:ezsp:uart --> ACK  (3) +0ms
  zigbee-herdsman:adapter:ezsp:uart --> [83401b7e] +4ms
  zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 4480010500 +5ms
  zigbee-herdsman:adapter:ezsp:ezsp <== 0x5: {"_id_":5,"_cls_":"nop","_isRequest_":false} +1ms
  zigbee-herdsman:adapter:ezsp:uart -+- waiting (5) success +1ms
  zigbee-herdsman:adapter:ezsp:ezsp Time to watchdog ... 0 +10s
  zigbee-herdsman:adapter:ezsp:ezsp ==> nop: null +0ms
  zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"nop","_id_":5,"_isRequest_":true} +0ms
  zigbee-herdsman:adapter:ezsp:uart --> DATA (5,3,0): 4500010500 +10s
  zigbee-herdsman:adapter:ezsp:uart --> [530721a9512a0de17e] +10s
  zigbee-herdsman:adapter:ezsp:uart -?- waiting (6) +0ms
  zigbee-herdsman:adapter:ezsp:uart <-- [3607a1a9512accc07e] +10s
  zigbee-herdsman:adapter:ezsp:uart <-- DATA (3,6,0): 3607a1a9512accc07e +6ms
  zigbee-herdsman:adapter:ezsp:uart --> ACK  (4) +0ms
  zigbee-herdsman:adapter:ezsp:uart --> [8430fc7e] +6ms
  zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 4580010500 +7ms
  zigbee-herdsman:adapter:ezsp:ezsp <== 0x5: {"_id_":5,"_cls_":"nop","_isRequest_":false} +1ms
  zigbee-herdsman:adapter:ezsp:uart -+- waiting (6) success +1ms
  zigbee-herdsman:adapter:ezsp:ezsp Time to watchdog ... 0 +10s
  zigbee-herdsman:adapter:ezsp:ezsp ==> nop: null +0ms
  zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"nop","_id_":5,"_isRequest_":true} +0ms
  zigbee-herdsman:adapter:ezsp:uart --> DATA (6,4,0): 4600010500 +10s
  zigbee-herdsman:adapter:ezsp:uart --> [640421a9512a04fe7e] +10s
  zigbee-herdsman:adapter:ezsp:uart -?- waiting (7) +0ms
  zigbee-herdsman:adapter:ezsp:uart <-- [4704a1a9512a222e7e] +10s
  zigbee-herdsman:adapter:ezsp:uart <-- DATA (4,7,0): 4704a1a9512a222e7e +5ms
  zigbee-herdsman:adapter:ezsp:uart --> ACK  (5) +1ms
  zigbee-herdsman:adapter:ezsp:uart --> [8520dd7e] +6ms
  zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 4680010500 +7ms
  zigbee-herdsman:adapter:ezsp:ezsp <== 0x5: {"_id_":5,"_cls_":"nop","_isRequest_":false} +1ms
  zigbee-herdsman:adapter:ezsp:uart -+- waiting (7) success +1ms
  zigbee-herdsman:adapter:ezsp:ezsp Time to watchdog ... 0 +10s
  zigbee-herdsman:adapter:ezsp:ezsp ==> nop: null +1ms
  zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"nop","_id_":5,"_isRequest_":true} +0ms
  zigbee-herdsman:adapter:ezsp:uart --> DATA (7,5,0): 4700010500 +10s
  zigbee-herdsman:adapter:ezsp:uart --> [750521a9512af18b7e] +10s
  zigbee-herdsman:adapter:ezsp:uart -?- waiting (0) +1ms
  zigbee-herdsman:adapter:ezsp:uart <-- [5005a1a9512a5aba7e] +10s
  zigbee-herdsman:adapter:ezsp:uart <-- DATA (5,0,0): 5005a1a9512a5aba7e +5ms
  zigbee-herdsman:adapter:ezsp:uart --> ACK  (6) +0ms
  zigbee-herdsman:adapter:ezsp:uart --> [8610be7e] +5ms
  zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 4780010500 +7ms
  zigbee-herdsman:adapter:ezsp:ezsp <== 0x5: {"_id_":5,"_cls_":"nop","_isRequest_":false} +0ms
  zigbee-herdsman:adapter:ezsp:uart -+- waiting (0) success +2ms
  zigbee-herdsman:adapter:ezsp:ezsp Time to watchdog ... 0 +10s
  zigbee-herdsman:adapter:ezsp:ezsp ==> nop: null +0ms
  zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"nop","_id_":5,"_isRequest_":true} +0ms
  zigbee-herdsman:adapter:ezsp:uart --> DATA (0,6,0): 4800010500 +10s
  zigbee-herdsman:adapter:ezsp:uart --> [060a21a9512a1f0e7e] +10s
  zigbee-herdsman:adapter:ezsp:uart -?- waiting (1) +0ms
  zigbee-herdsman:adapter:ezsp:uart <-- [610aa1a9512a556f7e] +10s
  zigbee-herdsman:adapter:ezsp:uart <-- DATA (6,1,0): 610aa1a9512a556f7e +5ms
  zigbee-herdsman:adapter:ezsp:uart --> ACK  (7) +0ms
  zigbee-herdsman:adapter:ezsp:uart --> [87009f7e] +5ms
  zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 4880010500 +6ms
  zigbee-herdsman:adapter:ezsp:ezsp <== 0x5: {"_id_":5,"_cls_":"nop","_isRequest_":false} +1ms
  zigbee-herdsman:adapter:ezsp:uart -+- waiting (1) success +1ms
  zigbee-herdsman:adapter:ezsp:ezsp Time to watchdog ... 0 +10s
  zigbee-herdsman:adapter:ezsp:ezsp ==> nop: null +1ms
  zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"nop","_id_":5,"_isRequest_":true} +0ms
[... 501 lines cut ...]
  zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"sendUnicast","_id_":52,"_isRequest_":true,"type":0,"indexOrDestination":29206,"apsFrame":{"clusterId":32,"profileId":260,"sequence":144,"sourceEndpoint":1,"destinationEndpoint":1,"groupId":0,"options":320},"messageTag":145,"message":{"type":"Buffer","data":[16,110,0,0,0]}} +1ms
  zigbee-herdsman:adapter:ezsp:uart --> DATA (5,2,0): 650001340000167204012000010140010000909105106e000000 +4ms
  zigbee-herdsman:adapter:ezsp:uart --> [522721a9602a15a42b904b05aa5493099d4e273b7ccb77e5fdc663d5dd7e] +6ms
  zigbee-herdsman:adapter:ezsp:uart -?- waiting (6) +1ms
  zigbee-herdsman:adapter:ezsp:uart <-- [2627a1a9602a154127097e] +19ms
  zigbee-herdsman:adapter:ezsp:uart <-- DATA (2,6,0): 2627a1a9602a154127097e +8ms
[... cut ...] ion":"Delta per interval, 20 by default","name":"delta","property":"delta","type":"numeric","value_min":0},{"access":2,"description":"Interval duration","name":"interval","property":"interval","type":"numeric","unit":"ms","value_min":0}],"name":"simulated_brightness","property":"simulated_brightness","type":"composite"}],"supports_ota":true,"vendor":"IKEA"},"friendly_name":"0x943469fffe3e27c3","ieee_address":"0x943469fffe3e27c3","status":"successful","supported":true},"type":"device_interview"}'

kirovilya, 2022-11-30:
on which version of z2m are you checking? try on dev/edge version or wait for release (December 01)

baylf2000, 2022-11-30:
> on which version of z2m are you checking? try on dev/edge version or wait for release (December 01)

Hi, thanks for the reply. I've tried the dev version as mentioned, but I'll give it another try in a couple of weeks. Thanks again.

closed 2023-01-07 by github-actions[bot]