Koenkk/zigbee-herdsman #659: New clusters aren't available in GUI Dev Console

danieledwardgeorgehitchcock, 2023-02-05:
I have just added a couple of commands / clusters and attributes here #653 and have updated my docker container to the latest dev branch.

I can confirm that the code that I submitted in the PR is present in the latest image however, when trying to work with the new clusters via the dev console in Z2M, I cannot see the new clusters in the dropdown menu of the attributes screen. furthermore, when trying to execute a command using one of the new clusters and commands, I get a `cluster has no command` error.

Are there some additional files that I need to change in order to get this working in the GUI?

Koenkk, 2023-02-05:
@nurikk correct me if I'm wrong, but I think they need to be added to https://github.com/nurikk/zigbee2mqtt-frontend/blob/dev/src/components/cluster-picker/index.tsx first.

danieledwardgeorgehitchcock, 2023-02-05:
Thanks once again! I'll add them in there. 

Is there any way this could automatically be parsed in or something?

Also, I'm not sure this would explain the no command error, as this seems to be coming back from Herdsman?

Koenkk, 2023-02-05:
I believe you need to fill in the command number, not the name.

nurikk, 2023-02-05:
> @nurikk correct me if I'm wrong, but I think they need to be added to https://github.com/nurikk/zigbee2mqtt-frontend/blob/dev/src/components/cluster-picker/index.tsx first.

No, all cluster get their banes from herdsman. So wee need to bump it version in frontend repo

danieledwardgeorgehitchcock, 2023-02-06:
That makes sense! looking at the `package.json` file in the frontend repo, I can see that it is on a version of herdsman that dates back to Sep 23, 2022 (0.14.59) latest being (0.14.90).

Is there any special process on upping the version dependency?

danieledwardgeorgehitchcock, 2023-03-06:
@nurikk - this still appears to be an issue. I have pulled the latest version of the frontend into my instance, and the clusters are still not showing - I think the frontend project is still using quite an old version of Herdsman..

I have also tried building the frontend with the dependencies updated but unfortunately, I cannot get a stable compilation following the instructions in the wiki..

nurikk, 2023-03-06:
> @nurikk - this still appears to be an issue. I have pulled the latest version of the frontend into my instance, and the clusters are still not showing - I think the frontend project is still using [...]

Sorry, I had no time to work on this task