Koenkk/zigbee-herdsman #665: deconz adapter/driver does not support backup and restore for ConBee/RaspBee by dresden elektronik?

Hedda, 2023-02-15:
Does deconz adapter/driver in zigbee-herdsman not support backup and backup restore for ConBee/RaspBee by dresden elektronik?

Or is it currently only the zstack adapter/driver in zigbee-herdsman that supports both Zigbee Coordinator backup and restore?

See related discussion here with users are also asking about migrating both to and from ConBee -> https://github.com/Koenkk/zigbee2mqtt/discussions/16478

deconz adapter/driver Zigbee network backup to file and restore from that file backup is needed for backup and restore support + cross-hardware migration in order to allow Zigbee2MQTT and IoBroker users to from one adapter to another, regardless of radio type used:

* https://www.zigbee2mqtt.io/guide/faq/#how-do-i-migrate-from-one-adapter-to-another
  * https://www.zigbee2mqtt.io/guide/adapters/flashing/copy_ieeaddr.html  
  * https://www.zigbee2mqtt.io/guide/faq/#how-do-i-move-my-zigbee2mqtt-instance-to-a-different-environment
  * https://www.zigbee2mqtt.io/guide/faq/#what-does-and-does-not-require-repairing-of-all-devices

zigbee-herdsman has implemented the support for the standard "Open ZigBee Coordinator Backup Format" as an open file format for storing all information about a Zigbee network in JSON file needed to in a backup and restoring of that backup:

https://github.com/zigpy/open-coordinator-backup

zigpy-deconz which also uses deCONZ Serial Protocol does support backup and backup restore to  ConBee/RaspBee using this format:

https://github.com/zigpy/zigpy-deconz

https://github.com/dresden-elektronik/deconz-serial-protocol

Hedda, 2023-02-16:
FYI, [ChrisHae](https://github.com/ChrisHae) reply to https://github.com/dresden-elektronik/deconz-serial-protocol/issues/20 indicate this is in the roadmap for future deconcz firmware.

Hedda, 2023-02-17:
FYI, Also see related feature request for Dresden-Elektronik's deCONZ/Phoscon -> https://github.com/dresden-elektronik/deconz-rest-plugin-v2/issues/12

Hedda, 2023-02-23:
> FYI, Also see related feature request for Dresden-Elektronik's deCONZ/Phoscon -> [dresden-elektronik/deconz-rest-plugin-v2#12](https://github.com/dresden-elektronik/deconz-rest-plugin-v2/issues/12)

FYI as reference, in related news [ChrisHae](https://github.com/ChrisHae) posted in a reply in https://github.com/dresden-elektronik/deconz-rest-plugin-v2/issues/12 that he will look into "[Open ZigBee Coordinator Backup Format](https://github.com/zigpy/open-coordinator-backup)" export and import in the "deCONZ REST plugin" for the deCONZ/Phoscon API and application. See:

https://github.com/dresden-elektronik/deconz-rest-plugin-v2/issues/12

Note! To clarify, he only said there that he can look into it for deCONZ/Phoscon, so he did not mention that he would also look into the backup and restore feature for the deconz adapter in zigbee-herdsman or Zigbee2MQTT.

Hedda, 2023-03-26:
@ChrisHae any updates or feedback on this?

Hedda, 2023-04-27:
Not stale. Still valid and should be left open.

@ChrisHae Any news about this feature?

Hedda, 2023-05-28:
Still needed

Hedda, 2023-06-28:
Not stale

Hedda, 2023-07-30:
Do not stale

Hedda, 2023-10-02:
Still and issue for ConBee and RaspBee users

Hedda, 2023-11-03:
Not stale

Hedda, 2023-12-04:
@ChrisHae any updates or feedback on this deconz adapter backup and restore features still an open issue in zigbee-herdsman?

Hedda, 2023-12-05:
@ChrisHae This feature request is even more relevant now that Zigbee2MQTT users want to migrate to ConBee III (ConBee 3), ex:

https://github.com/Koenkk/zigbee2mqtt/discussions/19785

Hedda, 2024-03-18:
> This feature request is even more relevant now that Zigbee2MQTT users want to migrate to ConBee III (ConBee 3), ex:
> 
> [Koenkk/zigbee2mqtt#19785](https://github.com/Koenkk/zigbee2mqtt/discussions/19785)

Any news on ConBee III (ConBee 3) backup and restore support in zigbee-herdsman's deconz adapter driver for Zigbee2MQTT?

Hedda, 2024-09-16:
Still an issue that discourage the use of ConBee/RaspBee as get to backups + prevent migration to and from ConBee/RaspBee:

* https://www.zigbee2mqtt.io/guide/adapters/deconz.html

While I can only speak for myself, think I can still say that as it stands right now with important features such as backup and restore missing, many of us who are a part of the Zigbee2MQTT and overall Zigbee open source projects communities can currently not with good conscience recommend ConBee II / RaspBee II or even the ConBee III to new Zigbee2MQTT end-users. 

* https://www.zigbee2mqtt.io/guide/usage/mqtt_topics_and_messages.html#zigbee2mqtt-bridge-request

vsatmydynipnet, 2024-09-16:
I am in the same position.  Running Z2M with a Conbee II having the new Conbee II already on my desk. But redoo all of  the Z2M setup of all devices is not the way to go.  Where is the problem to not be able backup/restore from old to new stick?

Hedda, 2024-09-16:
> Where is the problem to not be able backup/restore from old to new stick?

The reason is explained in detail in the original feature request issue above (at the top ) here -> https://github.com/Koenkk/zigbee-herdsman/issues/665 but as a short summary; while the deconz serial protocol can be used to achieve backup and restore such a feature is not yet supported for deconz based adapters like ConBee/RaspBee in Zigbee2MQTT for the simple reason that no developer yet programmed support for it to the zigbee-herdsman's code for its deconz driver/adapter -> https://github.com/Koenkk/zigbee-herdsman/tree/master/src/adapter/deconz 

Until this is added end-users have the option to use the unofficial third-party workaround discussed in this other thread that you can find here -> https://github.com/Koenkk/zigbee2mqtt/discussions/16478 (but please respect that end-user usage of that is off-topic here but end-users are welcome to discuss that there is you like).

PS: Please take note and remember that this is a free and open-source hobby project so it is mainly only worked on by volunteers.

Hedda, 2025-01-26:
Still valid issue I believe?

Hedda, 2025-05-28:
Still an open issue as far as I know, but I no longer got a ConBee or RaspBee adapter to test.

closed 2025-07-04 by Nerivec