Koenkk/zigbee-herdsman #677: link_quality is always 255

gemo-bi, 2023-03-06:
Hello, the link_quality of my devices are 255 or 10
Zigbee-Adapter 1.8.10
Nodejs: v18.14.2
few days ago it still worked. Is it related to a firmware update from the Conbee2 stick?

debug-log in ioBroker:
zigbee.0 | 2023-02-28 18:55:13.671 | warn | ELEVATED publishToState: value generated '255' from device a4c13804e80ab84a for 'Link quality'

asgothian's commented:
This is an issue with the implementation of the deconz hardware in the zigbee-herdsman. There is nothing the zigbee adapter can do to fix this.

FredWst, 2023-03-20:
Hi,

I've the same with 2 dongles ZDongle-E sonoff and SkyConnect.
with NCP firmware all fine but with RCP firmware had 255 LQI on all devices.

mamrai1, 2023-03-21:
> 

but ZBDongle-E has no hardware cts/rts. Does it?

ksjh, 2023-03-21:
It has. Not sure the original Sonoff Firmware supports hardware handshaking, but the ZBDongle-E uses Port D Pin 3 as CTS and Port D Pin 4 as RTS, so you can flash a firmware which supports RTS-CTS handshaking.

FredWst, 2023-03-21:
Hi,

I think it has according to : https://github.com/ksjh/silabs-firmware-builder/issues/4

Fred

MattWestb, 2023-03-28:
Flow control hard or software is only blocking then the buffer is being full and normally is not happening so you need filling up the buffer for see if its haling sending or you is loosing packages.

ksjh, 2023-03-28:
So the hardware is luckily implemented in a way that the software sees both RTS and CTS as asserted. Good. If one or both would always be seen as cleared (not set), the communication would never start.

ksjh, 2023-03-28:
@MattWestb great, thank you for quick answer. I will experiment a bit. I was aware of the 250 kbit/s in IEEE 802.15.4 per channel, but uncertain how the RTS/CTS is used in this implementation. I will experiment a bit more with higher data rates, since in normal cases, I expect much more messages from the wireless side to the host than from the host to the wireless side ("air interface") except for special occasions like OTA updates. My CH9102F on the ZBDongle-E should support data rates of up to 4 Mbps if they did not mess up the routing of the board. I mean, it does not look great, 2 vias each, a few centimeters track, but mainly separated through a ground fill, but also not awful. So, I expect 460800 baud should be not that much of a problem.

mamrai1, 2023-03-28:
@ksjh just tried the ncp firmware for zbdongle-none but it doesn't connect at all with z2m.. Had to manually put it in bootloader to reflash it.. 'None' is supposed to be software flow control or none?

ksjh, 2023-03-28:
@mamrai1 The zbdongle-none uses ``uartdrvFlowControlNone``. Some other person tried it, and for them, it basically worked (please note: 115200 baud, also when flashing from a running application, i.e., not bootloader).
But may I suggest that **we continue this discussion somewhere where it is more relevant**, i.e. [here](https://github.com/ksjh/silabs-firmware-builder/discussions) or [here](https://github.com/ksjh/silabs-firmware-builder/issues)?
**Thank you!**

MattWestb, 2023-03-28:
Sky Connect can running without problem with 460800 if the host side have OK USB implantation. 
And you mus calculating the overhead in every layer of all protocols for knowing where the bottleneck is and one user with Zigbee and OTBR have have problems with 60 OT devices with 115200 but the Zigbee is working OK at the same time.

HA have changing the standard for RCP to 460800  for not getting problems with much IPV6 traffic to and from the 802.15.4 mesh network over the comport bottleneck.

ksjh, 2023-03-28:
Of course, you are right, there might be overhead from other layers, especially with ``zigbeed`` and the resulting encapsulation, the ballpark figures often provide a good enogh estimate (plumbing analysis). But I know, you are right, I should be more careful here, especially since I work at a university in the field of network performance evaluation...

MattWestb, 2023-03-28:
Then you shall knowing how the flow control is working and can doing nice performance test on the 802.15.4 network :-))

ksjh, 2023-03-28:
@MattWestb Would be a nice project.
The flow control just confused me since I was not expecting that unconnected RTS and CTS pins in this hardware implementation are regarded as asserted, i.e., the signals are active low and the pins pulled to ground. If the manufacturer would have them left floating, communication would not have worked and I would have noticed earlier.

darkxst, 2023-03-28:
> eless side to the host than from the host to the wireless side ("air interface") except for special occasions like OTA updates. My CH9102F on the ZBDongle-E should support data rates of up to 4 Mbps [...]

In terms of RF design 4mbps is still very slow, you generally dont need to worry too much about impedance matching and all that fun transmission line stuff, like you would with say the differential pairs of High Speed USB.

closed 2023-05-06 by github-actions[bot]