Koenkk/zigbee-herdsman #686: Unable To Send Commands To Clusters Which Share Same ID

danieledwardgeorgehitchcock, 2023-03-28:
There is an existing cluster in Herdsman `manuSpecificUbisysDeviceSetup` which has a cluster id  of `64512`

I created a new cluster `manuSpecificAssaDoorLock` which shares the same cluster id of `64512` or Hex `0xFC00`

These are manufacturer specific clusters and therefore do not share anything in common, other than their id's.

When trying to send a command to `manuSpecificAssaDoorLock` using the cluster id (as the cluster name cannot be used as far as I am aware), I get an error response stating that the cluster `manuSpecificUbisysDeviceSetup` has no command `X`.

This presents an interesting problem, as these manufacturer specific clusters can have overlapping command id's that have completely different functions so, cannot be merged together.

Is there a way to control which cluster Herdsman (and Z2M) uses?

Koenkk, 2023-03-28:
This is indeed a known issue, but why not just use the name instead of ID?

danieledwardgeorgehitchcock, 2023-03-28:
The frontend (I don't believe) allows for this so I thought this might be a limitation with ZH.. If it isn't, then I will raise it as an issue in that project instead..

Koenkk, 2023-03-28:
It would be better if we indeed allow specifying the cluster name instead of ID in the frontend.

Koenkk, 2023-03-29:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-03-29 by Koenkk