Koenkk/zigbee-herdsman #690: Is it possible to create multiple controllers with this lib?

willieee802, 2023-04-03:
As stated, is it possbile to create multiple instances of class Controller, giving different params and start them at the same time, and maintaining all instances in my program? Is it technically possible with this library ?
Thanks in advance.

willieee802, 2023-04-03:
Currently if I create multiple Controllers, they can all start, but if I start a second one. The first one becomes slient, no messages emitted if I toggle devices paired on first controller, while the second controller works fine.

Koenkk, 2023-04-03:
You can, but you will need to use multiple dongles.

willieee802, 2023-04-03:
> You can, but you will need to use multiple dongles.

This is what I am currently doing. I got multiple EZSP controllers via IP. They all works fine independently. But if I create multiple Controller() with this lib, with each instance connects to a seperate EZSP zigbee gateway.
Once I start the second controller, the first one always stopped receiving message event and cannot send control either. Just wanna know if this is a limitation of the lib, which just does not work with multiple controllers at same time. Or it may relates to hardware or firmware of my ezsp adapter, or anything I might did wrong using the library
Many thanks.

Koenkk, 2023-04-04:
This should be possible, it might help to check the herdsman debug logging for any pointers. (https://www.zigbee2mqtt.io/guide/usage/debug.html)

willieee802, 2023-04-04:
> This should be possible, it might help to check the herdsman debug logging for any pointers. (https://www.zigbee2mqtt.io/guide/usage/debug.html)

Debug mode does give me quite a lot more info. I can see no message event is emitted because the Device.byNetworkAddr() is not returning the device, even though the zcl data payload is valid(looks same before and after creatinga second controller), so the method that processes zcl data stops here.

I can see the Entity class maintains the Database instance, but it is static. While Device extends Entity, it accesses database via Entity.database. When creating a Controller, it calls Entity.injectDatabase, to keep the database instance, so creating a second controller overrides the first controller's database instance, then Device.byNetworkAddr() does not work because it cannot find the device from the second controller's database, which is empty.
Looks like only one database is maintaned, is that correct?

Koenkk, 2023-04-05:
You should use unique database paths for each instance (https://github.com/Koenkk/zigbee-herdsman/blob/3dd461166bba970a6ac87e2bc1782b512a3dc48c/src/controller/controller.ts#L25)

willieee802, 2023-04-06:
Yes that is also what I am currently doing. I can see respective database files are created. What I was saying is that, every time I created a new Controller() instance, with different database paths passed in, it does create the respective db files, but it then just saved the db instance in Entity class as static member (
https://github.com/Koenkk/zigbee-herdsman/blob/61b701e8826d0120833a5485ebd595d4a2a5cfe9/src/controller/controller.ts#L126
https://github.com/Koenkk/zigbee-herdsman/blob/61b701e8826d0120833a5485ebd595d4a2a5cfe9/src/controller/model/entity.ts#L6
)

So event though it created unique db files via Database.open() (
https://github.com/Koenkk/zigbee-herdsman/blob/61b701e8826d0120833a5485ebd595d4a2a5cfe9/src/controller/controller.ts#L125
),  
only one is maintained by Entity class, and being used by say, the Device class via Entity.database (
https://github.com/Koenkk/zigbee-herdsman/blob/61b701e8826d0120833a5485ebd595d4a2a5cfe9/src/controller/model/device.ts#L366
), that's why the lastly created controller instance always works, but not previous ones.

Koenkk, 2023-04-06:
Ah I see now, then creating multiple controller instances is not supported, you have to create them in separate processes.

closed 2023-05-15 by github-actions[bot]