Koenkk/zigbee-herdsman #692: deconz ConBee/RaspBee adapters - probe the protocol version

famzah, 2023-04-06:
Hello,

I want to get the "last hop address" on APS_DATA_INDICATION but this feature is supported "since protocol version 0x0108" as described in the [deCONZ Serial Protocol](https://deconz.dresden-elektronik.de/raspbian/deCONZ-Serial-Protocol-en_1.21.pdf) docs. While it's easy to get and use the protocol version value, and I can code this myself, I don't have enough understanding of the code base in "zigbee-herdsman" to decide how to share the protocol version value between the different components:
- driver/driver.ts -- here we set the flag for "last hop address" if protocol version is 0x0108 or greater
- driver/frameParser.ts -- here we parse the "last hop address" if protocol version is 0x0108 or greater
- adapter/deconzAdapter.ts -- here in getCoordinator() we should probably get the protocol version (along with "ieeeAddr" and "nwkAddr"), so that we can store the version value and later use it in "driver/driver.ts" and "driver/frameParser.ts"

My current patch which doesn't check if the protocol version is 0x0108 or greater looks like this:
```diff
diff -u -r deconz-backup/driver/driver.js deconz/driver/driver.js
--- deconz-backup/driver/driver.js      2022-09-22 22:37:29.082154562 +0300
+++ deconz/driver/driver.js     2023-04-05 23:10:43.678988803 +0300
@@ -597,7 +597,7 @@
     sendReadReceivedDataRequest(seqNumber) {
         debug(`DATA_INDICATION - sending read data request - SeqNr. ${seqNumber}`);
         // payloadlength = 0, flag = none
-        const requestFrame = [constants_1.default.PARAM.APS.DATA_INDICATION, seqNumber, 0x00, 0x08, 0x00, 0x01, 0x00, 0x01];
+        const requestFrame = [constants_1.default.PARAM.APS.DATA_INDICATION, seqNumber, 0x00, 0x08, 0x00, 0x01, 0x00, 0x03];
         this.sendRequest(requestFrame);
     }
     sendEnqueueSendDataRequest(request, seqNumber) {

diff -u -r deconz-backup/driver/frameParser.js deconz/driver/frameParser.js
--- deconz-backup/driver/frameParser.js 2022-09-22 22:37:29.512137113 +0300
+++ deconz/driver/frameParser.js        2023-04-05 23:58:25.391089996 +0300
@@ -239,14 +239,17 @@
             i++;
         }
         response.asduPayload = payload;
+        response.lasthopAddr16 = view.getUint16(view.byteLength - 10, littleEndian);
+        let lasthopAddr = response.lasthopAddr16.toString(16);
         response.lqi = view.getUint8(view.byteLength - 8);
         response.rssi = view.getInt8(view.byteLength - 3);
         let newStatus = response.deviceState.toString(2);
         for (let l = 0; l <= (8 - newStatus.length); l++) {
             newStatus = "0" + newStatus;
         }
         debug("DATA_INDICATION RESPONSE - seqNr. " + response.seqNr + " srcAddr: 0x" + srcAddr + " destAddr: 0x" + destAddr + " profile id: 0x" + response.profileId.toString(16) + " cluster id: 0x" + response.clusterId.toString(16) + " lqi: " + response.lqi);
         debug("response payload: " + payload);
+        console.log("DATA frame routing info: srcAddr: 0x" + srcAddr + " destAddr: 0x" + destAddr + " lqi: " + response.lqi + " lasthopAddr: 0x" + lasthopAddr);
         frameParserEvents.emit('receivedDataPayload', response);
         frameParserEvents.emit('receivedDataNotification', response.deviceState);
         return response;
```

Currently I have the following questions:
- Which function is the the best place to get the protocol version, so that we can later use its value? It should be one of the one-time initializing states.
- How exactly in terms of code style is the best way to share the protocol version value between the mentioned functions?
- Do we ever need to probe this, or protocol version 0x0108 is so old that we can assume that all devices today already support it?

Koenkk, 2023-04-06:
> Which function is the the best place to get the protocol version, so that we can later use its value? It should be one of the one-time initializing states.

https://github.com/Koenkk/zigbee-herdsman/blob/61b701e8826d0120833a5485ebd595d4a2a5cfe9/src/adapter/deconz/driver/driver.ts#L182

> How exactly in terms of code style is the best way to share the protocol version value between the mentioned functions?

Since frameparser is not a class, easiest things is to pass it as an argument to the functions where you need it

> Do we ever need to probe this, or protocol version 0x0108 is so old that we can assume that all devices today already support it?

How old is 0x0108?

Question to you: what is your use case and how do you expect to expose this to the user?

famzah, 2023-04-06:
This feature can be used to get real-time information about the map of your Zigbee network. It turns out that ConBee II doesn't properly list all end-devices connected to it (the coordinator) - see this [bug report](https://forum.phoscon.de/t/lqi-query-doesnt-show-end-devices-in-z2m/3475). But if we look into the "last hop address" of each packet, we know if it was a direct communication with the coordinator or if it was routed through one of the routers in the network (we also know the address of the last router).

The information can be logged if we request debugging for this particular info in "zigbee-herdsman" by the DEBUG environment variable. A separate debug tag could be added.

P.S. I have no idea how old 0x0108 is. I was hoping that somebody here may have this knowledge.

famzah, 2023-04-08:
I'll close this for now until I find time to implement it properly and do a PR. Thank you for your clarifications.

closed 2023-04-08 by famzah