Koenkk/zigbee-herdsman #750: Does it support WTSK + radio board in NCP mode instead of USB-doggles?

GanesanGuru123, 2023-08-23:
Hi fellas,

As I'm having WTSK + radio board (xGM210P032), I can operate them in NCP mode with Z3Gateway example in RPI as NCP host. As in USB dongle, too its gonna work in NCP & linux xystm/RPi as gonna act as NCP host, can I use WTSK+radio setup to achieve the same? If so, whether I should use any dedicated firmware to xGM210P032?

As of now, I've dumped xGM210P032 with _ncp-uart-hw_ and RPi/NCP host with _Z3Gateway_.When I tried to install **Zigbee2MQTT**, I endedup with **Error while starting zigbee-herdsman** and along with it **Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)**. 

Any suggestions?

kirovilya, 2023-08-23:
add to configuration:
`adapter: ezsp`

GanesanGuru, 2023-08-23:
> add to configuration: `adapter: ezsp`

Thanks a lot @kirovilya

closed 2023-09-30 by github-actions[bot]