Koenkk/zigbee-herdsman #765: Sudden failure to startup - timeout after SRSP - ZDO - startupFromApp

thargy, 2023-09-19:
## Symptom

Home Assistant Z2M addon no longer starts, reporting `Error: SRSP - ZDO - startupFromApp after 40000ms`.  No changes were made that could explain the sudden fatal failure.  Multiple hardware reboots have failed to have an impact.

## Description

I've been running two instances of Z2m for about six months on two separate machines (an Intel NUC as a HAOS addon and a Rasberry Pi as a standalone Z2m instance) to allow me to run with two separate channels.  Today at around 13:50, my Z2m HAOS addon instance on the Intel NUC died and refuses to restart.  I had made no changes and was not interacting with Home Assistant or the hardware when it happened, and I have only been able to deduce the failure time from when various devices became 'unavailable' to HA.  I did trigger a reboot around that time, though.

Sadly, my addon wasn't configured for Herdsman logging nor SSH access via 22222.  By the time I got it all set up, the initial logs were gone.  However, I eventually managed to grab the full log below.

I've checked, and my Sonoff Zigbee 3.0 USB Dongle Plus is still connected to the correct USB port. From the logs, there appears to be initial communication until it gets to `SRSP - ZDO - startupFromApp`, at which point it times out, and the docker container is unloaded (making grabbing logs difficult).

Have you any idea what may be causing this issue?  I have a very large network (~150 devices), and rebuilding it would be a nightmare.  

I have a coordinator backup from last night, so I could try restoring that to the dongle and upgrading to coordinator revision 20230507, which I've been running on the second dongle for several months.  However, I wanted to let you look first and see if there were any logs or debug steps you'd like me to try.  However, I can no longer control any of my house lights or automations, so I would appreciate any help you can provide quickly.

## Data

<details>
<summary>System Properties for Intel NUC HA addon instance</summary>

| Property  | Value  |
| - | - |
| Hardware | Beelink EQ12 Pro Intel 12th N305 |
| RAM | 16G (DDR5) |
| Zigbee2MQTT version | 1.33.0-1 |
| Coordinator | Sonoff Zigbee 3.0 USB Dongle Plus |
| Coordinator Chip | CC2652P |
| Coordinator type | zStack3x0 |
| Coordinator revision | 20221226|
</details>

<details>
<summary>Docker Log (up to timeout)</summary>

```
[22:37:26] INFO: Preparing to start...
[22:37:26] INFO: Socat not enabled
[22:37:26] INFO: Zigbee Herdsman debug logging enabled
[22:37:26] INFO: Starting Zigbee2MQTT...
2023-09-19T21:37:28.106Z zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'
2023-09-19T21:37:28.107Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyUSB0"},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
2023-09-19T21:37:28.108Z zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with {"path":"/dev/ttyUSB0","baudRate":115200,"rtscts":false,"autoOpen":false}
2023-09-19T21:37:28.111Z zigbee-herdsman:adapter:zStack:znp:log Serialport opened
2023-09-19T21:37:28.112Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2023-09-19T21:37:28.112Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2023-09-19T21:37:28.365Z zigbee-herdsman:adapter:zStack:znp:log Writing CC2530/CC2531 skip bootloader payload
2023-09-19T21:37:28.365Z zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239]
2023-09-19T21:37:29.367Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2023-09-19T21:37:29.368Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2023-09-19T21:37:29.620Z zigbee-herdsman:adapter:zStack:znp:log Skip bootloader for CC2652/CC1352
2023-09-19T21:37:29.943Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [0]
2023-09-19T21:37:29.944Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [0]
2023-09-19T21:37:30.077Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2023-09-19T21:37:30.078Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2023-09-19T21:37:32.981Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,6,65,128,0,2,1,2,7,1,192]
2023-09-19T21:37:32.981Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [0,254,6,65,128,0,2,1,2,7,1,192]
2023-09-19T21:37:32.983Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 6 - 2 - 1 - 128 - [0,2,1,2,7,1] - 192
2023-09-19T21:37:32.984Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- SYS - resetInd - {"reason":0,"transportrev":2,"productid":1,"majorrel":2,"minorrel":7,"hwrev":1}
2023-09-19T21:37:32.985Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-09-19T21:37:36.080Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2023-09-19T21:37:36.080Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2023-09-19T21:37:36.084Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,2,97,1,89,6,61]
2023-09-19T21:37:36.084Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,2,97,1,89,6,61]
2023-09-19T21:37:36.084Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 2 - 3 - 1 - 1 - [89,6] - 61
2023-09-19T21:37:36.085Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - ping - {"capabilities":1625}
2023-09-19T21:37:36.085Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-09-19T21:37:36.086Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - version - {}
2023-09-19T21:37:36.086Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,2,35]
2023-09-19T21:37:36.089Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,10,97,2,2,1,2,7,1,42,141,52,1,0,252]
[... 105 lines cut ...]
2023-09-19T21:37:36.264Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,14,103,0,0,139,7,126,37,0,75,18,0,254,255,7,0,0,225]
2023-09-19T21:37:36.264Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 14 - 3 - 7 - 0 - [0,139,7,126,37,0,75,18,0,254,255,7,0,0] - 225
2023-09-19T21:37:36.265Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- UTIL - getDeviceInfo - {"status":0,"ieeeaddr":"0x00124b00257e078b","shortaddr":65534,"devicetype":7,"devicestate":0,"numassocdevices":0,"assocdeviceslist":[]}
2023-09-19T21:37:36.265Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-09-19T21:37:36.265Z zigbee-herdsman:adapter:zStack:startup starting adapter as coordinator
2023-09-19T21:37:36.265Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - startupFromApp - {"startdelay":100}
2023-09-19T21:37:36.265Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,2,37,64,100,0,3]
```
</details>

<details>
<summary>Z2M Log (on timeout)</summary>

```
error 2023-09-19 21:38:16: Error while starting zigbee-herdsman
error 2023-09-19 21:38:16: Failed to start zigbee
error 2023-09-19 21:38:16: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
error 2023-09-19 21:38:16: Exiting...
error 2023-09-19 21:38:16: Error: SRSP - ZDO - startupFromApp after 40000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:559:17)
    at processTimers (node:internal/timers:502:7)

```
</details>

thargy, 2023-09-19:
OK, so being an unforgivably impatient sort, I backed up the coordinator and updated to 20230507.  This went flawlessly, but I had exactly the same issue on starting the addon.

You can see from the log that the firmware has updated:

<details>
<summary>
Docker Log
</summary>

```
[23:54:07] INFO: Preparing to start...                                                                                                                                                                                                                                                                     
[23:54:07] INFO: Socat not enabled                                                                                                                                                                                                                                                                         
[23:54:07] INFO: Zigbee Herdsman debug logging enabled                                                                                                                                                                                                                                                     
[23:54:07] INFO: Starting Zigbee2MQTT...                                                                                                                                                                                                                                                                   
2023-09-19T22:54:09.106Z zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'                                                                                                                                                                                                    
2023-09-19T22:54:09.107Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyUSB0"},"datab
asePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'                                                                        
2023-09-19T22:54:09.107Z zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with {"path":"/dev/ttyUSB0","baudRate":115200,"rtscts":false,"autoOpen":false}                                                                                                                                          
2023-09-19T22:54:09.111Z zigbee-herdsman:adapter:zStack:znp:log Serialport opened                                                                                                                                                                                                                          
2023-09-19T22:54:09.112Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}                                                                                                                                                                                                       
2023-09-19T22:54:09.112Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]                                                                                                                                                                                                              
2023-09-19T22:54:09.364Z zigbee-herdsman:adapter:zStack:znp:log Writing CC2530/CC2531 skip bootloader payload                                                                                                                                                                                              
2023-09-19T22:54:09.365Z zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239]                                                                                                                                                                                                                       
2023-09-19T22:54:10.368Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}                                                                                                                                                                                                       
2023-09-19T22:54:10.368Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]                                                                                                                                                                                                              
2023-09-19T22:54:10.619Z zigbee-herdsman:adapter:zStack:znp:log Skip bootloader for CC2652/CC1352                                                                                                                                                                                                          
2023-09-19T22:54:11.077Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}                                                                                                                                                                                                       
[... cut ...]
However, whilst I was coming here to post, the watchdog started it again, and this time it booted up as usual!!!

I have restarted several times and turned off herdsman logging to see if it survives restarts, and it does.  However, out of 157 devices, only 58 have reported back in so far, and it seems to have stalled at that number, so I need to give it some time to see if I can get the full network to rebuild.

<details>
<summary>
Docker Log (truncated after startupFromApp succeeds)
</summary>

```
```
</details>

thargy, 2023-09-20:
Update: Z2M ran for approx. 12 hours, it was mostly issuing regular warnings such as:
```
warn  2023-09-20 14:23:44: Failed to ping 'Hallway Rear Spot 1' (attempt 1/1, Read 0x001788010d974eb5/11 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))
```
As a number of devices remained 'offline' to Z2m (but were responding to group messages).  About 2/3rds of devices were communicating OK, the rest were responding to the groups (even if showing as 'No Network route' to Z2m.

After 12 hours the errors started appearing in the logs, e.g.
```
error 2023-09-20 14:35:34: Publish 'set' 'state' to 'Bedroom 2 Spots' failed: 'Error: Command 7 genLevelCtrl.moveToLevelWithOnOff({"level":0,"transtime":50}) failed (SRSP - AF - dataRequestExt after 6000ms)'
```

About 3-4 hours later devices stopped responding entirely and went offline, and pretty much every error and warning was for `(SRSP - AF - dataRequestExt after 6000ms)`

I noticed about 3 hours later and attempted to restart Z2m, at which point the above failure started and z2m is back to the refusing to start state, due to the timeout `Error: SRSP - ZDO - startupFromApp after 40000ms`.  I haven't included a log as it's essentially identical to the first log seen in the [above comment](https://github.com/Koenkk/zigbee-herdsman/issues/765#issuecomment-1726670954).

thargy, 2023-09-20:
OK, I updated to 20230901 to see if that improved things.  The first start of the addon caused the same error, but, as with the last time I updated the firmware, it succeeded on the second attempt.

thargy, 2023-09-20:
So, a reboot after about 2 hours caused it to go back into the failed state.  After multiple unsuccessful reboot attempts, I again reflashed the firmware and restored the network backup.  However, this time, it failed to reboot after multiple attempts, so the 'pattern' of succeeding on the 2nd attempt after a flash was a coincidence and probably more related to time passing.

I've set the HAOS addon watchdog back on so that it repeatedly attempts a reboot, to see if it eventually succeeds and will update this comment.

thargy, 2023-09-21:
> Could it be that there are 2 zigbee networks in your house?

Yes, as I mentioned in the description:
> I've been running two instances of Z2m for about six months on two separate machines (an Intel NUC as a HAOS addon and a Rasberry Pi as a standalone Z2m instance) to allow me to run with two separate channels.

The primary network runs on channel 11 and has 159 devices.  Z2m runs as a Home Assistant Add-on on an Intel NUC.
The secondary network runs on channel 12 and has 17 'noisy' sensors.  Z2m runs directly on a Rasberry Pi as the only service.

I split the networks due to the issues caused by the 17 'noisy' presence sensors that made it unstable.  It has been relatively stable for about four months with this setup.

I also have a Philips Hue bridge with a few lights attached (about 4 or 5).  I don't know what channel it runs on, but it's coordinator is on the other side of the house.

They added a 'smart' meter to my electricity supply about a month ago.  However, it doesn't work due to an inability to sync to the radio network.  As such, I haven't set up the monitor.  It too, uses Zigbee, but I don't believe it is active.

thargy, 2023-09-21:
Z2m booted up again after about 9 hours this morning, and has been running since, though only 100 of 159 devices are 'online'.  I have ordered some new coordinators as backups, and am crossing my fingers I can keep the network up in the meantime.  Do you have any suggestions as to why it timeouts at this point?

Koenkk, 2023-09-21:
I suggest to downgrade to 20221226 (https://github.com/Koenkk/Z-Stack-firmware/tree/Z-Stack_3.x.0_coordinator_20221226/coordinator/Z-Stack_3.x.0/bin), problems have been reported with 20230507: https://github.com/Koenkk/Z-Stack-firmware/discussions/474

thargy, 2023-09-21:
> I suggest to downgrade to 20221226

Yeah, sorry, being Autistic I'm overly verbose, so I don't blame you for not reading all the walls of text.  I just wanted to give you all the info.

You'll see from the details on the first comment, that the problems started whilst it wa running 20221226.

I am running 20230507 on the second coordinator (which isn't showing this issue).  I've never run it on the primary.

I upgraded the primary to 20230901 as that's effectively the same as 20221226 just a newer SDK.

The symptom has been the same on both 20221226 and 20230901.  I've included logs showing the failure on both versions.

Hope that clears things up.

Koenkk, 2023-09-21:
If I recall correctly, `startupFromApp` can hang when there are conflicting networks. To test if this is the case, can you try wrapping the adapter in thin foil and try to start it?

thargy, 2023-09-22:
> If I recall correctly, `startupFromApp` can hang when there are conflicting networks. To test if this is the case, can you try wrapping the adapter in thin foil and try to start it?

So wrapping the adapter in tin toil, and removing the antennae allows Z2M to start, but _all_ communications to devices fail with `failed (SRSP - AF - dataRequestExt after 6000ms)'`, so it doesn't get me anywhere, except suggest a conflicting network.  That's strange as it's been stable for months, but it is channel 11 which is the most commonly used I suppose.

The prospect of changing to Channel 25 and repairing 150+ devices is more than a little daunting, especially as the vast majority are Philips hue bulbs, and they would require unpairing first... which would be a challenge.  Some of the lights are in very inaccessible places too.

So, I then replaced my Sonoff Dongle with a new one straight from the box and flashed with 20221226 and the network backup.  It too failed to start with `Error: SRSP - ZDO - startupFromApp after 40000ms` - effectively ruling out bad NVRAM, or a faulty device.

The real issue is how to identify the culprit that's stomping on my zigbee channel...

thargy, 2023-09-22:
So, I unplugged my Philips Hue hub (which is 2 floors away) and the second coordinator (which is supposed to be on Channel 12).  I can't physically disconnect the Smart Meter, but it's in the garage and has no devices connected to it.  I can't disconnect my Google Wifi Pro 6 (which was the worst purchase decision ever btw) and I can't see what channel it's running on, never mind change it (because it's SH*T), but I would expect Channel 12 to be stomped on too if the issue was Wifi related.

None of these steps made any difference.

I am running the coordinator on a USB extension cord already, and, as mentioned, it's been relatively stable until 3 days ago.

thargy, 2023-09-23:
UPDATE:

* Not hardware fault - I have now swapped USB wires, ports, and coordinator hardware and confirmed the issue remains exclusive to the primary network, and not the secondary network.

* Not Wifi - Further, as the secondary network is Channel 12, and the primary is on Channel 11, it is unlikely that Wifi is causing the issue (as it would stomp over 12 even more than 11).  

* Unlikely to be other Zigbee network - I'm in quite a remote location and my only neighbour barely has wifi.  As such a competing network is unlikely.

* Likely to be too many devices/a device on network - However, after 18 hours of being unable to start Z2M I switched off all my light circuits and rebooted, this time Z2M started straight away.  I could then bring up each light circuit separately and wait to stabilize (which is what I'm in the process of doing now).

So my working theory is one of:
* There is a device(s) on my network that does something when the coordinator initially starts that causes the startup to fail.
* The coordinator does not handle starting in a large network well.

I haven't got a sniffer, many of the CC2531 modules I can see require a separate programmer board that I'm not sure how to source or use.

Koenkk, 2023-09-23:
Since it is starting now but the coordinator is crashing, can you try 20221226 again? (it's seems to be the most stable fw for now based on the feedback of many users)

thargy, 2023-09-23:
> Since it is starting now but the coordinator is crashing, can you try 20221226 again? (it's seems to be the most stable fw for now based on the feedback of many users)

I'm not sure what that helps demonstrate? As I've already stated, the problems started when I was on 20221226 already (and had been for about 6 months.  However, I did as asked and, sure enough, it wouldn't start up again at all, until I switched off all my light circuits (unpowering about 130 devices).  However, switching on the circuits after a reboot almost immediately caused the dreaded `failed (SRSP - AF - dataRequest after 6000ms))` to start - i.e.

```
[20:35:00] INFO: Preparing to start...
[20:35:00] INFO: Socat not enabled
[20:35:00] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:warn  2023-09-23 20:35:45: Failed to ping 'Balcony Spot 2' (attempt 1/1, Read 0x001788010cf4f4cb/11 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))
Zigbee2MQTT:warn  2023-09-23 20:35:50: Failed to ping 'Garage Switch Right' (attempt 1/1, Read 0x54ef4410005c8290/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))
Zigbee2MQTT:warn  2023-09-23 20:35:57: Failed to ping 'Bedroom 1 Spot 2' (attempt 1/1, Read 0x001788010cf2df14/11 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)))
Zigbee2MQTT:error 2023-09-23 20:36:07: Publish 'set' 'brightness' to 'Kitchen Spots' failed: 'Error: Command 22 genLevelCtrl.moveToLevelWithOnOff({"level":163,"transtime":10}) failed (SRSP - AF - dataRequestExt after 6000ms)'
Zigbee2MQTT:warn  2023-09-23 20:36:13: Failed to ping 'Bedroom 1 Spot 4' (attempt 1/1, Read 0x001788010cf6d54b/11 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms))
Zigbee2MQTT:error 2023-09-23 20:36:19: Publish 'set' 'color_temp' to 'Kitchen Spots' failed: 'Error: Command 22 lightingColorCtrl.moveToColorTemp({"colortemp":400,"transtime":10}) failed (SRSP - AF - dataRequestExt after 6000ms)'
Zigbee2MQTT:warn  2023-09-23 20:36:25: Failed to ping 'Bedroom 1 Spot 5' (attempt 1/1, Read 0x001788010cf2d2db/11 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms))
Zigbee2MQTT:warn  2023-09-23 20:36:36: Failed to ping 'Bedroom 1 Spot 6' (attempt 1/1, Read 0x001788010cf6c741/11 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms))
Zigbee2MQTT:warn  2023-09-23 20:36:47: Failed to ping 'Bedroom 1 Spot 7' (attempt 1/1, Read 0x001788010cf69b23/11 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms))
```

thargy, 2023-09-26:
After a few more days of testing, I identified a couple of Moes Thermostats that are super chatty and moved them to my secondary network. This improved things _slighty_, but I'm still having the same problems getting the network to start.

What I'm seeing is that, when busy, Z2M continues to send messages into an already busy environment, which causes a cascade failure, resulting in `BUFFER_FULL`, timeouts (`AF`), etc.  This is particularly the case when the network is starting up.  Sometimes, the coordinator can slowly recover from this state, but usually, it is fatal.

Because of this, particularly chatty devices, like sensors, are problematic.  Also, the number of pings and messages being spammed by Z2M on startup (like querying the device state when it first sees the device on startup) contributes to the issue of starting up stably in a large network environment.

I've started looking through the code base to see if there's any throttling ability built into sending messages to the coordinator based on the current frames/second.  I suspect such a feature would be really useful and lead to a much more stable coordinator in busy environments.

What would be good is for the herdsman to track the current reception and transmission rate (preferably in kb/s, but even msg/s would be useful). Exposing these two figures would be useful for debugging.

With these two rates, it would then be good to update the waitress to delay sending messages until the combined rate is below a configurable threshold.  This would reduce pressure on the coordinator and prevent contributing to the saturated network environment.  It is better for the message to timeout higher up the stack than on the network itself.

As a final stage, adding a priority to message types would allow messages such as pings, OTA updates, etc., to be placed at a lower priority.  Also, splitting the timeout into two - the timeout after the initial send (when the waitress receives the message) and the timeout after the actual transmit (when it releases it to the coordinator after the network saturation is deemed low enough) would allow messages such as availability pings to be given a low priority but a longer send timeout whilst keeping the transmit timeout short.

Later, Z2M itself could use the network saturation data to determine optimum times to do maintenance work (like querying initial state, pinging availability, etc.), helping to balance the network load.

What do you think as a way forward to support larger networks?

TL;DR
I propose adding the following features (in order):

1. Report transmit and receive rates (expose via UI, MQTT, logs).
2. Allow configuration of a maximum network saturation (based on combined transmit & receive rate), and update waitress only to send messages whilst the network is below the configured rate.
3. Add message priorities and update the waitress to send messages based on priority (now that it can delay sending a message to the coordinator based on network saturation).
4. Split timeouts into timeout from initial send, which should be greater than or equal to timeout from transmit.  This allows low-priority messages to sit in the waitress queue for longer without hitting their timeout.
5. Utilise network saturation data within Z2M to create more advanced strategies around availability and state checking to help reduce network load.

Koenkk, 2023-09-26:
You can configure the amount of concurrent messages, see https://www.zigbee2mqtt.io/guide/configuration/adapter-settings.html#mdns-zeroconf-discovery

thargy, 2023-09-26:
I'll try reducing to 8 as that is half the recommended value, and see what happens.

closed 2023-11-03 by github-actions[bot]

WhistleMaster, 2024-03-26:
I have a similar issue https://github.com/Koenkk/zigbee2mqtt/issues/21962. Did you figure out what the problem was ? Thanks !